pvvx/THB2 #1: Tuya BLE sensor: BTH01

user784213, 2023-12-27:
Hello
I have a very similar sensor: **BTH01**
Marking on SoC: PHY6222QCW04IYOJSH (PHY6222QC-W04C)
Marking on sensor: 8305 ([CHT8305C](https://www.sensylink.com/upload/1/net.sensylink.portal/1689159407292.pdf))
The case looks cheap, same as TS0202, disassembles very easily
Current price on Aliexpress is less than $2
[image]

pvvx, 2023-12-27:
Разработка PHY6222 только началась.
Пока не будет ОТА, не будет и новых прошивок.

Development for PHY6222 has just begun.
As long as there is no OTA, there will be no new firmware.

pvvx, 2023-12-28:
[CHT8305](https://unagidojyou.com/document/2023/03/CHT8305.pdf)
https://github.com/pvvx/ZigbeeTLc/blob/master/src/sensor_cht8305.c

pvvx, 2024-01-14:
## Соединения на печатной плате [BTH01](https://pvvx.github.io/BTH01/)

N   ножки чипа | Куда соединен | Маркировка | Название вывода чипа
-- | -- | -- | --
1 | Через R2 на С11 + Сенсор Vdd | CHT8305_VDD | GPIO_P00
32 | Сенсор SCL + R4 на Сенсор Vdd | CHT8305_SCL | GPIO_P34
31 | Сенсор SDA + R5 на Сенсор Vdd | CHT8305_SDA | GPIO_P33
7 |   | TM | TM
8 |   | TX1 | GPIO_P09
9 |   | RX1 | GPIO_P10
11 | Кнопка S1 (работает на GND) + C2  + R8 на Vbat |   | GPIO_P14
20 |   | RX2 | GPIO_P18
21 |   | TX2 | GPIO_P20
22 | C10   на GND | REST_N | RST_N
15 | Через R7 на светодиод D1 |   | GPIO_P15
27 |   | P25 | GPIO_P25

pvvx, 2024-01-14:
Тестовая прошивка для [BTH01](https://pvvx.github.io/BTH01/) готова, протестирована.
[image]

theanch92, 2024-01-15:
Hi, first of all thank you  for the hard work done! I just flashed your latest firmware BTH01_v05.hex on a BTH01, Bluetooth communication works but the temperature and humidity values were 0, the battery value is fine. 
Looking on pcb, my BTH01 appears to have the AHT30 sensor and not the CHT8305C. 

[image]
[image]

I have hope that you can also implement this sensor? Thanks so much again!

pvvx, 2024-01-20:
I don't have AHT30 yet. But I added processing code. There is no opportunity to check yet...
OTA version coming soon - It will be possible to change the firmware regardless of the sensors.

stefanstidlffg, 2024-01-21:
hi! Thank you for your work here!
here is a link you could add to the readme
https://developer.tuya.com/en/docs/iot/burn-and-authorize-phy6222-chip

theanch92, 2024-01-23:
> I don't have AHT30 yet. But I added processing code. There is no opportunity to check yet... OTA version coming soon - It will be possible to change the firmware regardless of the sensors.

Flashed FW 0.7 and the AHT30 works perfectly!
Also installed on TH05 and here too everything is perfect!

Great job

NeoMatrixJR, 2024-01-23:
> Flashed FW 0.7 and the AHT30 works perfectly! Also installed on TH05 and here too everything is perfect!
>
> Great job

Did you flash OTA with the example site or did you solder in?  Is this fully functioning firmware to add to home assistant via proxy and still has functional screen or is this just testing getting custom firmware on the device?  I tried the example web page but it wouldn't list my test device.

sstidl, 2024-01-24:
@pvvx Hi Victor! I have v0.6 flashed via serial. can i update to 0.7 OTA? or do I have to do this over serial a last time?

pvvx, 2024-01-24:
v0.x is the firmware for tests.
While the firmware number is not 1.x or higher, these are test firmwares and many things do not work in them.
Normal OTA will be from version 1.0.

g6094199, 2024-01-25:
he guys.

sorry but i dont gt how to get the chip to flash mode?
must TM pulled to high (3.3V)? or have i to pull RST (which PIN is this?) to GND?
or what else is wrong?

sstidl, 2024-01-25:
> he guys.
> 
> sorry but i dont gt how to get the chip to flash mode? must TM pulled to high (3.3V)? or have i to pull RST (which PIN is this?) to GND? or what else is wrong?

In my case it worked if TM left open and RST shortly pulled to GND 
At the same time try to connect with the python script

pvvx, 2024-01-26:
| USB-UART | PSB |
|---|---|
| RX | TX1 |
| TX | RX1 |
| RTS | REST_N|
| GND | -Vbat |
| +3.3V | +Vbat |

g6094199, 2024-02-02:
> USB-UART 	PCB
> RX 	TX1
> TX 	RX1
> RTS 	REST_N
> GND 	-Vbat
> +3.3V 	+Vbat

with RTS to REST_N i get 

```
PHY62x2 - Reset Ok
Error read Revision!
```

but with " RST shortly pulled to GND" it works.

pvvx, 2024-02-04:
> but with " RST shortly pulled to GND" it works.

Linux ?  - Everything is crooked there.

g6094199, 2024-02-04:
> > but with " RST shortly pulled to GND" it works.
> 
> Linux ? - Everything is crooked there.

yes, never used windoze in 20 years. so linux is my natural OS.

TheNitek, 2024-02-10:
> ```
> PHY62x2 - Reset Ok
> Error read Revision!
> ```

I have the same error with windows, trying to reset manually (I use a arduino uno with board with the atmel removed as a serial adapter and reset line is not availble, I think.)

g6094199, 2024-02-10:
its tricky...even with another instance inbetween. i started the flash and grounded the pin afterwards shortly. you maybe have to play with the timing

TheNitek, 2024-02-10:
Since i get the "Reset ok" message, I guess my timing is ok, but it's still unable to read the revision

g6094199, 2024-02-10:
also did get the "reset ok" as you can read in my post, but it was not "ok".

tERB00, 2024-02-24:
I did this:
- Connected all the pins including Rst to Gnd
- Started the script
- Pulled off Rst lead from Gnd

Flashed that way first try with no issues.

TheNitek, 2024-02-24:
That's what I am trying, but It doesn't work even after many, many attempts. I feel like my board might be different in some way ...
[image]
[image]

pvvx, 2024-02-24:
REST_N must be released when the Reset OK message occurs.
Without an RTS signal from the adapter to REST_N, the connection is difficult.

pvvx, 2024-02-24:
> > > but with " RST shortly pulled to GND" it works.
> > 
> > 
> > Linux ? - Everything is crooked there.
> 
> yes, never used windoze in 20 years. so linux is my natural OS.

On Linux there are always problems with setting the character reception timeout and RTS/DTR on USB-UART adapters.

tERB00, 2024-02-25:
> That's what I am trying, but It doesn't work even after many, many attempts. I feel like my board might be different in some way ... 

I've got exact the same board. And yes, some usb adapters doesn't work with some devices but work with others... I used CH341A programmer in TTL mode on mine.

TheNitek, 2024-02-26:
Different USB2Serial worked the first time, even though I did the reset manually.

electro-nic, 2024-03-03:
Hi! yesterday I flashed the sensor and I would like to set the advertising period to 1 min as on xiaomi but max value 10000
[image]

electro-nic, 2024-03-03:
Firmware 1.6 
8:52:06: Поиск устройств
08:52:28: Ожидание соединения с BTH01-DB601F
08:52:38: Model: BTH01
08:52:38: Firmware: github.com/pvvx
08:52:38: Hardware: 0014
08:52:38: Software: V1.4
08:52:38: Device info # hw: 0014, sw: 0014, services: 00004398, sd: 0000
08:52:38: Устройство подключено.
08:53:10: Установка времени на устройстве (23763AE465)...

draekko, 2024-03-15:
Just an FYI for future readers of this issue thread, in my case with my serial adapter i had to tap REST_N to ground to get it working instead of just going straight to ground and then releasing like was suggested.  Was using an older cp210x adapter.

pvvx, 2024-03-15:
The script has been fixed to work on Linux.

draekko, 2024-03-15:
good to know

gvieiraaa, 2024-03-15:
Sorry if I got this wrong, but I can't use OTA first, I have to flash using UART?

And could I do it with a raspberry pi instead of with a uart usb adapter?

pvvx, 2024-03-15:
On rPi you need to switch something to output RTS... /boot/config.txt ...

Romno7, 2024-03-18:
Не работает опрос состояния RX2 (при замыкании на GND) ничего не происходит.
Версия прошивки 1.6
17:42:01: Model: BTH01
17:42:01: Firmware: github.com/pvvx
17:42:01: Hardware: 0014
17:42:01: Software: B1.6
17:42:01: Device info # hw: 0014, sw: 0016, services: 00000281, sd: 0000
17:42:01: OTA ver: 01

Romno7, 2024-03-18:
Также нет возможности управлять TX2. Это аппаратное ограничение для данного (BTH01) датчика?

NonPIayerCharacter, 2024-04-13:
Have one of those, initially it was with CHT8305, but with original firmware I didn't find a use for it. Later removed 8305, as i needed it as a spare part for something else. Found this firmware, looked through source, and seen that it supports AHT2x. Instead of 8305, soldered AHT21, as they are almost pin compatible (AHT21 lacks AD0 and ALERT pins, instead they are NC), and can confirm that this mod works.

Nikolay-Ch, 2024-05-09:
@pvvx - is it not expected to support other data packet formats (for example ATC1441), like for devices on the Telink chipset?

pvvx, 2024-05-09:
BTHome is the only open standard with a registered UUID and paid for in the Bluetooth SIG by Allterco.

strix-iwan, 2024-09-23:
подпаял pl2303 а что дальше не понял - где скрипт прошивки ? Какой программой шить?

pvvx, 2024-09-24:
https://github.com/pvvx/THB2/blob/master/README-ru.md#%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0

strix-iwan, 2024-10-02:
как понять что прошилось то? 
[image]

strix-iwan, 2024-10-08:
значит никто не подскажет (( жаль

pvvx, 2024-10-08:
У вас нелады с Python.
В Windows обычно используется `python.exe` вместо `python3`.

xyzzy42, 2024-10-19:
> The script has been fixed to work on Linux.

The copy in `bthome_phy6222/rdwr_phy62x2.py` is fixed, but there is a second copy in `ota_boot/rdwr_phy62x2.py`, this other copy is not updated and still opens and closes the serial port when changing baud rate.

edwardtich1, 2024-11-21:
```
C:\THB2-1.8>python3 rdwr_phy62x2.py -p COM5 -e -r wh BOOT_BTH01_v19.hex
Python
```

what does it mean?

edwardtich1, 2024-11-21:
it worked!
thanks a lot!

**maybe someone will find it useful**

connect RESET and GND

`C:\THB2-1.8>python rdwr_phy62x2.py -p COM5 -e -r wh BOOT_BTH01_v19.hex`
press enter

```
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
```
disconnect RESET and GND

```
PHY62x2 - Reset Ok
Revision: b'001340c4 6222M005'
FlashID: 1340c4, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008b88
Segment: 1fff0000 <- Flash addr: 0000bb88, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000bf88, Size: 00002bca
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008b88
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000b88 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bb88, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bb88... ok
Segment: 1fff1838 <- Flash addr: 0000bf88, Size: 00002bca
Write 0x00002000 bytes to Flash at 0x0000bf88... ok
Write 0x00000bca bytes to Flash at 0x0000df88... ok
----------------------------------------------------------
Write Flash from file: BOOT_BTH01_v19.hex - ok.
Send command 'reset' - ok
```

edwardtich1, 2024-11-23:
Now I want to adjust the temperature and humidity values.

**Actual info:**
Temperature: 25.2 °C
Humidity: 53 %

**Sensor info:**
Temperature: 22.50 °C
Humidity: 50.00 %

```
Sensor parameters CHT8305, I2C Address: 0x40
Correction coefficients
Temperature: 16500
Humidity: 10000

Display correction
Temperature: -4000
Humidity: 0
```
how can I correct the data?

pvvx, 2024-11-23:
The coefficients are specified in the datasheet for the CHT8305 sensor.
The program uses coefficients multiplied by 100.

Datasheet for the sensor CHT8305
https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8305C%20Advanced%20Datasheet%20Ver2.5%202021-10-26.pdf

edwardtich1, 2024-11-23:
> Коэффициенты указаны в паспорте датчика CHT8305. Программа использует коэффициенты, умноженные на 100.
> 
> Технический паспорт датчика CHT8305 https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8305C%20Advanced%20Datasheet%20Ver2.5%202021-10-26.pdf

I understand. cannot be adjusted. Thank you very much!

pvvx, 2024-11-23:
The coefficients are changed in the program PHY62x2BTHome.html.
Use fw "APP", not "BOOT".

edwardtich1, 2024-11-23:
> Коэффициенты меняются в программе PHY62x2BTHome.html. Используйте fw "APP", а не "BOOT".

I saw where they were changing
but I don’t understand at all what needs to be changed and what)

pvvx, 2024-11-23:
Linear equation with one variable (7th grade school)

edwardtich1, 2024-11-23:
> I understand. cannot be adjusted. Thank you very much!

I understand. cannot be adjusted. Thank you very much!

g6094199, 2024-11-23:
> Now I want to adjust the temperature and humidity values.
> 
> **Actual info:** Temperature: 25.2 °C Humidity: 53 %
> 
> **Sensor info:** Temperature: 22.50 °C Humidity: 50.00 %
> 
> **Set values**
> 
> ```
> Sensor parameters CHT8305, I2C Address: 0x40
> Correction coefficients
> Temperature: 16500
> Humidity: 10000
> 
> Display correction
> Temperature: -4000
> Humidity: 0
> ```
> 
> how can I correct the data?

just to claify:

EVERY senser has its unaccuracies which is expectable and can be retrieved from the datasheets (to bad that chinese oem tend to "optimise" their products for marketing. also is the quality management.  so i wont trust +-0.5° for this very cheap CHT8305 sensor. while a sensirion sht4x sensor  with +-0.2° cost as much as the whole bht01 module. same for the dht22)! 

eg 
sensor 1 has +-1,5° (common for very cheap sensors)
sensor 2 has +-0,8° (which is quiet good sensor)
but anyway in worst case sensor 1 has a positve max strung of 1,5° and sensor 2 of -0,8°. so this is a  worst case an results in measuring diff of 2,3°. this is near to the values u mentioned and expecable. so i would say there is nothing wrong with the sensors at first sight. 

just to adjust the offests you need to know WHICH sensor is wrong...this CANT be done with only 2 sensors. 
you need a calibrated sensor or a bunch of uncalibrated good quality senors to measure a mean value. this value should tend to the real/correct value. from there you can do offsets.

edwardtich1, 2024-11-23:
> **Actual info:**
> Temperature: 25.2 °C
> Humidity: 53 %

calibrated sensor

pvvx, 2024-11-23:
> calibrated sensor

Measurements must be carried out in a special stable cabinet with constant temperature and humidity. Any convection must be excluded. Temperature and humidity fluctuations must not exceed the accuracy of the results over several hours of measurement. Otherwise, you will not calibrate the sensor.

The battery temperature changes over a long period of time. This affects the sensor, as it is located in a single plastic housing. Therefore, the measurements in the temperature-stabilized cabinet will be equalized only after a couple of hours.

pvvx, 2024-11-23:
> EVERY senser has its unaccuracies which is expectable and can be retrieved from the datasheets (to bad that chinese oem tend to "optimise" their products for marketing. also is the quality [...]

CHT8305 is not designed to operate from CR2032.
CHT8305 has a minimum supply voltage of 2.5V and a large delta of readings in the operating voltage range. A stabilized power supply is required.
And the thermometer's voltage supplied by the CR2032 battery changes depending on the load. And the load is pulsed - 6..10 μA and 5..15 mA. As a result, the voltage of a half-discharged CR2032 battery pulses from 3.0 to 2.2..2.6V.

At voltages below 2.5V, the CHT8305 produces jumping, understated temperature readings by tens of degrees...

pvvx, 2024-11-25:
[image]

[image]

Thermometers: CGG1M, TS0201 ([TS0201S1](https://github.com/pvvx/ATC_MiThermometer/issues/571))

At different speeds of temperature change - different readings. Depends on the speed of heating of the sensor with the elements surrounding it. Plus different air convection on the shelf.

pvvx, 2024-11-25:
> and you see how crap the sht85 are.

Place and lock all your sensors in a cardboard box and in a closed cabinet.
After a couple of hours, compare the readings.
Please remember to ensure that the battery voltage for CHT8305 is greater than 2.5V.

[18 more replies]