pvvx/THB2 #8: TH-05 with CHT8305

zipfo, 2024-01-24:
Здравствуйте, приехал TH05, но с неизвестным LCD драйвером и CHT8305. Сейчас прошивка TH05 адаптирована конкретно под AHT20?

[image]

almirus, 2024-01-24:
А где достать распиновку для th05
Куда подпаиваться для прошивки?

pvvx, 2024-01-24:
C другой стороны платы, там всё пописано - см. обновленный https://pvvx.github.io/TH-05/
| USB-UART | PSB |
|---|---|
| RX | TX1 |
| TX | RX1 |
| RTS | REST_N|
| GND | -Vbat |
| +3.3V | +Vbat |

almirus, 2024-01-24:
у меня старая bl55028

pvvx, 2024-01-24:
Ужасная [BL55028](https://pvvx.github.io/TH-05/BL55028_V1.3_en.pdf). Кушает много.

almirus, 2024-01-24:
@pvvx Уверен что ваша прошивка сделает намного лучше

`
C:\Users\user\AppData\Local\Programs\Python\Python312>python rdwr_phy62x2.py -p COM5 -r rc 0x11000000 0x80000 ff_thb2.bin                                                                                                                                                              =========================================================                                                                                                                                                                                                                               PHY62x2 Utility version 11.01.24                                                                                                                                                                                                                                                        ---------------------------------------------------------                                                                                                                                                                                                                               Connecting...                                                                                                                                                                                                                                                                           PHY62x2 - Reset Ok                                                                                                                                                                                                                                                                      Error read Revision!                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            C:\Users\user\AppData\Local\Programs\Python\Python312>python rdwr_phy62x2.py -p COM5 -e -r wh BOOT_TH05_v07.hex                                                                                                                                                                        =========================================================                                                                                                                                                                                                                               PHY62x2 Utility version 11.01.24                                                                                                                                                                                                                                                        ---------------------------------------------------------                                                                                                                                                                                                                               Connecting...                                                                                                                                                                                                                                                                           PHY62x2 - Reset Ok                                                                                                                                                                                                                                                                      Revision: b'001364c8 6222M005'                                                                                                                                                                                                                                                          FlashID: 1364c8, size: 512 kbytes                                                                                                                                                                                                                                                       PHY62x2 - connected Ok                                                                                                                                                                                                                                                                  ---- Segments Table -------------------------------------                                                                                                                                                                                                                               Segment: 11003000 <- Flash addr: 00003000, Size: 00008d64                                                                                                                                                                                                                               Segment: 1fff0000 <- Flash addr: 0000bd64, Size: 00000400                                                                      [... cut ...]                                                                                           Write Flash from file: BOOT_TH05_v07.hex - ok.                                                                                                                                                                                                                                          Send command 'reset' - ok                                              
`

Не смог оригинальную сохранить, для th05 другие адреса?

almirus, 2024-01-24:
успех, при запуске показало "07" несколько сек, это отладочное?

```
00:25:51: Нет подключения
00:25:54: Поиск устройств
00:26:28: Ожидание соединения с TH05-5F21F7
00:26:37: Model: TH05
00:26:37: Firmware: github.com/pvvx
00:26:37: Hardware: 0001
00:26:37: Software: V0.7
00:26:37: Dev info # hw: 0015, sw: 0007, services: 000002A3, sd: 0000
00:26:37: OTA ver: 01
00:26:37: Устройство подключено.
```

pvvx, 2024-01-24:
> успех, при запуске показало "07" несколько сек, это отладочное?

Это номер версии.

pvvx, 2024-01-24:
> Уверен что ваша прошивка сделает намного лучше

Менее 13 мкА в режиме сна у TH-05 не удалось выжать. Такие чипы ставят китайские разгильдяи под марку Tuya.
У оригинальной прошивки сон - 23 мкА. Там они вообще не заботятся о потреблении.
И LCD c драйвером [BL55028](https://pvvx.github.io/TH-05/BL55028_V1.3_en.pdf) не приспособлен для работы от CR2032. По мере посадки батареи напряжение для CR2032 под нагрузкой падает и считается нормой до 2.0..2.2В = 0%.
А тут чип с нормой питания от 2.5В.
В итоге вы уде ничего не увидите на экране когда CR2032 поработает на 30..40%, если не будете присматриваться и крутить его под разными углами.
При 2.5В:
[image]

При 3.3В:
[image]

В итоге расходовать батарейку до 0% не выйдет. А это сокращает время работы от батареи....

almirus, 2024-01-24:
А при понижении ниже 2.5 можно программно вырубать экран?

pvvx, 2024-01-24:
Экран на данном чипе не отключается. Т.е. если давать команду сброса контроллеру, то он будет ещё больше жрать. А уменьшить менее чем при текущих настройках когда показывает - нет возможности.

almirus, 2024-01-30:
Успешно перешил на 0.8, раньше не работало установка времени

```
21:13:17: Старт программирования...
21:14:07: Программирование завершено за 50.104 секунды
21:14:16: Переподключение
21:14:16: Устройство отключено.
21:14:16: Ожидание соединения с TH05-5F21F7
21:14:19: Устройство отключено.
21:14:19: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
21:14:19: Переподключение 0 из 5
21:14:19: Ожидание соединения с TH05-5F21F7
21:14:26: Model: TH05
21:14:26: Firmware: github.com/pvvx
21:14:26: Hardware: 0001
21:14:26: Software: V0.8
21:14:26: Dev info # hw: 0015, sw: 0008, services: 000002B2, sd: 0000
21:14:26: OTA ver: 01
21:14:26: Устройство подключено.
21:14:38: Время на устройстве: 1970-01-02 08:51:29
21:14:41: Установка времени на утройстве (23C166B965)...
21:14:41: Время на устройстве: 2024-01-30 21:14:41
21:14:44: Время на устройстве: 2024-01-30 21:14:44
21:14:50: Отключение
21:14:50: Устройство отключено.
```

pvvx, 2024-01-31:
Подождите выхода первого релиза. Он скоро.
Будет в v1.0.

almirus, 2024-02-06:
Чет не могу обновить 0.8 на 1.1
Датчик лежит рядом с компом, показания считываются, переходит в boot режим не с охотой, но отображает на пару секунд 08 и переходит обратно в стандартный режим (иконка соединения постоянно горит)
```
23:06:38: Не подключено
23:06:40: Поиск устройств
23:06:49: Ожидание соединения с TH05-5F21F7
23:06:53: Файл: TH05_v11.bin
23:06:53: Файл id:PHY6, Сегментов: 4, Старт: 0x1FFF1838, Размер: 50304 байт
23:06:53: Размер файла: 50308 байт
23:06:53: Счетчик: 3145 блоков
23:07:01: Model: TH05
23:07:01: Firmware: github.com/pvvx
23:07:01: Hardware: 0001
23:07:01: Software: V0.8
23:07:01: Dev info # hw: 0015, sw: 0008, services: 000002B2, sd: 0000
23:07:01: Устройство подключено.
23:07:02: Ответ на команду (33): ff
23:07:43: Отключение
23:07:43: Устройство отключено.
23:10:21: Поиск устройств
23:10:52: Ожидание соединения с TH05-5F21F7
23:10:57: Model: TH05
23:10:57: Firmware: github.com/pvvx
23:10:57: Hardware: 0001
23:10:57: Software: V0.8
23:10:57: Dev info # hw: 0015, sw: 0008, services: 000002B2, sd: 0000
23:10:57: Устройство подключено.
23:10:57: Ответ на команду (33): ff
23:11:43: Строка конфигурации: 55000000001f501d00023c0200
23:11:46: Dev sensor # tk: 1250, hk: 625, tz: -5000, hz: 0
23:11:46: Dev sensor # id: 0000-AAAA, i2c_addr: 0x38
23:11:49: Ответ на команду (44): ff
```
пробовал с телефона, тож самое

pvvx, 2024-02-06:
Версия 0.8 ещё не поддерживала нормальный OTA.
Придется проводами прошить Boot версию от 1.0+.

Shestoperd, 2024-02-06:
Пришли два клона TH-05, вместо BL55028 стоит GNV1792S, сенсор S-8310. У GNV1792S напряжение питания 2.5-5.5 вольт, потребление среднее 7.5мкА, максимальное 20мкА, это если верить китайскому datasheet-у. Ток покоя при выключенном дисплее и генераторе 5мкА.
[image]
[image]
Пока не пробовал прошивать, не очень понятно как быть с сигналом RST, которого у моего UART-а нет, нужно просто перед началом замкнуть на землю и отпустить?

pvvx, 2024-02-06:
> Пришли два клона TH-05, вместо BL55028 стоит GNV1792S,

TH05 больше. Пока "освоена" [только одна версия ](https://pvvx.github.io/TH-05/)- на плате надпись TH05_V1.4.
Поддерживаемые датчики описаны в [README](https://github.com/pvvx/THB2#%D0%BE%D1%81%D0%BD%D0%BE%D0%B2%D0%BD%D1%8B%D0%B5-%D1%85%D0%B0%D1%80%D0%B0%D0%BA%D1%82%D0%B5%D1%80%D0%B8%D1%81%D1%82%D0%B8%D0%BA%D0%B8).

Другие варианты TH05 видимо имеют различие в разводке сегментов LCD. Заказал разные, пока идут по почте с али...

Вот ещё версия TH05_v1.3:
[image]
[image]
[image]
Фото с али...

> не очень понятно как быть с сигналом RST, которого у моего UART-а нет

Будет проблемно.

pvvx, 2024-02-06:
> У GNV1792S напряжение питания 2.5-5.5 вольт, потребление среднее 7.5мкА, максимальное 20мкА, это если верить китайскому datasheet-у. Ток покоя при выключенном дисплее и генераторе 5мкА.

У BL55028 тоже много чего написано. но не работает

Shestoperd, 2024-02-06:
> Поддерживаемые датчики описаны в [README](https://github.com/pvvx/THB2#%D0%BE%D1%81%D0%BD%D0%BE%D0%B2%D0%BD%D1%8B%D0%B5-%D1%85%D0%B0%D1%80%D0%B0%D0%BA%D1%82%D0%B5%D1%80%D0%B8%D1%81%D1%82%D0%B8%D0%BA%D0%B8).

Я изначально прочитал это, т.к. там указан CHT8310, то думаю и S-8310 сгодится.
Но вот что мне интересно, зачем передавать данные измерять температуру/влажность каждые 5-10 секунд, раз в минуту для комнатного датчика ИМХО достаточно. 

> У BL55028 тоже много чего написано. но не работает

Просто чуть выше было обсуждение потребления, потому сразу написал сколько заявляют китайцы.

> Будет проблемно.

Я как нибудь попробую без RST, выйдет хорошо, не выйдет, буду искать другой адаптер.

pvvx, 2024-02-07:
> Но вот что мне интересно, зачем передавать данные измерять температуру/влажность каждые 5-10 секунд, раз в минуту для комнатного датчика ИМХО достаточно.

А мне нет. Нагреватели и кондиционеры быстрее отрабатывают и при меньшей частоте передачи температура гуляет на несколько градусов, т.к. не отключаются нагреватели и прочее.

Для наблюдения за температурой и влажностью купите более дешевый вариант и наблюдайте :)
А для авто-управления домом требуется совсем другое...

Shestoperd, 2024-02-07:
> Для наблюдения за температурой и влажностью купите более дешевый вариант и наблюдайте :)

Так эти датчики как раз для наблюдения, а не управления, не?

> А для авто-управления домом требуется совсем другое...

Как бы в помещении в разных местах температура будет изменяться по разному. Если поставить датчик под струю кондиционера, то наверное температура будет скакать как заяц, при включении/выключении, но если расположить за пределами воздушного потока, то уже будет заметный лаг.
Впрочем я просто спросил, точность этих датчиков не сказать чтобы высокая, потому мне показалась странной такая частота опроса.

pvvx, 2024-02-07:
Даже если используется масляный обогреватель с простым вкл/выкл, то его инерция:
[image]

На Xiaomi Aqara Zigbee уже ничего не сделать - он тупит хуже даже такого обогревателя...

pvvx, 2024-02-07:
> Так эти датчики как раз для наблюдения, а не управления, не?

CHT8310 и другие устанавливаемые датчики сами измеряют с интервалом 1 сек. И так их програмно перестраиваю на 5 сек.

Т.е. TH05 по Вашему нельзя использовать для управления?
Или частота передачи мешает наблюдению?

Shestoperd, 2024-02-07:
> Мне просто интересно - какова причина требований у некоторых передачи замеров раз в час?

Да нет никаких требований, я просто спросил почему выбран именно такой интервал, и все. Я не прошу что-то переделывать или добавлять.

> Существенной экономии батарейки на этом не получите - жалкие пару %.

Все, предлагаю на этом закончить обсуждение интервала обновлений.

pvvx, 2024-02-07:
> У GNV1792S 

Аналог в некоторых Xiaomi - **BU9792FUV**
[MJWSD05MMC](https://pvvx.github.io/MJWSD05MMC/) , [CGDK2](https://pvvx.github.io/CGDK2/), варианты LYWSD03MMC, ...

almirus, 2024-02-07:
> Версия 0.8 ещё не поддерживала нормальный OTA. Придется проводами прошить Boot версию от 1.0+.

Все получилось, спасибо

Shestoperd, 2024-02-07:
Попытался считать родную прошивку, получаю ошибку:
```
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX and Chip Power!

```
Адаптер на PL-2303HX, сигналы RTS и DTR не выведены на гребенку, но я подпаялся к нужной ноге и через резистор 39 Ом подключил его к Reset. Адаптер точно рабочий, во первых, я им без проблема прошиваю ESP8266 и другие аналогичные платы, а также подключаюсь к различным одноплатникам. Во вторых, я в терминале вижу то, что передает датчик по UART. 
[image]

Сигнал RTS  тоже рабочий, посмотрел осциллографом, при старте считывания низкий уровень держится 316мс, чего должно быть достаточно для сброса. 
Может кто-то уже сталкивался с таким, что это может быть?
Под рукой еще есть программатор на CH341A, который также может работать в режиме serial, в нем тоже RTS не выведен, хотя вроде как есть.

CMY-Reader, 2024-02-08:
> TH05 больше. Пока "освоена" [только одна версия ](https://pvvx.github.io/TH-05/)- на плате надпись TH05_V1.4. Поддерживаемые датчики описаны в [...]

Прошивка версии 1.1 работает на 
 - TH05_V1.5 2023-05-23 с BL55028 и ATH20
 - TH05_V1.6 2023-06-02 с BL55028 и ATH20

Очень странно что сохранение прошивки идет очень медленно. Около !!30!! минут. А прошивка через тот же адаптер происходит за 1 минуту. По воздуху 2мин.

Shestoperd, 2024-02-08:
> Пробуйте программу от PHY -

Не подключается. Делал по инструкции [отсюда](url=https://developer.tuya.com/en/docs/iot-device-dev/bluetooth_platform_ble_phy6222?id=Kcnreamkf9r1a), т.к. китайскую ниасилил. 
Получил бесконечное ожидание:
```
Current port: COM41
Current baudrate: 9600
Current stopBits: 1
Current parity: No
Serial opened!! 
*******************************
UART TX ASCII: UXTDWU
........
UART TX ASCII: UXTDWU
Serial port COM41 closed!!
*******************************
```
Но резет тем не менее работает, если подключиться на скорости 115200, и нажать резет, то видно что при старте выводятся основные параметры:
```
Current port: COM41
Current baudrate: 115200
Current stopBits: 1
Current parity: No
Serial opened!! 
*******************************
UART RX : DEVICE_NAME = THWL
DEVICE_VERSION = 1.5
PID = iv7hudlj
MAC = DC2....7FF4
DID = uuid291....111
KEY = QvYPcN.....C8RxHRTibOG
Device_authentication = 1
user_dt.ty_dt.updata_cnt = 0 
....................begin = 1
flash_dt.ty_fdt.unit = 0
flash_dt.ty_fdt.status1 = 3
power on status 1 = 2
vid = 8215 mid = 5959
Smaple_data: 258, 798, 798, 798, 798, 
Temperature: 243 Humidity: 38 Battery: 100
temp_alert: 2 hum_alert: 2
Serial port COM41 closed!!
*******************************
```
Пробовал по китайским картинкам провести автоматический подбор битрейта, но не дождался его окончания.
В общем, никак не получается ввести загрузчик чипа в режим программирования :( Или я что-то упускаю.

pvvx, 2024-02-08:
> Очень странно что сохранение прошивки идет очень медленно. Около !!30!! минут. 

Это типа "хак" и в PHY не рассчитывали, что прошивку можно считать.
Но большого смысла считывать нет, т.к. выложена уже считанная Full Flash: [Tuya of.version](https://pvvx.github.io/TH-05/bin/fullFlashTuyaTH05.zip).

pvvx, 2024-02-08:
> > Пробуйте программу от PHY -
> 
> Не подключается. Делал по инструкции [отсюда](https://developer.tuya.com/en/docs/iot-device-dev/bluetooth_platform_ble_phy6222?id=Kcnreamkf9r1a#title-12-Flashing%20method), т.к. китайскую ниасилил. 

Мы ушли от китайской версии. Она рассчитана на SDK для сборки на проприетарном Keil.
И PhyPlusKit использует фиксированную разметку для прошивки, что уже не совместимо с версиями из данного репозитория. У них другой OTA.

Текущая SDK, из данного репозитория, переработана на gcc.

Shestoperd, 2024-02-09:
> Это работает уже сама прошивка.
> Для Boot-ROM PHY6222 надо 9600 baud и после Reset посылать 'UXTDWU' до ответа 'cmd>>:' или 'fct>>:'.

Именно так я себе и представлял, после резета оригинальная прошивка сначала выдает информационные данные, а затем периодично посылает состояние датчиков.
Решил все же попробовать CH341A в качестве USB->uart, и получилось подключиться, но я сразу попытался шить через PhyPlusKit и стер оригинальную прошивку :( При этом записать ничего не удалось, то писало что нет файла, то просто падало.
После некоторых плясок с бубоном я наловчился нажимать ресет в нужный момент и все же смог залить BOOT_TH05_v11.hex :) Обновился по OTA, но к сожалению, ни S-8310, ни GNV1792S, не работают :( При этом очень быстро садится батарея.

pvvx, 2024-02-09:
У меня пока нет[ такого варианта ](https://github.com/pvvx/THB2/issues/8#issuecomment-1930943920)
Возможно https://pvvx.github.io/TH-05/bin/fullFlashTuyaTH05.zip распознает?

Shestoperd, 2024-02-09:
> Возможно https://pvvx.github.io/TH-05/bin/fullFlashTuyaTH05.zip распознает?

Не завелась. Впрочем у меня есть второй экземпляр, с него могу сохранить прошивку, опыт уже есть, но пока записал BOOT_TH05_v11.hex, вдруг к Вам приедут такие же, или могу выступить тестером прошивок.

Shestoperd, 2024-02-12:
Выяснил что название модели моих неудачных HT-05 = UA660-WL-B0, по крайней мере так они видятся в туевском облаке:
```      "category": "wsdcg",
      "model": "UA660-WL-B0",
      "name": "Bluetooth Temperature Humidity Sensor 3",
      "product_id": "iv7hudlj",
      "product_name": "卫力-温湿度传感器-UA660-WL-B0",
```
Решил прозвонить что куда подключено, для того чтобы сравнить с предыдущими версиями:
```
№  | Название  | функции             | куда соединен
2  | P1        | GPIO 1              | кнопка при нажатии замыкает на землю резистор r15 подключенный к VCC
7  | TM        | test mode enable    | TM  
8  | P9        | GPIO 9              | TX   
9  | P10       | GPIO 10             | RX   
10 | P11/AIO_0 | GPIO 11/ADC input 0 | ADC, делитель напряжения R1, R3, R2
14 | VDD3      | 3.3V power supply   | VCC   
20 | P18/AIO_7 | GPIO 18             | CHT8310 SDA
21 | P20/AIO_9 | GPIO 20             | CHT8310 SCL
22 | RST_N     | reset, active low   | RESET  
31 | P33       | GPIO 33             | GNV1792s SCL
32 | P34       | GPIO 34             | GNV1792s SDA
```
Странно что контроллер экрана и сенсор на разных портах, возможно это как-то влияет на потребление, в предыдущих версиях сделано так же? Мне казалось что на I2C можно вешать множество самых разных устройств, и они не будут друг другу мешать.

Deoptim, 2024-02-12:
> Менее 13 мкА в режиме сна у TH-05 не удалось выжать. 

А Вы можете сделать возможность отключения использования BL55028 через web-конфигуратор ?
Там BL55028 подключена напрямую к батарейке или питается от GPIO?
Если напрямую то можно как-то BL55028 загнать в глубокий сон вместо отключения?

pvvx, 2024-02-12:
На BL55028 не работают команды сна. И он подключен к +Vbat одной ногой, а вторая через GPIO.
Если опустить GPIO - потребление только увеличивается и сильно.

Deoptim, 2024-02-12:
> На BL55028 не работают команды сна. 

> https://pvvx.github.io/TH-05/BL55028_V1.3_en.pdf Найдите там команду отключения :) Были проверены все комбинации, включая неописанные, по аналогии с чего был сделан этот клон, и выставлено самое минимальное потребление.

Ну там в даташите ясно написано, чтобы включился  power-saving mode нужно :

> SDA and SCL must be connect to high level(by pull up
 resistor),otherwise the device maybe can not go into power saving mode
и
> In power-saving mode, SCL frequency must be less than 21KHz
Т.е. никаких команд не нужно.

Это в вашей прошивке реализовано?

pvvx, 2024-02-12:
Вы схему то смотрели? 
Или пытаетесь отнять время у других на ваши причуды?

Deoptim, 2024-02-12:
> Вы схему то смотрели? Или пытаетесь отнять время у других на ваши причуды?

Я к чему это, я к тому, что если в phy6222 используется аппаратный I2C то вообще может так быть что меньше 40-100kHz не поставите, много таких микроконтроллеров я видел. Может придется делать software I2C (bitbang) чтобы уменьшить скорость.

pvvx, 2024-02-12:
Если опустить SCL или SDA то там резисторы на пару кОм включатся в питание :P
Цель уменьшения скорости, если команды он воспринимает?

pvvx, 2024-02-12:
> Я к чему это, я к тому, что если в phy6222 используется аппаратный I2C то вообще может так быть что меньше 40-100kHz не поставите,

Данный SoC позволяет установить скорости I2C и меньше. Поправьте значения в исходных кодах и соберите себе прошивку...

Shestoperd, 2024-02-13:
> Исходные коды есть - кто вам мешает вписать (bitbang) и прочее?

Для тех кто не особо в этом разбирается, чем собирать? Я посмотрел в mk_windows.cmd вроде нужно скачать Arm GNU Toolchain Version 13.2.Rel1, зашел на https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads ,но что именно надо скачать не разобрался. Который из тулчейнов надо скачать?

pvvx, 2024-02-13:
> Который из тулчейнов надо скачать?
Последний: Arm GNU Toolchain Version 13.2.Rel1 Released: October 30, 2023
или любой старый. На 10-ом тоже собирается...

https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

Deoptim, 2024-02-13:
До изменения прошивки у меня показывал мой мультиметр 16мкА.
Ну без аппаратной переделки я добился 11-12мкА (при этом индикация видна как при разряженной батарейки) и дело не в I2C тут уже максимум энергосбережение судя по даташиту это 9мкА самой BL55028 + 3мкА SoC:
```
diff --git a/bthome_phy6222/source/lcd_th05.c b/bthome_phy6222/source/lcd_th05.c
index 038e2e0..6dcad7b 100644
--- a/bthome_phy6222/source/lcd_th05.c
+++ b/bthome_phy6222/source/lcd_th05.c
@@ -63,7 +63,7 @@ uint8_t display_out_buff[LCD_BUF_SIZE+1];
 const uint8_t lcd_init_cmd[]   =       {
                // LCD controller initialize:
                0xea, // Set IC Operation(ICSET): Software Reset, Internal oscillator circuit
-               0xC8, // Mode Set (MODE SET): Display enable, 1/3 Bias, power saving
+               0xd8, // Mode Set (MODE SET): Display enable, 1/3 Bias, power saving
                0xbc, // Display control (DISCTL): Power save mode 3, FRAME flip, Power save mode 1
                0x80, // load data pointer
                0xf0, // blink control 0xf2
diff --git a/bthome_phy6222/source/main.c b/bthome_phy6222/source/main.c
index 23469c0..b2056d9 100644
--- a/bthome_phy6222/source/main.c
+++ b/bthome_phy6222/source/main.c
@@ -176,7 +176,7 @@ const ioinit_cfg_t ioInit[] = {
 #elif (DEVICE == DEVICE_TH05)
                { GPIO_P00, GPIO_PULL_UP }, // GPIO_SPWR Sensor Vdd
                { GPIO_P01, GPIO_PULL_DOWN },
-               { GPIO_P02, GPIO_PULL_UP }, // GPIO_LPWR
+               { GPIO_P02, GPIO_FLOATING }, // GPIO_LPWR
                { GPIO_P03, GPIO_PULL_DOWN },
                { GPIO_P07, GPIO_PULL_DOWN },
                { GPIO_P09, GPIO_PULL_UP }, // TX1
```
Пробовал еще команду 0xEC из того же семейства LCD драйверов микросхему BL55070/BL55077 в этих семействах оно в режиме sleep должно брать 2мкА:
[image]
Дисплей отключается, т.е. как-то реагирует на команду, но энергосбережение остается прежним.

В принципе можно было бы добавить кнопочку в web-конфигуратор для экономии 5мкА (это если существенно не нужен дисплей) по примеру патча выше, а может и не стоит того....

Shestoperd, 2024-02-13:
> Ну без аппаратной переделки я добился 11-12мкА (при этом индикация видна как при разряженной батарейки)

Вы пробовали менять LCD Bias? Это по идее двухуровневая регулировка контрастности, снижая контрастность можно снизить потребление. 

> В принципе можно было бы добавить кнопочку в web-конфигуратор для экономии 5мкА (это если существенно не нужен дисплей) по примеру патча выше, а может и не стоит того....

Можно прицепить на кнопку, тогда не нужно будет менять конфигуратор.

pvvx, 2024-02-13:
Всё было опробовано. Там всего 2 байта влияющих на режимы, а в них всего 4 бита, которые что-то делают.
0xD8 уже было.  Это отключение отображения, но не работы контроллера LCD :P
Не забывайте сохранить совместимость с другими аналогами данного чипа. Народу попадется другой чип и будет толпа issue.

Shestoperd, 2024-02-13:
Собрал себе на пробу прошивку, без поддержки GNV1792s потребление 4мкА в простое. Это так просто к сведению.
Еще пробовал включить GNV1792s, но без сенсора, экран показал галиматью, вероятно программно этот контроллер совместим с BL55028, но вот сегменты отличаются, хотя думаю не очень сильно.

pvvx, 2024-02-13:
Есть команда "60xxxxxx", где xxxx - буфер LCD - 6 байт.
Вводите 6001000000000000 - рисуете сегмент byte0.bit0
Далее 6002000000000000 - рисуете сегмент byte0.bit1
и т.д.
Когда пройдете все биты - киньте картинку, [как тут](https://github.com/pvvx/THB2/)

pvvx, 2024-02-13:
> потребление 4мкА в простое. 

BTH01 
При 3.3В питании, sleep = 3.64 мкА:
[image]

При 2.2В питании, sleep = 3.35 мкА:
[image]

Это так просто к сведению. Еще пробовал включить GNV1792s, но без сенсора, экран показал галиматью, вероятно программно этот контроллер совместим с BL55028, но вот сегменты отличаются, хотя думаю не очень сильно.

Shestoperd, 2024-02-13:
> Есть команда "60xxxxxx", где xxxx - буфер LCD - 6 байт.

Я похоже что-то не так делаю, иногда загораются сразу несколько сегментов, а часть сегментов не гаснет при выполнении команды.

> Когда всё подготовлено, тогда и просто :)

Я писал к тому что GNV1792s может потреблять совсем немного.

pvvx, 2024-02-13:
TH05, текущая версия, при 2.2В, sleep = 14.5 мкА

[image]

pvvx, 2024-02-13:
TH05, текущая версия, но при **3.3В**, sleep = 18.22 мкА
[image]
CR2032 выдает около 2.8..2.95В 70..80% жизни, при работе. При новой батарее от 3.0 до 3.3В, но это напряжение очень быстро падает до 2.95В.
В итоге на TH05 с BL55028 на 2.9В как раз и выходит к 17 мкА.
Напряжение меньше - меньше ток утечки...

Shestoperd, 2024-02-13:
> Есть команда "60xxxxxx", где xxxx - буфер LCD - 6 байт.

Часть рожицы не гаснет, шестой байт нигде не отображается.
```
/*
 *  UA660 LCD buffer:  byte.bit

         --0.1--         --1.1--            --2.1--          BAT
  |    |         |     |         |        |         |        3.1 | 3.5 | 3.6 | 3.7 | 3.8
  |   0.5       0.2   1.5       1.2      2.5       2.2
  |    |         |     |         |        |         |      o 3.3
 0.8     --0.6--         --1.6--            --2.6--          +--- 3.3
  |    |         |     |         |        |         |     3.3|
  |   0.7       0.3   1.7       1.3      2.7       2.3       ---- 3.2
  |    |         |     |         |        |         |     3.3|
         --0.4--         --1.4--     *      --2.4--          ---- 3.4
                                    2.8
                                        --4.1--         --5.1--
                                      |         |     |         |
          x.x      x.x               4.5       4.2   5.5       5.2
          / \      / \                |         |     |         |
    4.8(  \_/  x.x \_/  )4.8            --4.6--         --5.6--
          x.x  / \ x.x                |         |     |         |
               \_/                   4.7       4.3   5.7       5.3     %
               x.x                    |         |     |         |     5.8
                                        --4.4--         --5.4--
                           OO 1.8
 None:  
*/

```
первая фотка 60ffffffffffffff, вторая 6000000000000000
[image] [image]

pvvx, 2024-02-14:
Всё бы хорошо, но PHY62x2 кушают больше из-за долгого пробуждения после сна.
Если сравнять период рекламы с Xiaomi LYWSD03MMC B1.4 до 2.5 сек, тогда PHY62x2 потребляет столько, сколько LYWSD03MMC полностью с экраном.

pvvx, 2024-02-14:
Вот такие токи в моменты пробуждения при передаче BLE рекламы на 3 канала имеются на TH05:
Цикл измерения по умолчанию - 2 периода.
Первое событие: 
[image]
[image]
Второе событие:
[image]
[image]
Хвост - так потребляет AHT30.

pvvx, 2024-02-14:
И для прикола надо обязательно сравнить с [nRF Power Profiler II](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2)
[image]
[image]

Deoptim, 2024-02-14:
@pvvx Может оно не существенно, но я в исходниках не нашел команды экономного режима для датчика AHT20-AHT30.

```
diff --git a/bthome_phy6222/source/sensors.c b/bthome_phy6222/source/sensors.c
index 4438b76..1af985a 100644
--- a/bthome_phy6222/source/sensors.c
+++ b/bthome_phy6222/source/sensors.c
@@ -75,6 +75,7 @@ int read_sensor_ahtxx(void) {
        init_i2c(I2C_400KHZ);
        if(!read_i2c_nabuf(thsensor_cfg.i2c_addr, reg_data, 7)
                        && (reg_data[0] & 0x80) == 0) { // busy
+               send_i2c_wreg(thsensor_cfg.i2c_addr, AHT2x_CMD_TMS, AHT2x_DATA_LPWR);
                deinit_i2c();
                _temp = ((reg_data[3] & 0x0F) << 16) | (reg_data[4] << 8) | reg_data[5];
                measured_data.temp = ((uint32_t)(_temp * thsensor_cfg.coef.temp_k) >> 16) + thsensor_cfg.coef.temp_z; // x 0.01 C
@@ -87,6 +88,7 @@ int read_sensor_ahtxx(void) {
                measured_data.count++;
                return 0;
        }
+       send_i2c_wreg(thsensor_cfg.i2c_addr, AHT2x_CMD_TMS, AHT2x_DATA_LPWR);
        deinit_i2c();
        return 1;
 }
diff --git a/bthome_phy6222/source/sensors.h b/bthome_phy6222/source/sensors.h
index 5fc3ee9..713fd7d 100644
--- a/bthome_phy6222/source/sensors.h
+++ b/bthome_phy6222/source/sensors.h
@@ -132,6 +132,8 @@ struct __attribute__((packed)) _cht8305_config_t{
 #define AHT2x_CMD_TMS  0x0AC  // Trigger Measurement Command
 #define AHT2x_DATA_TMS 0x3300  // Trigger Measurement data
 #define AHT2x_CMD_RST  0x0BA  // Soft Reset Command
+#define AHT2x_DATA_LPWR        0x0800 // go into low power mode
+

 typedef struct __attribute__((packed)) _measured_flg_t {
```

pvvx, 2024-02-14:
> Может оно не существенно, но я в исходниках не нашел команды экономного режима для датчика AHT20-AHT30.

Они и так не отличаются точностью...
В TH05 датчик AHT20-F. Уже есть клон GXHT3X (ставят в TH03Z).

[image]
[image]

pvvx, 2024-02-14:
> При работающем экране минимальное потребление 10мкА, это без сенсора и при напряжении питания 2,4 вольта. Вполне неплохо.
Tuya = 54 мкА (3.3В)
[THB1](https://pvvx.github.io/THB1)

Shestoperd, 2024-02-14:
> Tuya = 54 мкА (3.3В)
> [THB1](https://pvvx.github.io/THB1)

Это как раз мое устройство в облаке туя обозначается как UA660-WL-B0, откуда Вы брали название THB1/BT1?
Потихоньку колупаю прошивку, но т.к. я не слишком хорошо знаю программирование, застрял сейчас на битовых операциях при выводе на экран разных символов. Почему-то при выводе некоторых символов калечится буфер, иногда в неожиданных местах.

[91 more replies]