pvvx/THB2 #10: Информация

pvvx, 2024-01-31:
# Распределение Flash 512 килобайт

| Адрес | Описание | Размер |
|---|---|---|
| 0x00000 | Используется ROM | 8 килобайт |
| 0x02000 | Boot Info для ROM | 4 килобайта |
| 0x03000 | FW Boot с функцией OTA | 52 килобайта |
| 0x10000 | FW OTA | 128 килобайт |
| 0x30000 | Запись истории | 304 килобайт |
| 0x7C000 | Сохранение настроек (EEP) | 16 килобайт |

# FW Boot и OTA

`FW Boot` имеет функцию OTA, но не имеет функции записи истории.
`FW OTA` не имеет функции OTA, для OTA перезагружается в `FW Boot`.

pvvx, 2024-02-09:
[image]

CMY-Reader, 2024-02-09:
А есть возможность запросить счетчик Открытий и состояние Контакта (геркона)?
Или еще лучше историю открытий и закрытий (как история температуры и влажности).

pvvx, 2024-02-09:
> А есть возможность запросить счетчик Открытий и состояние Контакта (геркона)?

Показывает в PHY62x2BTHome.html.

> Или еще лучше историю открытий и закрытий (как история температуры и влажности).

Ну-ну - если к контакту прикоснуться пальцем при питании термометра от БП, то имеем наводку в 50Гц. Это кол-во срабатываний 100 шт в секунду. И куда такая история влезет? На SSD в пару ТБайт?

CMY-Reader, 2024-02-09:
> > А есть возможность запросить счетчик Открытий и состояние Контакта (геркона)?
> 
> Показывает в PHY62x2BTHome.html.
Как я понимаю он счетчик получает вместе с беконами в "момент срабатывания". 

Вопрос был про чтение счетчика "через час" после открытия.
 
> Ну-ну - если к контакту прикоснуться пальцем при питании термометра от БП, то имеем наводку в 50Гц. Это кол-во срабатываний 100 шт в секунду. И куда такая история влезет? На SSD в пару ТБайт? [...]

Тут вопрос к аппаратному методу борьбы с дребезгом. Но можно и на программном уровне. Например можно историю писать не чаще чем раз в секунду. Так сказать "усреднять" значение параметра. Добавить время и значение счетчика если за этот период он изменился.

pvvx, 2024-02-09:
Смысл этого всего, если тот-же HA ведет историю?

CMY-Reader, 2024-02-09:
Во-первых все радио можно заглушить. 
Во-вторых сервер можно выключить..перезагрузить.
В-третьих. Один из xiaomi сенсоров вожу в сумке велосипеда.  И данные температуры за поездку считываются по возвращению. 

Тема с "герконом" очень неожиданная, но клевая. Пока не знаю как можно применить. Но история событий будет полезна в любом случае.

pvvx, 2024-02-09:
>И данные температуры за поездку считываются по возвращению.

Считать кол-во оборотов колеса :)

CMY-Reader, 2024-02-09:
> > И данные температуры за поездку считываются по возвращению.
> 
> Считать кол-во оборотов колеса :)

Или ям на дороге :)

pvvx, 2024-02-09:
История open/close возможно будет в https://pvvx.github.io/TS0207_TZ3000/ , https://pvvx.github.io/TS0202_TZ3000/ , https://pvvx.github.io/DS01_zbeacon/ , https://pvvx.github.io/iSearching/ , ... и в аналогичных

Romno7, 2024-03-21:
Как обнулить (сбросить) счетчик?

pvvx, 2024-03-21:
Питанием

h0m3us3r, 2024-11-16:
> Требуется PCIe адаптер

А можете посоветовать PCIe адаптер?