pvvx/THB2 #11: TH05 appears as Demo, no service found

spupetic, 2024-02-03:
Hi!

I have tried to flash a TH05 with the 1.0 firmware via OTA, but the device has a name `Demo`, not TH05, and when I connect to it with the web app, it says no services found. According to the Tuya app, it has the latest firmware with version 2.5.

```
7:11:33 PM: Not connected
7:11:35 PM: Search for devices
7:11:41 PM: Waiting for connection with Demo
7:11:42 PM: NotFoundError: No Services found in device.
7:11:42 PM: Reconnect 0 of 5
```

Is it possible to do OTA from the Tuya firmware, or only when I have initially flashed firmware via UART?

pvvx, 2024-02-03:
There is no OTA from the Tyua option yet.
The first firmware is only via UART.

spupetic, 2024-02-03:
I see, thank you.

closed 2024-02-03 by spupetic

NeoMatrixJR, 2024-02-04:
Willing to uart flash, how do I remove and replace the screen safely?