pvvx/THB2 #12: TH-05 after flashing BOOT_TH05_v10.hex not visible via BT

BubbleRep, 2024-02-03:
Hi,

flashed the BOOT_TH05_v10.hex 

`PS H:\Downloads\THB2-1.0> python rdwr_phy62x2.py -p COM9 -e -r wh  .\bin\BOOT_TH05_v10.hex
=========================================================
PHY62x2 Utility version 25.01.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00009100
Segment: 1fff0000 <- Flash addr: 0000c100, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000c500, Size: 00002de2
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00009100
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00001100 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000c100, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000c100... ok
Segment: 1fff1838 <- Flash addr: 0000c500, Size: 00002de2
Write 0x00002000 bytes to Flash at 0x0000c500... ok
Write 0x00000de2 bytes to Flash at 0x0000e500... ok
----------------------------------------------------------
Write Flash from file: .\bin\BOOT_TH05_v10.hex - ok.
Send command 'reset' - ok
PS H:\Downloads\THB2-1.0>`

looke successful for me, but the [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html) tool doesn't see the devices, neither does the nrf connect scan on an android device.
The PCB  seems to be a bit different version:
[image]
[image]

Any idea how to proceed?

BR

pvvx, 2024-02-04:
[README](https://github.com/pvvx/THB2?tab=readme-ov-file#%D0%BE%D1%81%D0%BD%D0%BE%D0%B2%D0%BD%D1%8B%D0%B5-%D1%85%D0%B0%D1%80%D0%B0%D0%BA%D1%82%D0%B5%D1%80%D0%B8%D1%81%D1%82%D0%B8%D0%BA%D0%B8)
Кнопка используется для быстрого подключения к старым BT-адаптерам. Нажатие кнопки переключает интервал BLE рекламы на более короткий период (1562.5 мс). Действие продолжится 60 секунд, затем интервал восстановится на установленный в настройках.

The button is used to quickly connect to old BT adapters. Pressing the button switches the BLE advertising interval to a shorter period (1562.5 ms). The action lasts 60 seconds, then the advertising interval set in the device will work.

dromie, 2024-05-23:
I got almost the same board, mine is TH05Y_V1.1
And my guess is that you used a bad BOOT image, and firmware, I'll try with the instructions for https://pvvx.github.io/TH05F/

closed 2024-12-01 by pvvx