pvvx/THB2 #13: BTH01 Error: Open COM8, 9600 baud!

fralloger, 2024-02-06:
Привет, пытаюсь прошить BTH01 и возникает такая ошибка:

`PS C:\Users\dmitr\Downloads\THB2> python rdwr_phy62x2.py -p COM8 -e -r wh  .\bin\BOOT_BTH01_v11.hex
=========================================================
PHY62x2 Utility version 25.01.24
---------------------------------------------------------
Error: Open COM8, 9600 baud!`

В чем может быть причина? Драйвера стоят, бодрейт проверял, пробовал запускать PS от админа, но всё равно не помогло.

lehanspb, 2024-02-11:
Аналогичная проблема с Python 3.11. На 3.10 всё работает

pvvx, 2024-02-11:
```
python3 --version
Python 3.12.2
```

```
python3 rdwr_phy62x2.py -p COM11 -e -r wh ./bin/BOOT_THB2_v12.hex 
=========================================================
PHY62x2 Utility version 25.01.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008cc8
Segment: 1fff0000 <- Flash addr: 0000bcc8, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000c0c8, Size: 00002dc6
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008cc8
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000cc8 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bcc8, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bcc8... ok
Segment: 1fff1838 <- Flash addr: 0000c0c8, Size: 00002dc6
Write 0x00002000 bytes to Flash at 0x0000c0c8... ok
Write 0x00000dc6 bytes to Flash at 0x0000e0c8... ok
----------------------------------------------------------
Write Flash from file: ./bin/BOOT_THB2_v12.hex - ok.
Send command 'reset' - ok
```

fralloger, 2024-02-13:
Спасибо, помогло!

closed 2024-02-13 by fralloger