pvvx/THB2 #14: TH05Y_V1.2 with 2332 LCD not working.

rbonat, 2024-02-07:
Tested TH05_V1.2 device like in picture - everything seems to work ok on TH05_v11.bin except display. Any plans to support 2332 display in the future?
log:

13:22:03: Ожидание соединения с TH05-92848F
13:22:18: Model: TH05
13:22:18: Firmware: github.com/pvvx
13:22:18: Hardware: 0001
13:22:18: Software: V1.1
13:22:19: Dev info # hw: 0015, sw: 0011, services: 000042B0, sd: 0000
13:22:19: Устройство подключено.

[image]
[image]

pvvx, 2024-02-07:
Today there is only [TH05_V1.4](https://pvvx.github.io/TH-05).
I ordered a few more options on aliexpress.
I'll see which ones come...

jurgen2005, 2024-02-08:
I have this model but get the following message:
Connecting...
PHY62x2 - Reset Ok
Error read Revision!

something I did wrong?

pvvx, 2024-02-08:
RTS is not connected to Reset?

jurgen2005, 2024-02-08:
I think it is, like this?

[image]

CMY-Reader, 2024-02-09:
> I have this model but get the following message: Connecting... PHY62x2 - Reset Ok Error read Revision!
> 
> something I did wrong?

Linux OS?

Try to do manual reset. 
Disconnect reset wire. Run script. Then short once reset to the ground (BAT-).

jurgen2005, 2024-02-09:
Works! Thx!
OTA also works, appears as BThome, just no display...

`PHY62x2 Utility version 25.01.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00009100
Segment: 1fff0000 <- Flash addr: 0000c100, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000c500, Size: 00002de2
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00009100
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00001100 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000c100, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000c100... ok
Segment: 1fff1838 <- Flash addr: 0000c500, Size: 00002de2
Write 0x00002000 bytes to Flash at 0x0000c500... ok
Write 0x00000de2 bytes to Flash at 0x0000e500... ok
----------------------------------------------------------
Write Flash from file: BOOT_TH05_v11.hex - ok.
Send command 'reset' - ok
`

SaiQmon, 2024-02-22:
Any news on this v1.2 display? Flashed with USB, then OTA and now it's BTHome compatible OK - without local display :( 

Also, the "TH05_v12.bin" disappeared from BIN folder? Why?

SaiQmon, 2024-02-22:
Also, do not OTA flash the working v1.2 with "TH05V13_v14.bin", device goes fully unresponsive. I need to try USB flash to retrieve it.

pvvx, 2024-02-23:
In the file name "_v1.4.hex" is the firmware version number, not the device.

pvvx, 2024-02-23:
[Added firmware for TH05F](https://github.com/pvvx/THB2)

rbonat, 2024-02-23:
I can confirm that TH05Y_V1.2 works on TH05F_v14.bin with display :)
Excellent work!

SaiQmon, 2024-02-23:
Confirmed, TH05Y_V1.2 works on TH05F_v14.bin, display working. Great work, thanks!

VolunPotest, 2024-03-09:
Добрый день, @pvvx.
А что у этой версии по эноргопотреблению? (на оригинальной и вашей прошивке)
Есть ли разница с TH05?

pvvx, 2024-03-09:
Минимум двукратная по сравнению с оригинальной от Tuya.

VolunPotest, 2024-03-09:
> Минимум двукратная по сравнению с оригинальной от Tuya. Пока есть только такая статистика ![image](https://private-user-images.githubusercontent.com/12629515/311461799-3c312fc5-ad5f-4e45-8733-bd7c04f33edd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwMTI3MTIsIm5iZiI6MTcxMDAxMjQxMiwicGF0aCI6Ii8xMjYyOTUxNS8zMTE0NjE3OTktM2MzMTJmYzUtYWQ1Zi00ZTQ1LTg3MzMtYmQ3YzA0ZjMzZWRkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA5VDE5MjY1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzODRjZmRkZmMyMDFkMjNhM2Q1YzA2NmE3MGZiMmQ1ZmVkZTU5Mjg1MTdlMWU2MjQ2YzZhN2U4MDk4ZDhhYjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AJR3MHds3SvwzjIFOAkLYYKbb29PVNf_2xUnLntUtEc)

Двухкратная в лучшую сторону или худшую? Вижу на графики, что потребление снизилось после 6 марта

pvvx, 2024-03-09:
Напряжение CR2032 зависит от температуры и сильно.