pvvx/THB2 #15: TH05 new version V1.6

Notrial, 2024-02-09:
I got this one from Ali about two weeks ago. I suppose it is the latest one.
Can't get it to connect through Chrome, but I am on Mac so could that be the case?
Will try on PC a bit later today.

Photos of the board are attached.

[image]
[image]

spupetic, 2024-02-09:
You cannot flash OTA at first time, you need to flash it via serial and the commands in the readme, after that you can use the OTA website. See this issue: [https://github.com/pvvx/THB2/issues/11](https://github.com/pvvx/THB2/issues/11)

Notrial, 2024-02-10:
> You cannot flash OTA at first time, you need to flash it via serial and the commands in the readme, after that you can use the OTA website. See this issue: #11

Can I use this one to flash it as well?

[image]
[image]

spupetic, 2024-02-10:
> > You cannot flash OTA at first time, you need to flash it via serial and the commands in the readme, after that you can use the OTA website. See this issue: #11
> 
> Can I use this one to flash it as well?
> 
> ![2D9B87E9-C6A0-40C9-B22E-93BD38F6DFD6_4_5005_c](https://private-user-images.githubusercontent.com/36201754/303856825-fd00ad28-3f39-44d4-b6c7-5526e5618ada.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1ODg4MzYsIm5iZiI6MTcwNzU4ODUzNiwicGF0aCI6Ii8zNjIwMTc1NC8zMDM4NTY4MjUtZmQwMGFkMjgtM2YzOS00NGQ0LWI2YzctNTUyNmU1NjE4YWRhLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxMFQxODA4NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OGY4MzE3YmRkMTU1YTc4M2I5MjQ0NGVlNmU3M2QwZTgzMjBlNDk1YjliODk4YTg1OGNmZDhmMzg1N2EzODA1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NhUIAgAoU6lcuQfLYwMx0IDourfMo4fL8Ip-AIWmIjM) ![E0E09E01-0952-45A1-8F76-0BC73396EC98_4_5005_c](https://private-user-images.githubusercontent.com/36201754/303856826-48dad4e1-2457-4544-8bac-5f044d2015bc.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1ODg4MzYsIm5iZiI6MTcwNzU4ODUzNiwicGF0aCI6Ii8zNjIwMTc1NC8zMDM4NTY4MjYtNDhkYWQ0ZTEtMjQ1Ny00NTQ0LThiYWMtNWYwNDRkMjAxNWJjLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxMFQxODA4NTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MGExNDk2YjZmN2Q0MzM0ZGM4YWI2ZTIzMjE2NTU1NzZjZTE1ZDU1MDg1ZjNkMzU4YjFlYzFmNTU3Yzc2NmRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ozGIW_dRk0NCGcxQruuBhpjP-0UG8xsQDGCEogfMm3w)

If it has 3.3V logic level, you can. But be sure to check it first, because it might kill the board if it is for 5V devices.

Notrial, 2024-02-10:
There is a switch on it for 3.3 vs 5v, so should be good :)

spupetic, 2024-02-10:
> There is a switch on it for 3.3 vs 5v, so should be good :)

Yes, it should be good.

Notrial, 2024-02-15:
Just to confirm that the procedure works for V1.6.

Flashing boot one first and then full firmware.
I did have to short reset with VBAT- for a millisecond after running the script, otherwise, it wouldn't work.
[image]

dicamarques14, 2024-02-16:
> Just to confirm that the procedure works for V1.6.
> 
> Flashing boot one first and then full firmware. I did have to short reset with VBAT- for a millisecond after running the script, otherwise, it wouldn't work.
> 
> Not sure why I can't read the values in ESPHome though...I read MAC from the web tool and configured it like this:
> 
> <img alt="image" width="296" src="https://private-user-images.githubusercontent.com/36201754/305034916-d20b4169-7286-4744-b3a8-706f0a373770.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwOTg5NjksIm5iZiI6MTcwODA5ODY2OSwicGF0aCI6Ii8zNjIwMTc1NC8zMDUwMzQ5MTYtZDIwYjQxNjktNzI4Ni00NzQ0LWIzYTgtNzA2ZjBhMzczNzcwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE2VDE1NTEwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY1MzA3N2VlODc3OTQ0ZjA0ODY3ODVkM2ZhZDcxMmVjMDZjNGJmNzBiNjc3MjljY2VlOTUyNDRhODk5NjQ0OTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.850C6z-yOqqgXq189ocIgSmMXDVcoUJ9iixewvcq4wk">
> ![IMG_5426](https://private-user-images.githubusercontent.com/36201754/305034332-8bea5d65-f593-425b-acd1-a22cd6593e53.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwOTg5NjksIm5iZiI6MTcwODA5ODY2OSwicGF0aCI6Ii8zNjIwMTc1NC8zMDUwMzQzMzItOGJlYTVkNjUtZjU5My00MjViLWFjZDEtYTIyY2Q2NTkzZTUzLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNlQxNTUxMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMTI0NTRjNmQ0NDA1MjMyODI0YTQ4YzcyMjIxNTIwMDZmOTRmYmQ5YjY5OWZmN2E2NjA5ZWNmOTNiMTUxNzg3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TCgpNBoo3MP09I_YkyJtlWSWNBcEfSdZERYiZmLK5n4)
> 
> EDIT: I am an idiot and haven't realized this only works through BTHome. It broadcasts in there perfectly!

Im trying to flash my V1.5 with a bte17-06b based in CH340E and I cant get it to work... which pins did you connect to?
I tried to short the rst and tm pins in the board but I still dont get a connection on the python

Notrial, 2024-02-17:
I did what is says bellow...except the REST_N that i didn't connect to anything but shorted after starting python with VBAT-...you have to time it right.

[image]

pvvx, 2024-02-20:
On adapters without "RTS" you need to manually close "RESET" to "GND" and release it when starting rdwr_phy62x2.py.
It is difficult to manually open the contact at the right time. Therefore, automatic control via "RTS" is used.

pvvx, 2024-02-21:
[README](https://github.com/pvvx/THB2)

_Кнопка используется для быстрого подключения к старым BT-адаптерам. Нажатие кнопки переключает интервал BLE рекламы на более короткий период (1562.5 мс). Действие продолжится 60 секунд, затем интервал восстановится на установленный в настройках._

_The button is used to quickly connect to old BT adapters. Pressing the button switches the BLE advertising interval to a shorter period (1562.5 ms). The action will last 60 seconds, then the interval will be restored to the one set in the settings._

NeoMatrixJR, 2024-03-06:
is RESET = REST_N?  I'm getting the same errors, but I can't find "RESET"

Notrial, 2024-03-07:
Correct.

dicamarques14, 2024-03-07:
For those with trouble getting this right:
In the python script: rdwr_phy62x2.py at line 227, i set the timeout to "0.5", and after line 233 i added "print(read)"
So i just kept shorting RST with GND (like quick taps) until I saw a print with "cmd>>:"

NeoMatrixJR, 2024-03-07:
Turns out I may have had a janky adapter.  I switched over to windows with no luck, then changed out my adapter and it worked.

igorkietrz, 2024-03-19:
> For those with trouble getting this right: In the python script: rdwr_phy62x2.py at line 227, i set the timeout to "0.5", and after line 233 i added "print(read)" So i just kept shorting RST with GND (like quick taps) until I saw a print with "cmd>>:"

If you get it wrong does it say Error read Revision?

LordArrin, 2024-04-27:
Bought a new revision, flash it with BOOT_TH05F_v16.hex by instruction, and device didn't show in list. Any help?
[image]
 
[image]

[image]

[image]

pvvx, 2024-04-28:
TH05_V1.6 works fine. The [README](https://github.com/pvvx/THB2) describes what firmware is used for TH05_V1.6
Why did you flash the firmware from [TH05F](https://pvvx.github.io/TH05F/) if this is another option?

LordArrin, 2024-04-29:
Looks like it is my fault) Try it soon

Rudo1111, 2024-06-12:
Hello, I also have a problem with Error read Revision! out of 11, I flashed 9 and 2 still can't be flashed, has anyone encountered a similar situation? I tried advice like line 227 "0.5" and 233 "print(read)" and TM + VCC. It didn't help, could there be errors or try it on another computer and try again and again? :-) Thanks typ: 
[image]
[image]
TH05Y_V1.2 2023-12-06