pvvx/THB2 #16: Не восстановить дамп оригинальной прошивки на THB2

lehanspb, 2024-02-11:
Пытался и тот дамп, который сам снял:
python.exe rdwr_phy62x2.py -p COM3 -b 5000000 we 0 .\ff_thb2.bin

И тот, что на странице https://pvvx.github.io/THB2/ есть:
python.exe rdwr_phy62x2.py -p COM3 -b 5000000 we 0 .\ff_r11000000-00040000.bin

Одинаковая ошибка:
```
=========================================================
PHY62x2 Utility version 25.01.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
Reopen COM3 port 1000000 baud... ok
Write Flash data 0x00000000 to 0x00040000 from file: .\ff_r11000000-00040000.bin ...
Error set ext.Flash size 00400000!
```

Также пробовал перед восстановлением ключи -a и -e - без результата.

pvvx, 2024-02-11:
-b 5000000 не много ли?
И не все адаптеры могут установить ровно 1 000 000 baud rate. Ставят 921600 или что ещё.
Пробуйте другую скорость. И на 115200 не так уж долго будет работать.

lehanspb, 2024-02-11:
> -b 5000000 не много ли? И не все адаптеры могут установить ровно 1 000 000 baud rate. Ставят 921600 или что ещё. Пробуйте другую скорость. И на 115200 не так уж долго будет работать.

Да, многовато. Опечатался видимо. 
1000000 точно пробовал. Перепроверю ещё раз.
На самом деле хотел попробовать вернуть стоковую прошивку, чтобы удостовериться точно ли у меня THB2, т.к. пока могу судить только по фото платы (совпадает).

BOOT_THB2_v12.hex успешно залил, а вот далее по BLE уже подключиться не удаётся, хотя устройство что-то вещает с именем THB2XXXX. С кнопкой Reset пробовал.
https://pvvx.github.io/THB2/web/PHY62x2BTHome.html запускаю в Chrome под Android. 5 из 5 неуспешных попыток подключиться. К ATC_XXXX (lywsd03mmc) подключается.

lehanspb, 2024-02-11:
> Не указывайте параметр -b xxx. Будет по умолчанию 115200. На 1 Mbit может плохо работать из-за самих адаптеров и т.д.

Спасибо. Действительно в этом дело было. 

Залил ff_r11000000-00040000.bin. Устройство вещает с именем TH05. Так и должно быть?

lehanspb, 2024-02-11:
Попробовал заново:

` python.exe rdwr_phy62x2.py -p COM3 -e -r wh .\bin\BOOT_THB2_v12.hex
=========================================================
PHY62x2 Utility version 25.01.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008cc8
Segment: 1fff0000 <- Flash addr: 0000bcc8, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000c0c8, Size: 00002dc6
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008cc8
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000cc8 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bcc8, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bcc8... ok
Segment: 1fff1838 <- Flash addr: 0000c0c8, Size: 00002dc6
Write 0x00002000 bytes to Flash at 0x0000c0c8... ok
Write 0x00000dc6 bytes to Flash at 0x0000e0c8... ok
----------------------------------------------------------
Write Flash from file: .\bin\BOOT_THB2_v12.hex - ok.
Send command 'reset' - ok`

Далее результат https://pvvx.github.io/THB2/web/PHY62x2BTHome.html

`16:27:15: Поиск устройств
16:27:18: Ожидание соединения с THB2-FD88E4
16:27:33: Устройство отключено.
16:27:33: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with device.gatt.connect.
16:27:33: Переподключение 0 из 5
16:27:33: Ожидание соединения с THB2-FD88E4
16:27:44: NetworkError: Connection Error: Connection attempt failed.
16:27:44: Переподключение 1 из 5
16:27:44: Ожидание соединения с THB2-FD88E4
16:27:49: NetworkError: Connection Error: Connection attempt failed.
16:27:49: Переподключение 2 из 5
16:27:49: Ожидание соединения с THB2-FD88E4
16:27:54: NetworkError: Connection Error: Connection attempt failed.
16:27:54: Переподключение 3 из 5
16:27:54: Ожидание соединения с THB2-FD88E4
16:27:59: NetworkError: Connection Error: Connection attempt failed.
16:27:59: Переподключение 4 из 5
16:27:59: Ожидание соединения с THB2-FD88E4
16:28:04: NetworkError: Connection Error: Connection attempt failed.
16:28:04: Подключиться не удалось!`

Что я делаю не так?

pvvx, 2024-02-11:
[README](https://github.com/pvvx/THB2#)
* _Кнопка используется для быстрого подключения к старым BT-адаптерам. Нажатие кнопки переключает интервал BLE рекламы на более короткий период (1562.5 мс). Действие продолжится 60 секунд, затем интервал восстановится на установленный в настройках._

lehanspb, 2024-02-11:
> [README](https://github.com/pvvx/THB2#)
> 
> * _Кнопка используется для быстрого подключения к старым BT-адаптерам. Нажатие кнопки переключает интервал BLE рекламы на более короткий период (1562.5 мс). Действие продолжится 60 секунд, затем интервал восстановится на установленный в настройках._

Как и писал выше, эту рекомендацию видел и пробовал. Пробовал также подавать питание с нажатой кнопкой. Только имя устройства сменилось с THB2-FD88E4 на THB2-2ADD16.

pvvx, 2024-02-11:
* При стирании Flash стирается и прошлый MAC.
* Если прошивка не соответствует типу устройства, то кнопка не работает. Она на разных выводах у разных устройств.

lehanspb, 2024-02-11:
> * При стирании Flash стирается и прошлый MAC.
> * Если прошивка не соответствует типу устройства, то кнопка не работает. Она на разных выводах у разных устройств.
> 
> При нажатии кнопки загорается светодиод.

Устройство 1-в-1 как на этом фото https://pvvx.github.io/THB2/
Контакты соответствуют. На чипе PHY6222
При нажатии на кнопку синий светодиод загорается только в момент нажатия на неё. Может какая модификация? Датчик температуры с Али. Заказывал разные. У этого на коробке указано THB2

pvvx, 2024-02-11:
Значит ваш BT адаптер какой-то не такой. Пробуйте на другом устройстве.
Тут ничего не сделать.

lehanspb, 2024-02-11:
> Значит ваш BT адаптер какой-то не такой. Пробуйте на другом устройстве. Тут ничего не сделать.

lywsd03mmc успешно прошиваю вашей прошивкой этими адаптерами (и в смартфоне и на ноутбуке). Или для PHY требуется что-то особенное?

pvvx, 2024-02-11:
Период рекламы увеличен до 5 сек. У lywsd03mmc по умолчанию 2.5 секунды.

lehanspb, 2024-02-11:
Понял. Поищу другой BT-адаптер и попробую с ним.

P.S.: Благодарю за ваш труд! Около 20 штук lywsd03mmc в работе с вашей прошивкой. Даже в связке с tasmota на esp32 срок жизни батареи год или чуть более.

closed 2024-02-11 by lehanspb