pvvx/THB2 #20: TH-05 version and serial adapter  clarification

vasil-github, 2024-02-16:
Hi there and thanks for the work.

Firs I want to know is the TH-05 V1.2 supported or not?
In the documentation/readme it is stated that v1.4 is supported.  At the same time there are people who flashed  TH-05 v1.2 and you advise them how to complete the full OTA flash.

Second - When doing the initial flash via serial IS IT TTL or RS232 serial adapter needed?

rbonat, 2024-02-17:
Hi, 
Short answers:
TH05_V1.2 works ok as remote sensor but without display (display is blind). 
To flash bootloader you will need standard USB to TTL adapter working with 3.3V supply and logic levels.

vasil-github, 2024-02-20:
Thank you.
I did flash with TTL adapter a v1.2 device and as your wrote no display. But for testing purpose is ok.
One more thing. 
Is the BT advertisement/data readable/understandable by OpenMQTT gateway ? If you know.
The esp32 does "see" the TH sensor but data is not recognized/decoded in my setup.

rbonat, 2024-02-21:
I haven't tested openmqttgateway.
I'm using esphome bluetooth proxy on esp32.
Devices has been discovered automatically after flashing, as new
 BTHome device, and added to Home Assistant without any problems.
[image]

pvvx, 2024-02-21:
uuid: 0xfcd2 - BTHome 
data: "40000f015c02c00603dd120c6f0b"
40 - BTHome V2
000F - Msg count 15
015c - Battery 0x5c = 92%
02c006 - Temp: 0x06C0 = 17.28 C
03dd12 - Humi: 0x12DD = 48.29 %
0c6f0b - Vbat:  0x0b6f = 2927 mV

rbonat, 2024-02-23:
So you have at least three options now :)
- decode servicedata on controller side (f. ex. by HA template sensors)
- use one more esp32 with ble proxy loaded
- or use simple usb BT dongle (any with CSR 8510 chip works fine).
However from my experience esp32 have better range than BT dongle

pvvx, 2024-02-23:
> However from my experience esp32 have better range than BT dongle

RTL8761BU has the best reception characteristics in the class of cheap BT adapters. ESP32 is many times worse.
CSR8510 A10 is only BT4.2.

rbonat, 2024-02-23:
Thanks for this tip - do you think it's worth to replace my CSR8510 dongle by RTL8761BU one?

vasil-github, 2024-02-23:
> So you have at least three options now :)
>
>     * decode servicedata on controller side (f. ex. by HA template sensors)
>
>     * use one more esp32 with ble proxy loaded
>
>     * or use simple usb BT [...]

Using another USB BT adapter defeats the point of remote monitoring-that is why I am using OMG after all :-)
Using another ESP32 is overkill for reading just 1 another temp sensor. For the price of the ESP i will buy just new sensor, that is supported and if I decide I can flash it of over to ZigBee
I guess I will try the option for decode servicedata on controller side (f. ex. by HA template sensors) just for the fun of it. I guess I will spend some time reading and fooling around before I get bored or stuck and order new sensor :-)

Thank you for the effort and support!

pvvx, 2024-02-23:
> Thanks for this tip - do you think it's worth to replace my CSR8510 dongle by RTL8761BU one?

The CSR8510 receives the most BLE advertisements per second.
The RTL8761BU is significantly smaller. But it works with "LE long Range".
I use 2 adapters:
RTL8761BU - for receiving from remote sensors (from my second house 500 meters away)
CSR8510 - for receiving a local group of sensors (currently 40 pcs)

rbonat, 2024-02-23:
Thanks for clarification - as always there is no simple answer yes or no :) Maybe I will go your way, ending up with two receivers.

pvvx, 2024-02-23:
CSR8510
[image]

RTL8761BU https://github.com/pvvx/ATC_MiThermometer/issues/297#issuecomment-1877575447
ESP32 https://github.com/pvvx/ATC_MiThermometer/discussions/328#discussioncomment-7838023

vasil-github, 2024-02-23:
WOW 40 ble devices.... I will NOT do that to my self :-D
I preferre ZigBee for such number of devices. Much more reliable and extendable range. I have currently 72 ZB devices. 
I did have too many issues with BT over the years (disconnections, encryptions, range, sending commands, etc., etc.) Then I discovered OMG and BT is working flawlessly for me. Everything is send over MQTT, discovered automatically, not to mention that for 2 usd im adding bidirectionalls RF and IrDa to the gateway. And that is the first time I face some challenge with a device. Yes that was way before BLE proxy :-)  
Anyways the CSR8510 and RTL8761BU info is useful to known.

vasil-github, 2024-02-23:
Deferent problems, deferent solutions after all :-D