pvvx/THB2 #25: Настройки при включении Boot загрузчика. (защита от дурака)

Balaganoff, 2024-02-27:
Ситуация. Прошил бут, прошил ота , все ок. Начал счелкать настройки. В итоге установил максимум (255сек ) интервал посылок. 
Ну и я потерял устройство. Не находится. Так неохота опять выдергивать, плату чтобы стереть и перешить.
Как я понимаю защита от дурака, такого как я )) не предусмотрена и при загрузке в boot настройки в дефолтные не сбрасывает? 

И да ! Спасибо за отличную работу!

froloffw7, 2024-02-27:
Загрузитесь в FW BOOTб как описано в [README](https://github.com/pvvx/THB2/blob/master/README.md#fw-boot-%D0%B8-ota) и заново выполните OTA.

pvvx, 2024-02-28:
Выньте батарею и зажмите кнопку. Вставьте батарею. Должен загрузиться в boot. 
Нажатие кнопки меняет интервал на 1.6 сек для подключения

Balaganoff, 2024-02-28:
nRFConnect соединяется с первого раза , подаю команду 56, при следующем обновлении уже другое значение. 
Пробовал и в boot и в обычном режиме и батарейку откидывал. 
Не видит chrome на смартфоне и все тут ((  Причем я именно с этого смартфона OTA прошивал. 
[image]
[image]

pvvx, 2024-02-28:
Наверно в js [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html) стоит фильтр - не показывает устройства без имени... Надо поменять... Но это позже...

А пока введите '0100'  в сервис 0xFCD2 c характеристикой 0xFFF4. Сделает имя по умолчанию.

Balaganoff, 2024-02-28:
> Наверно в js [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html) стоит фильтр - не показывает устройства без имени или с хитрыми символами... Надо поменять... Но это позже, т.к. может быть проблема в "Web Blouetooth API" у Chrome и ничего сделать не выйдет.
> 
> А пока введите `0100` в сервис 0xFCD2 c характеристикой 0xFFF4. Сделает имя по умолчанию.

Вот спасибо ! Все поехало !

pvvx, 2024-02-28:
Не выходит заставить  "Web Bluetooth API" сканировать устройства со всеми возможными именами (региональными символами). Но русские вписал:
		deviceOptions.filters = "ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz_#@!*0123456789';,.<>{}[]абвгдеёжзийклмнопрстуфхцчшщъыьэюяАБВГДЕЁЖЗИЙКЛМНОПРСТУФХЦЧШЩЪЫЬЭЮЯ"
Т.е. только символ из списка может быть первым для того, чтобы Chrome нашел устройство.

[image]

[image]

Если есть какие другие варианты исправить фильтр сканирования в  "Web Bluetooth API" - напишите как.

almirus, 2024-03-25:
@pvvx позвольте мини оффтоп, а что за устройства с именем TY у меня в кв они тож есть (наверняка соседские)?

pvvx, 2024-03-25:
Обычно это Tuya WiFi розетки. Они настраиваются по BLE, если там не ESP82xx. В таких розетках всё работает и по BLE и по WiFi. Но это никого из "писателей" Умных домов не интересует. 
Таким розеткам не обязательно соединение с Cloud Tuya при включении в сеть, в отличии от розеток сделанных на ESP чипе.

closed 2024-12-11 by pvvx