pvvx/THB2 #32: Error read revision on TH05_V1.5 date 2023-05-17

ianderso, 2024-03-11:
I have connected a fairly standard FTDI to the board, soldering leads to TX1, RX1, vbat+, vbat-, and REST_N. with the REST_N connected to RTS, TX1 into RX, RX1 into TX. 

When I run this command:
`python3 rdwr_phy62x2.py -p /dev/cu.usbserial-A50285BI -e -r wh bin/BOOT_TH05_v16.hex` 

This is the result:

```
PHY62x2 Utility version 26.02.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Error read Revision!
```

having put several print statements into the code, it is returning the 'cmd>>' with the initial connect, but not responding at all with the "rdrev+ " command in ReadRevision.

Any help would be appreciated!

pvvx, 2024-03-11:
https://github.com/pvvx/THB2/issues/21#issuecomment-1989207530

ianderso, 2024-03-12:
worked in windows. thank you!

closed 2024-03-12 by ianderso