pvvx/THB2 #35: Standalone Android App

draekko, 2024-03-15:
Not directly related to this firmware but related to the devices, got myself a BTH01, have managed to get the firmware installed (using v1.6 currently) but i was wondering if anyone had knowledge of an Android app to get the data without using Home Assistant or other home server setup and read the devices directly? Preferably open sourced.

My use case for using these is for on the go but after a couple of days of web searches i can't find anything that isn't related to home automation which all seem to require a server setup of some kind which i don't want or need.

Thanks in advance.

pvvx, 2024-03-15:
https://apps.rustore.ru/app/ru.kompius.thermo_plung

draekko, 2024-03-16:
Thanks for the info, still new to the ble game.

draekko, 2024-03-30:
Ended up with a quick and dirty forking of the Sensirion Smart Gadget app and the updated code is in a temprory repo here https://github.com/draekko/BLESensor

The app was super old so spent part of yesterday and all of this morning updating it, works ok to report temp & humidity and gives some charting but its rough and dirty hack to the code base to see whats going but at least it works standalone.

Will probably end up writing one from scratch over time because there aren't many safeguards for permission checks and since its got none of the back end service updates that newer androids require it really needs to be redone from scratch.

Its a nice app too bad they never updated it over the last 8 yrs LOL.

repo is here https://github.com/draekko/BLESensor

[image]
[image]
[image]
[image]

draekko, 2024-03-30:
OOOps i cropped out the viewpager out of the images by mistake ... oh well it gives an idea of what it looks like.

draekko, 2024-03-30:
Oh and write now it only polls the device and it does so too often in my opinion but thats how the original code was, also before i forget it doesn't pull down any history from the device or do anything else.