pvvx/THB2 #36: Trying to connect the TH05 through html page shows Выбрано неверное устройтво! (Invalid device)

igorkietrz, 2024-03-18:
Hello, how can I fix this problem?
[image]

pvvx, 2024-03-18:
You are required to read the [README](https://github.com/pvvx/THB2?tab=readme-ov-file#%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0).

igorkietrz, 2024-03-19:
> You are required to read the [README](https://github.com/pvvx/THB2?tab=readme-ov-file#%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0).

To open up this device i need to remove the 3 plastics holding the pcb?

pvvx, 2024-03-19:
You did not describe which version of TH05.

igorkietrz, 2024-03-19:
> You did not describe which version of TH05. And all TH05s have one cover that is removed to replace the battery. TH05_V1.4 is the worst. Has high battery consumption. TH05_V1.4 is not worth buying.

I unfortunately have the TH05_V1.4, so can I flash it through ota? Sorry if I’m asking but the Google translation of the page is kinda bad 😅

pvvx, 2024-03-19:
From the Tuya version, everything is flashed via UART.

igorkietrz, 2024-03-19:
> From the Tuya version, everything is flashed via UART.

Thank you

igorkietrz, 2024-03-19:
It’s really hard to perfectly short reset with gnd 😅, is it possible to program that using arduino uno with removed atmega?

igorkietrz, 2024-03-19:
Okay. I managed it to work, but unfortunately it shows Error read revision. I am using Windows and shortening the vbat- (connected to gnd) with REST_N. Is there something I can try?

igorkietrz, 2024-03-19:
> From the Tuya version, everything is flashed via UART.
> 
> To flash TH05_V1.4 firmware, you must completely remove the printed circuit board. Same thing with TH05F. Other models do not require this.

I managed to make it work. It’s now unfortunately showing Error read revision. I am using Windows and shortening the vbat-(connected to gnd) with REST_N. Is there something I can do to make it work? Thank you in advance

closed 2024-03-20 by igorkietrz