pvvx/THB2 #38: THB1 bluetooth problems

rainer38, 2024-04-01:
Hi,

thanks for your effort for this nice project. I already updated my Xiaomi Thermometers and they work fine with your firmware. :)

Now i got a few cheap Tuya THB1 and tried to flash your firmware on them.

Therefor i soldered an usb2ttl serial adapter to the board and executed the following two commands:

python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -e -r wh bin/BOOT_THB1_v16.hex
python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -r we 0x10000 bin/THB1_v16.bin

They executed successfully and the LCD shows "bot 16" after the first and "app 16" after the second command.
I can also see some bluetooth beacons with "hcitool lescan --duplicates".

The problem starts when i try to connect to the device through this webpage from yours:
https://pvvx.github.io/THB2/web/PHY62x2BTHome.html

I started the page in chromium on android to get the bluetooth web api and can see my device in the list. But when i try to connect to the device i get an connection failure error.

These are the screens from my android device:
[image]
[image]

The translation of the messages doesn't help me at all where the problem could be.

Maybe you have a hint for me what i can try to connect to the device.

Thanks
Rainer

pvvx, 2024-04-01:
1. On Linux, you must configure Bluez to be compatible with the Bluetooth SIG standard. This is done in the Bluez configuration files.
2. To quickly connect, you need to briefly press the button on the thermometer. A short press of the button shortens the BLE advertising transmission interval to one and a half seconds. The action lasts 80 seconds to enable scanning and connection to the thermometer on non-standard devices and for Linux (Bluez).

rainer38, 2024-04-03:
I flashed the other 9 devices and i can connect to all of them.
Only to the first one i can't connect.
But this is not a great problem compared to how cheap they were.

Thanks for your help and the ticket can be closed.

closed 2024-04-03 by rainer38

pvvx, 2024-04-05:
Try writing with erasing the entire flash (option `-a`):
```
python3 rdwr_phy62x2.py -p COMxx -a -r wh BOOT_xxx_vxx.hex
```

Areas protected from normal erase may contain some settings that are incorrect (RF, ADC).

rainer38, 2024-05-20:
Sorry for the late answer but now it works.
Today i reflashed v1.7 with your command and i can connect to the device.

Thanks for the help.