pvvx/THB2 #41: Tuya THB1

fcrovy, 2024-04-15:
Hi,
I didn't understand if I can use OTA ([PHY62x2BTHome.html web page) to change the original firmware of the THB1 sensor or if it is necessary to load, at least the bootloader, via USB-COM converter.
Because from OTA web page the device is seen with the name THWL but then the connection is rejected due to an unrecognized device.
Thank's in advance,
Fabio

kopchik, 2024-04-24:
I think it's necessary to flash the boot firmware first. It wasn't straightforward for me: https://github.com/pvvx/THB2/issues/44 .

fcrovy, 2024-04-29:
thank's a lot. I have flashed the boot without mistake and than loaded the firmware with OTA

DON5PEDRO, 2024-07-01:
After the firmware is wired, the OTA does not connect normally to the phone and the TUI gateway. All the firmware went fine - there were no errors. On the page [PHY62x2BTHome.html] connects, all settings are available and can be read and changed.
What's wrong? How can I restore the connection to the devices in normal mode? 
I pressed the reset button briefly, several times including - it does not help.

pvvx, 2024-07-01:
> After the firmware is wired, the OTA does not connect normally to the phone and the TUYA gateway

The firmware works with Home Assistant and other software running in the [BTHome](https://bthome.io/) format.

DON5PEDRO, 2024-07-02:
> > После прошивки OTA не подключается нормально к телефону и шлюзу TUYA
> 
> Прошивка работает с Home Assistant и другим ПО, работающим в формате [BTHome](https://bthome.io/) . Работа с Tuya не поддерживается. Для работы с Tuya восстановите оригинальную прошивку.

Thanks!
I flashed this sensor because it consumed too much power, I even converted its power supply to an 18650 battery - and even it ran out in less than 30 days (18650 is a new 3500mAh). Is there anything we can do about it?