pvvx/THB2 #44: flashing BTH01 issues

kopchik, 2024-04-24:
Hi!

JFYI I'm trying to flash BTH01 from UART, and I think I'm having some issues not described anywhere.

The first one is, I needed to connect TM (TestMode) pin to VBAT+. Otherwise PHY6222  doesn't communicate via UART.

The second issue is, my BTH01 starts serial at 115200, while `rdwr_phy62x2.py` script initiates connection at 9600. So, I had to change `START_BAUD` to `115200` in order to be able to connect to it.

My module looks exactly like on https://pvvx.github.io/BTH01/, pcb version `BTH01Y_V1.2` with datecode on PCB silkscreen `2024-02-29` .

pvvx, 2024-04-24:
I have never connected TM anywhere on dozens of devices and hundreds of firmwares. There were no problems. Same with UART speeds.

kopchik, 2024-04-25:
Ah, I see, the IC can also be programmed without TM pin.

May be something changed in recent firmware? I attach the dump here, in case this is of any interest. Sadly I didn't check the behavior after flashing your BOOT firmware.

I guess we can close this issue for now.

[fw_orig.bin.zip](https://github.com/pvvx/THB2/files/15119826/fw_orig.bin.zip)

closed 2024-04-25 by kopchik

333Violet, 2024-09-05:
Привет!
Я пытаюсь прошить BTH01 через UART ,  что делаю не так?  Не запускается утилита для прошивки.
PS C:\Users\Violet\AppData\Local\Programs\Python\Python312> cd C:\Users\Violet\AppData\Local\Programs\Python\Python312\THB2
PS C:\Users\Violet\AppData\Local\Programs\Python\Python312\THB2> pip3 install -r requirements.txt
Requirement already satisfied: pyserial in c:\users\violet\appdata\local\programs\python\python312\lib\site-packages (from -r requirements.txt (line 1)) (3.5)
PS C:\Users\Violet\AppData\Local\Programs\Python\Python312\THB2> python3 rdwr_phy62x2.py -p COM4 -e wh ./bin/BOOT_BTH01_v18.hex
Python
PS C:\Users\Violet\AppData\Local\Programs\Python\Python312\THB2>

valga, 2024-11-21:
@pvvx Got the same issue with BTH01Y_V1.2 2024-01-03.

UART is dead unless TM is connected to VCC.

And when TM is wired to VCC, device goes into programming mode even without sending `UXTDWU` to it, speed is 115200:

```
*******************************
Current port: COM9
Current baudrate: 115200
Current stopBits: 1
Current parity: No
Serial opened!! 
*******************************
UART RX : cmd>>:
UART TX ASCII: rdrev+
UART RX : 0x001364c8 6222M005 #OK>>:
Serial port COM9 closed!!
*******************************
```

The script works just fine after that:

```
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008b88
Segment: 1fff0000 <- Flash addr: 0000bb88, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000bf88, Size: 00002bca
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008b88
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000b88 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bb88, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bb88... ok
Segment: 1fff1838 <- Flash addr: 0000bf88, Size: 00002bca
Write 0x00002000 bytes to Flash at 0x0000bf88... ok
Write 0x00000bca bytes to Flash at 0x0000df88... ok
----------------------------------------------------------
Write Flash from file: ./bin/BOOT_BTH01_v19.hex - ok.
Send command 'reset' - ok
```