pvvx/THB2 #45: Advertising packets of the BTH-01 sensor does not contains usefull information

Nikolay-Ch, 2024-04-28:
Hi,

I have BTH-01 sensor, and I flash it with latest firmware THB2.
But ad-packets contains only RSSI info...
```
Advertisement received.
Device Name: BTH01-FD2387
Device ID: **ID of the device**
RSSI: -61
TX Power: -128
UUIDs:
```

No temperature, no pressure, no battery info in AD-packet...

Regards,
Nikolay

Nikolay-Ch, 2024-04-28:
Device configuration:
[image]

pvvx, 2024-04-28:
> I have BTH-01 sensor, and I flash it with latest firmware THB2.

The firmware for BTH-01 is BOOT_**BTH01**_v17.hex, not **THB2**.

Nikolay-Ch, 2024-04-28:
> The firmware for BTH-01 is BOOT_**BTH01**_v17.hex, not **THB2**.
I understand, I called it by the name of the repository
Of course, I flashed the correct firmware, otherwise the device would not have worked at all.

> Where do You look for BLE advertise data?
I use OpenMQTTGateway - it sends raw data, if not undertands it...
Another way is to use [Google Chrome](https://googlechrome.github.io/samples/web-bluetooth/watch-advertisements-async-await.html)

pvvx, 2024-04-28:
On Linux and Windows, scanning to the "Web Bluetooth API" does not work.
See the implementation status of the scanning function for the "Web Bluetooth API".