pvvx/THB2 #49: Can't flash TH05

B4nd1, 2024-05-16:
I have two TH05 device, but different revisions. I am trying to flash the Boot hex with a raspberry pi 4 via the UART pins. 
I don't have an RST pin so I am trying to close the RESET pin with GND(-Vbat) but It's not working
I can't connect to the device, I tried this https://github.com/pvvx/THB2/issues/15#issuecomment-1982967330 and I got this

```
(env) pi@raspberrypi:~/THB2 $ python3 rdwr_phy62x2.py -p /dev/ttyS0 -e -r wh ./bin/BOOT_TH05F_v17.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!
```

The device has power and I connected it right rx-tx, tx-rx
here is the device
[image]
[image]

The other one is a TH05_V1.6, and I am having the same problem.
I sucessfully flashed a BTH1 yesterday, that works great with home assistant.
I don't understand what am I missing

pvvx, 2024-05-16:
GND -> BAT-

B4nd1, 2024-05-16:
Yes, i connected it right, the board has power

pvvx, 2024-05-16:
You release RESET at the wrong time

B4nd1, 2024-05-16:
When should I release? Right now I changed it to be slower and I am trying to tap it. This worked with the BTH1

B4nd1, 2024-05-17:
I bought this device because it has a reset pin, but I still have the same issue
https://www.aliexpress.com/item/1005006739710034.html

B4nd1, 2024-05-17:
So, I figured it out. I had to close the TM pin to Vcc and keep it there while the flash happened.
I flashed the TH05_V1.6 and I installed the APP via bluetooth with the webapp succesfully.

I flashed the TH05Y_V1.2 too, with the /BOOT_TH05F_v17.hex, but I can't find it via bluetooth.
Here is the output of the flash, it looks great, but I can't find it with bluetooth. ( I already tried pushing the button, it makes no difference and I tried to find it with multiple devices)
```
python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -e -r wh ./bin/BOOT_TH05F_v17.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
b''
b''
b''
b''
b''
b'cmd>>:'
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008f7c
Segment: 1fff0000 <- Flash addr: 0000bf7c, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000c37c, Size: 00002bde
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008f7c
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000f7c bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bf7c, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bf7c... ok
Segment: 1fff1838 <- Flash addr: 0000c37c, Size: 00002bde
Write 0x00002000 bytes to Flash at 0x0000c37c... ok
Write 0x00000bde bytes to Flash at 0x0000e37c... ok
----------------------------------------------------------
Write Flash from file: ./bin/BOOT_TH05F_v17.hex - ok.
Send command 'reset' - ok
```

B4nd1, 2024-05-18:
I think my device has some issues with its antenna. I put my phone right on top of it and I found it. 

I edited my previous comment and added how did I flash the software successfully.

closed 2024-05-18 by B4nd1