pvvx/THB2 #51: TH05_V1.4 + TH05_v17 display not working

DavidAntonin, 2024-05-18:
Hi, flashed my TH05_V1.4 with serial usb adapter to BOOT_TH05_v17.hex Than I made OTA TH05_v17.bin. I have now TH05 in Home Assistant but display is off. There is no information even on battery connection. Display is still empty.

amateusz, 2024-05-20:
Same here. Toggling the setting in the web flasher "disconnect display" does not change it, still nothing on the LCD
(LCD is not broken, checked with an another device full original firmware)

pvvx, 2024-05-21:
> Same here. _EDIT: I realised I have a TH05Y_V1.2 (PCB marking). I flashed `BOOT_TH05_v17` and it still works, expect for LCD.. 😛_
[README](https://github.com/pvvx/THB2/tree/master#%D0%BF%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B8-boot-%D0%B8-app): 

_TH05Y_V1.1, TH05Y_V1.2 with QD01 2332 NT chip -> BOOT_TH05F_v17.hex_

pvvx, 2024-05-21:
> Hi, flashed my TH05_V1.4 with serial usb adapter to BOOT_TH05_v17.hex Than I made OTA TH05_v17.bin. I have now TH05 in Home Assistant but display is off. There is no information even on battery connection. Display is still empty.

[TH-05](https://pvvx.github.io/TH-05/) ?

closed 2024-05-21 by DavidAntonin

amateusz, 2024-05-21:
I guess we both made the same mistake by picking TH05_V1.4 from by the photo line-up, as the TH05F_V1.2 does not have its photo in readme. I might add it