pvvx/THB2 #56: battery level calculation

fcrovy, 2024-06-23:
Hi,
i see in the code that the battery level is 0% with 2 volts and 100% with 3 volts.
I have uder test the sensor BTH01 and, if the voltage is under 2.5 volts, the temperature value is in fault (more than 100°C).

In the [Tuya developer page](https://developer.tuya.com/en/docs/iot/zigbee-temperature-and-humidity-sensor-common-faq?id=K9tp1656vjg9r&_source=github) indicate that 0% is at 2.5V.

I suggest to change the code in according to Tuya indication.

Fabio

pvvx, 2024-06-24:
This is due to the fact that [BTH01](https://pvvx.github.io/BTH01/) contains the cheapest sensor CHT8305. But another sensor may be installed.
Datasheet: [CHT8305](https://pvvx.github.io/BTH01/CHT8305.pdf) -> Vcc min 2.5V, max 5 V
The CHT8305 is not designed to run on two AAA batteries.

pvvx, 2024-06-25:
Test BTH01. To test operation on a weak source, a panel was soldered in for the CR20xx battery. Runs on an old CR2025 battery. Then I replaced it with CR2032.
There were no changes in the readings at 2.4V.
[image]

ressu, 2025-05-05:
Funny you should mention 2.4V, I seem to find that that's the point where readings start to wobble a lot on TH05F. So this isn't specific to BTH01. Considering that different devices seem to get different performance from the battery voltage due to quality variance, it would be useful to have this as a configurable value (assuming there is storage space available)

[image]

pvvx, 2025-05-05:
This is a problem with the CHT8305 sensor and not the entire device.
I have always recommended not to buy such thermometers in which the CHT8305 sensor is installed.

> Datasheet: [CHT8305](https://pvvx.github.io/BTH01/CHT8305.pdf) -> Vcc min 2.5V, max 5 V

ressu, 2025-05-05:
Yeah, the mechanism for the issue is different. But the original request of configurable battery level lower boundary is pretty much the same. Being able to configure when the battery is considered to be 0% makes it easier to monitor the battery level and replace the batteries as needed.

I do agree that Tuya recommendations are not great, which is why a configurable lower boundary is a better solution.

pvvx, 2025-05-05:
> I do agree that Tuya recommendations are not great, which is why a configurable lower boundary is a better solution.

Setting the lower limit of the battery level is not suitable for many options.
At different temperatures, the voltage and remaining working charge of the battery are different. Especially for CR2032.
The current settings show the working battery level according to the levels of the main chip's ability to work.
There is no point in making corrections for Tuya's hackwork. All thermometers with CHT8305 sensors should be in the trash.