pvvx/THB2 #60: “TH05” with QS01 2340 NT

Nanganator, 2024-07-03:
I just received 2X “TH05” devices with the TH05F board layout but with the QD01 2340 NT chip on them. Is the QD01 2332 NT firmware compatible with this chip?
[image]
[image]
[image]

pvvx, 2024-07-03:
The type is determined by the display. By smiley symbol.
https://pvvx.github.io/TH-05/ or https://pvvx.github.io/TH05F/

Nanganator, 2024-07-03:
Could it perhaps be some kind of new combination. It has the smiley face of the TH05 like this one:
[image]
But flashing the TH05_v17.bin firmware makes the display stop working. All other functions are working.

pvvx, 2024-07-03:
The printed circuit board is clearly different - from TH05F.
We need to try the TH05F firmware.

closed 2024-07-12 by pvvx