pvvx/THB2 #61: How to start

Floriszz, 2024-07-25:
I read the readme and can't even get through the first step...I feel really stupid.
I already have python on my windows pc.
For some reason Github has made it difficult to download the repo, as there is now an 'Open' button for Gitpod only. Or I might have done something stupid.
But the requirements.txt file has only 'pyserial' in it. Trying to install this with:
`pip3 install -r pyserial`

```
C:\WINDOWS\system32>pip3 install -r  pyserial
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'pyserial'
```

J0k3r2k1, 2024-07-30:
click on "code" and download ZIP, unpack and copy all files to your python directory
open powershell
type: pip3 install -r requirements.txt
(this will pull+install all necessary files)
go on with the other steps...

HumbleDeer, 2024-08-17:
The `-r` flag for Pip is to specify a `requirements` file. What you're doing in this case is telling Pip to use a `requirements`-type file that's called `pyserial` in your current directory.

To install packages without a `requirements` file, ommit that `-r` flag. Get Pip help with `pip -h`.

HumbleDeer, 2024-08-17:
> move to your python directory
This should be
> Move to inside the project directory
i.e. the pvvx/THB2 repository, either downloaded and extracted or cloned using `git clone`.

Floriszz, 2024-09-22:
Thank you all for the support and tips. 
Managed to execute to find out I already had pyserial after all... Now waiting on the USB-com 3.3v adapter. And fingers crossed I can get the BTH01 flashed...

- Download flash package from github as zip; THB2-master.zip
- Extract to a folder like:  C:\Python\THB2-master
- Open command prompt (make sure python is in the path setting of Windows)
- cd to the folder where you extracted THB2-master.zip.  In my case : C:\Python\THB2-master
- Start the following command: C:\Python\THB2-master>pip3 install -r requirements.txt