pvvx/THB2 #63: Error writing the FW-APP

eph2, 2024-08-22:
Hello. I'm trying to write the FW APP to my BTH01.

The firmware was successful uploaded, but when I try to upload the FW APP some errors occurs.

I tryed from OTA page, and the command line method too. 

Here are the code from the command line:

```shell
% python3 rdwr_phy62x2.py -p /dev/cu.SLAB_USBtoUART -b 115200 -e wh bin/BOOT_BTH01_v17.hex
python3 rdwr_phy62x2.py -p /dev/cu.SLAB_USBtoUART -b 115200 -r we 0x10000 bin/BTH01_v17.bin
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008b64
Segment: 1fff0000 <- Flash addr: 0000bb64, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000bf64, Size: 00002bca
----------------------------------------------------------
Erase Flash work area... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008b64
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000b64 bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bb64, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bb64... ok
Segment: 1fff1838 <- Flash addr: 0000bf64, Size: 00002bca
Write 0x00002000 bytes to Flash at 0x0000bf64... ok
Write 0x00000bca bytes to Flash at 0x0000df64... ok
----------------------------------------------------------
Write Flash from file: bin/BOOT_BTH01_v17.hex - ok.
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
Write Flash data 0x00010000 to 0x0001bde4 from file: bin/BTH01_v17.bin ...
Erase sector Flash at 0x00010000... ok
Erase sector Flash at 0x00011000... ok
Erase sector Flash at 0x00012000... ok
Erase sector Flash at 0x00013000... ok
Erase sector Flash at 0x00014000... ok
Erase sector Flash at 0x00015000... ok
Erase sector Flash at 0x00016000... ok
Erase sector Flash at 0x00017000... ok
Erase sector Flash at 0x00018000... ok
Erase sector Flash at 0x00019000... ok
Erase sector Flash at 0x0001a000... ok
Erase sector Flash at 0x0001b000... ok
Write 0x00002000 bytes to Flash at 0x00010000... ok
Write 0x00002000 bytes to Flash at 0x00012000... ok
Write 0x00002000 bytes to Flash at 0x00014000... ok
Write 0x00002000 bytes to Flash at 0x00016000... ok
Write 0x00002000 bytes to Flash at 0x00018000... ok
Write 0x00001de4 bytes to Flash at 0x0001a000... error!
Error: Write Flash!
``` 
And here are the OTA error:

```shell
20:43:47: Start OTA flashing...
20:44:18: OTA: NotSupportedError: GATT operation failed for unknown reason.
20:44:18: Device is disabled.
``` 
On the status area this are the message when the FW area trying to be uploaded:
Block transferred 15 from 3039, 0% succeeded, total time 0.006 seconds

Sometimes, OTA returns this message too:
Error: Error (7) during block transmission 15 OTA: Package loss

I'm attaching the "resume" of the sensors to this post too, the voltage are ok! 
[image]

And, I'm attaching a photo of the board too.
[image]

eph2, 2024-08-23:
I'm using a MAC, when I try to upload the FW APP with OTA on a Windows computer, it worked.

pvvx, 2024-08-23:
Only Linux, Windows, Android are supported.

eph2, 2024-08-23:
> Only Linux, Windows, Android are supported. APP programming has problems in the ROM chip program. Therefore, it is recommended to use OTA for APP.

Nice, Victor. Thanks for your reply!

Straying from the topic of this post, but taking advantage of the fact that I have doubts.

I configured the BTH01 in Home Assistant, using BTHome, everything worked perfectly, but only two entities are coming without values ​​(as shown in the image below). When clicking on them, they say they are disabled. Is this normal?

[image]

Second question: when the sensors were connected to the Tuya system, the integration with Home Assistant notified me when the devices became unavailable. Is there any way I can monitor this now? I imagine it has to do with the entities mentioned above, but maybe there is another way. When I remove the batteries from the sensors, they just stop sending the signal and do not update the value. Should I use this time of the last update as a "sign" of unavailability?

Thanks in advance!

pvvx, 2024-08-24:
> I configured the BTH01 in Home Assistant, using BTHome, everything worked perfectly, but only two entities are coming without values ​​(as shown in the image below). When clicking on them, they say they are disabled. Is this normal?

This is diagnostic data and is not needed by everyone. When activated, the data will be written to the history, to the HA database. This will reduce the media resource and increase the size of the database.

> Second question: when the sensors were connected to the Tuya system, the integration with Home Assistant notified me when the devices became unavailable. 

This is configured in the integration package you are using (Bluetooth, BTHome).

eph2, 2024-08-28:
NIce, thanks, Victor!

pvvx, 2024-09-11:
Two options:
1. Long BLE advertising interval. Before attempting to connect, you must press the button on the thermometer.
2. Chip glitch. At supply voltage over 2.6 V, some batches of PHY6222 chips have a special instability of the internal timer. This is most often found in BTH01. Software correction of the internal oscillator drift is performed over a long interval. That is, try connecting later or install a discharged battery.

Staars, 2024-11-12:
Two observations on MacOS Sequoia on a M1Max:

1. In MacOS I did not even pass the first step of seeing a successful reset in many, many tries.
2. Using a virtual machine (Ubuntu Noble server/Virtualbox) solved this completely with a 100% success rate on the same Macbook.