pvvx/THB2 #66: TH05_V1.4 flashed okay, but no display

remy, 2024-09-30:
I've successfully flashed the TH05_v1.5 with the v18 firmware, and it appears on my home assistant BT Home device list.

However, I don't have any display (and I've checked the settings and "diconnect display" is *not* set).

I thought this might be a bad device, so I repeated the process on a second TH05 that I had, and exactly the same thing has happened - everything works, but I get no display.

Any ideas what this might be - or can I provide any more informtion to help debug?

pvvx, 2024-09-30:
TH05 exists in several versions, with different displays. Specify the model of your thermometer and flash it with the appropriate firmware.

[README.md](https://github.com/pvvx/THB2)
_The device type can be distinguished externally by the smiley face symbol on the screen._

remy, 2024-10-01:
Ah, so I went with a visual match which lead me to the TH05_1.4 (as per the subject), but now inspecting my PCB, it has TH05Y_V1.2 but the LCD matches the TH05_1.4 (i.e. using brackets for the sides of the face).

Anyway, now I've flashed based on the board with TH05F_v18 firmware, the LCD works.

closed 2024-10-01 by remy

CodeAsm, 2024-11-21:
This closed issue actually help me recheck my pcb. I dunno if its my dutch or the readme english, I thought purely checking the display look determines what version, but the one with ('-') can have different firmware, my TH05Y_V1.2 pcb worked with working display with the TH05F_v19 firmware (display was enabled by default, now also have time displaying)

So cool it works now. im gonna order another one for the other room now. Thanks @remy for asking and finding the trick