pvvx/THB2 #67: Feature Request: Temperature range calibration

Cossid, 2024-10-05:
The CHT8310 supports the following temperature ranges

```
Temperature accuracy with calibration:
±0.2℃ (typ.) from 0℃ to 80℃
±0.4℃ (typ.) -20℃ to 100℃
±0.5℃ (typ.) -40℃ to 125℃
```

And the default appears to be `±0.2℃ (typ.) from 0℃ to 80℃`

It would be nice to choose between these 3 ranges to make these sensors useful in freezers or cold outdoor settings.

This is just a sample from the CHT8310 specsheet, but CHT8305 and AHT20 also have similar calibration ranges with different accuracy zones.

pvvx, 2024-10-06:
https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8310.Advanced.Datasheet_Ver1.0.20230407.pdf

Cossid, 2024-10-06:
How about alkaline like the THB2 & BTH01?  I'm aware cold weather isn't ideal for any battery, and alkaline supplies a lower starting voltage that lithium, but they do hold up better in colder environments typically.

pvvx, 2024-10-07:
At the moment, there are statistics on the use of 2 AAA saline batteries in [TS0201](https://pvvx.github.io/TS0201_TZ3000/) in the freezer compartment of the refrigerator.
Works in BLE - LE Long Range (PHY Coded) mode so that the signal can penetrate the iron body of the refrigerator. This is a little more power consumption than a typical BLE (PHY 1M).
[image]
The following effect has been observed:
When batteries are heated briefly, their subsequent output increases. That is, AAA salt batteries freeze, and when defrosted briefly, partial recovery occurs.

pvvx, 2024-10-07:
Behavior of the CHT8305 sensor in [TH05_V1.3](https://pvvx.github.io/TH05-v1.3) when the battery voltage drops below 2.5V:
[image]

[THB1](https://pvvx.github.io/THB1/) (CHT8310) remains operational at 2.0V:
[image]

Cossid, 2024-10-07:
Right, the THB2 with CHT8310 and 2xAAA batteries is my desired target for low temperature sensing.

I know 2xAAA is typically the standard for other outdoor/low temperature sensors, but it would be nice to have these as a potential, even if less efficient, option, as the price point for these sensors is substantially lower than more robust solutions.

pvvx, 2024-10-08:
Thermometers with CR2032 are not suitable for use at large negative temperatures. Thermometers with larger CRxxxx batteries (CR2050, ...) withstand negative temperatures a little better, since CR2032 under normal conditions has a maximum output current equal to the pulse current consumption of the microcircuit.

Cossid, 2024-10-08:
I've had terrible quality on CR2450 battery supply here, even indoors, most of the ones available at a reasonable price only last about 6-8 months.

TLSR825x is hard to find in AA with a good sensor and not requiring physical mods.

Which is why I would like to see this calibration option available for the THB2 with the 8310 and AAA batteries.  AAA battery quality is easy to control, and even at 6 months in cold weather wouldn't be a huge deal for most, since they're so accessible.

pvvx, 2024-10-08:
> Which is why I would like to see this calibration option available for the THB2 with the 8310 and AAA batteries. 

THB2 does not have a display. Accordingly, it is better to make adjustments in the receiving program. This is easy to do in Home Assistant and/or other programs.

Cossid, 2024-10-08:
I don't follow.  I don't think it's as simple as an offset.  The readings are accurate in the 0-50C range, but below 0C, they become skewed, for example -18C actual temperature reads at about -13C

I think the sensor needs configuration changes in the 0x03 register (but I'm not certain).  It sounds like the default mode just isn't meant to accurately report below 0C.

pvvx, 2024-10-08:
> I think the sensor needs configuration changes in the 0x03 register (but I'm not certain).

There is no setup of measurement ranges.
There is only a switch to reduced accuracy.

Cossid, 2024-10-08:
[image]

I read that as it needs reduced accuracy to go below 0C

Cossid, 2024-10-08:
Most freezers are around -18C which is a reasonable thing to monitor.  I would not expect anyone to use these for -40C or +90C.

I don't fully understand the data format, but the spec sheet says the sensor should be able to handle it, but it currently is for sure skewed below 0C.  To be fair, Tuya's firmware has the same limitation.

Here is a comparison against an Inkbird IBS-TH2 (unsure what sensor is inside).
[image]

Above 0C they would be much closer together, so I'm assuming it is something with the configuration, as the CHT8310 clearly can handle ranges below 0C.

pvvx, 2024-10-08:
You haven't understood the most important thing: Tuya is a device from the trash heap (the cheapest configuration, often just defective parts from factories, parts that have been sitting in warehouses - instead of being recycled, assembled by small private companies, no guarantee, no checks, no certificates, ...)

Cossid, 2024-10-08:
I understand these are poorly made devices, but the specs say they should be able to handle this quite easily.

pvvx, 2024-10-08:
CHT sensors have very crooked characteristics.
Nothing good can be done with them.

pvvx, 2024-10-08:
> Above 0C they would be much closer together, so I'm assuming it is something with the configuration/driver, as the CHT8310 clearly can handle ranges below 0C.

Please set the coefficients for CHT8310 sensor correctly in the [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html) program.

Cossid, 2024-10-08:
25606 is already the present value (as initiated by the device-specific firmware installed).

And it looks like adjusting that applies a linear adjustment, which is not what is incorrect here.  At 0C-30C, this sensor is within tolerance of it's spec, but at -18C it is several degrees off.

Cossid, 2024-10-08:
That is saying humidity won't be accurate below 0C, and that is fine.  The line above says temperature to -40C.  It doesn't look like you can disable humidity reading in any way though, but I wouldn't expect it to have an effect on the temperature portion.

pvvx, 2024-10-08:
The Chinese factory that produces CHT sensors has translated the available PDF file from Chinese to English. This document cannot be trusted.
It is not known what sensors are in the THB2. They may be defective CHT8215 that failed the tests and were marked specifically for a small underground company that assembles THB2 and sells it on the market...

pvvx, 2024-10-08:
I put THB2 (with the case removed) in the freezer next to TS0201(BLE). There is also a regular thermometer - its readings coincide with TS0201(BLE).
Warm batteries in THB2 started to heat TS0201...
Without the case, the THB2's reaction to cooling is faster...
[image]

Cossid, 2024-10-30:
Thanks for all the feedback.  After further research, I have found I had an unknown warm spot in my freezer near the compressor.
Relocating the devices and also trying a different freezer improved the results.  Closing this issue, as those potential adjustments do not seem relevant.

closed 2024-10-30 by Cossid