pvvx/THB2 #72: Feature Request: RTC calibration

Deoptim, 2024-10-24:
Здравствуйте.
У меня два датчика **BTH01**, один Outside (на улице) диапазон +40 до -10 , другой в Freezer chamber (в морозилке) от -19 до -28.
Вроде бы кварц или сама микросхема должны показывать разное время отставания, но в принципе они примерно на одном уровне.
На скриншотах ниже оно показывает отставание за месяц.
<details>
 
[![Screenshot_2024-10-24-13-53-55-933_com android chrome][1]][1]
[![Screenshot_2024-10-24-13-55-43-711_com android chrome][2]][2]

[1]: https://github.com/user-attachments/assets/a9359e98-15dc-49a1-b627-78283fd2aadc
[2]: https://github.com/user-attachments/assets/56fbc66f-60a6-4250-92c2-2cf7f89321b9
</details>

Хотелось бы чтобы появилась функция ввода коэффициента калибровки RTC (даже желательно сделать чтобы можно было ввести три коэффициента калибровки RTC в зависимости от текущей температуры датчика).

pvvx, 2024-10-24:
В программе применен плавающий коэффициент поправки счета времени.
У чипов PHY622x2 существует  проблема - сверх большая нестабильность счета RC генератора от напряжения питания.
Некоторые партии PHY62222 невозможно заставить работать от напряжения питания более 2.5В. Возникает уход счета RC генератора несовместимый с параметрами спецификаций BLE для соединения. Стандартный SDK от производителя не справлялся с коррекцией - не работало соединение.
По этим причинам пришлось патчить стандартный SDK от производителя и вписывать другой алгоритм динамической поправки счета RC генератора. 
Этот алгоритм плавающий - поправка меняется каждый цикл по мере набора счетчика и в таком варианте говорить о каких-то статических коэффициентах нет смысла.

У меня более десятка устройств на PHY622x2 - работают для теста прошивок с момента первых публикаций альтернативных прошивок. Везде уход часов разный. На некоторых меняется со временем то в плюс, то в минус.

universam1, 2025-01-25:
made an automation script for https://github.com/pvvx/THB2/pull/125

pvvx, 2025-01-25:
@universam1 
```
Auto Time Sync
-------------
Program will run for 10 minutes
Status: Scanning for devices...
Status: Connecting to device A4:C1:38:05:4E:0A (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with A4:C1:38:05:4E:0A (attempt 1/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:50:38: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device A4:C1:38:05:4E:0A (attempt 2/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with A4:C1:38:05:4E:0A (attempt 2/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:50:48: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device CGG_11C3B2 (attempt 1/2)...
ERROR: Error with CGG_11C3B2 (attempt 1/2): Could not get GATT services: Unreachable
21:51:08: ERROR: Could not get GATT services: Unreachable
Status: Connecting to device CGG_11C3B2 (attempt 2/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with CGG_11C3B2 (attempt 2/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:51:23: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device CGG_11C3B2 (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with CGG_11C3B2 (attempt 1/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:51:42: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device CGG_11C3B2 (attempt 2/2)...
ERROR: Error with CGG_11C3B2 (attempt 2/2): Could not get GATT services: Unreachable
21:52:04: ERROR: Could not get GATT services: Unreachable
Status: Connecting to device TH5_F8CA3B (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with TH5_F8CA3B (attempt 1/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:52:15: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device TH5_F8CA3B (attempt 2/2)...
ERROR: Error with TH5_F8CA3B (attempt 2/2): Connection process timed out
21:52:52: ERROR: Connection process timed out
Status: Connecting to device TH5_F8CA3B (attempt 1/2)...
ERROR: Error with TH5_F8CA3B (attempt 1/2): Could not get GATT services: Unreachable
21:53:08: ERROR: Could not get GATT services: Unreachable
Status: Connecting to device TH5_F8CA3B (attempt 2/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with TH5_F8CA3B (attempt 2/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:53:18: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device TH05D-50B5B9 (attempt 1/2)...
ERROR: Error with TH05D-50B5B9 (attempt 1/2): Could not get GATT services: Unreachable
21:53:37: ERROR: Could not get GATT services: Unreachable
Status: Connecting to device TH05D-50B5B9 (attempt 2/2)...
ERROR: Error with TH05D-50B5B9 (attempt 2/2): Could not get GATT services: Unreachable
21:53:54: ERROR: Could not get GATT services: Unreachable
Status: Connecting to device TH05D-50B5B9 (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
Status: Connected to TH05D-50B5B9
21:54:05: Device time: 2025-01-26 01:22:45
21:54:05: Time difference: 1720 seconds
21:54:05: Time difference > 3s, updating device time...
21:54:05: Time set successfully
21:54:05: Time updated, disconnecting...
Status: Connecting to device TH5_14A68A (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with TH5_14A68A (attempt 1/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:54:13: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device TH5_14A68A (attempt 2/2)...
Status: Connected, discovering services...
[... 17 lines cut ...]
Status: Setting up notifications...
ERROR: Error with CGG_145087 (attempt 2/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:55:33: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device THSz36D56B (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with THSz36D56B (attempt 1/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:55:49: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device THSz36D56B (attempt 2/2)...
ERROR: Error with THSz36D56B (attempt 2/2): Could not get GATT services: Unreachable
21:56:10: ERROR: Could not get GATT services: Unreachable
Status: Connecting to device THSz36D56B (attempt 1/2)...
Status: Connected, discovering services...
Status: Setting up notifications...
ERROR: Error with THSz36D56B (attempt 1/2): Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
21:56:28: ERROR: Characteristic 0000fff4-0000-1000-8000-00805f9b34fb was not found!
Status: Connecting to device THSz36D56B (attempt 2/2)...

Shutting down due to user interrupt
21:56:44: Shutting down...
21:56:44: 10 minute runtime completed
```
OS Windows 10.
I can see more than 40 BLE with custom firmware through this adapter.
https://github.com/pvvx/ATC_MiThermometer have the same protocol but different UUID for commands.

universam1, 2025-01-26:
go ahead and adjust as needed, I can only test the THB2 devices for now

pvvx, 2025-01-26:
> go ahead and adjust as needed, I can only test the THB2 devices for now

With this approach, this script will drain all batteries of other devices.
Such a connection with a BLE device creates a large peak in consumption at the very beginning of the connection.
This is caused by Bleak reading the entire UUID table from the device with a short polling interval.
Plus, if the device has a PIN code, then starting the connection causes even more battery consumption.

In this regard, I do not recommend using this script.

universam1, 2025-01-26:
I don't see the need to save a few ms runtime when the script is run once a month or so, but if there is a portable pkg alternative available fine with me

pvvx, 2025-01-26:
Firstly, not a few ms, but tens of seconds.
And secondly, it consumes energy from all devices with its active scanning and even more from those working in the BTHome format.
Thirdly, when the CR2032 battery is discharged, there may not be enough energy for such a connection. But the device can still transmit BLE advertising with data for BTHome for months.
Do you need to provide a consumption chart of the devices this script is trying to connect to?