pvvx/THB2 #74: radio bLE issues with all THB01

universam1, 2024-10-28:
Amazing work with this project @pvvx - really appreciating and exiting!

I bought 6x THB01 and all show the same connection issues. Right from the first BOOT flash it is almost impossible (maybe 1 of 30 attempts) to connect to the devices to **try** OTA. So I went ahead with UART flashing that solved that problem. However, the connection is so extremly fragile, it disconnects after few seconds. OTA is basically impossible, I was only once (from 100x tries or so) able to do, trying to downgrade to 1.7 FW if that solves the issue.

Tried with 2 different Windows Laptops and 2 different Android devices, tablet and phone.

One day later all devices stopped to be announce via BLE now! Even with NRF connnect I can only guess by the RSSI which devices they are, connection impossible. I tried power unplug, oder batteries and of course pressing the buttons.

I sounds to me the PLL for the Radio might have an issue.

Any idea please, is it possible that the BLE setup is broken

universam1, 2024-10-28:
wonder if this could be related: https://github.com/pvvx/THB2/issues/38#issuecomment-2040327261

universam1, 2024-10-28:
Some more feedback, that makes a huge difference for the majority of the devices. Around 80% work now after clearing the whole flash once:

> Try writing with erasing the entire flash (option -a):
python3 rdwr_phy62x2.py -p COMxx -a -r wh BOOT_xxx_vxx.hex
Areas protected from normal erase may contain some settings that are incorrect (RF, ADC).

Is that something that helps to solve an underlying problem with tunings, maybe for the remaining devices where this does not help?

pvvx, 2024-10-28:
THB01 - what is this variant?

On some PHY6222 series chips, the RC oscillator is unstable.
The BLE connection stabilizes when the supply voltage drops to 2.5 V. Or over time, when the RC oscillator drift correction, calculated relative to the crystal oscillator, accumulates.

universam1, 2024-10-28:
Not sure if this photo helps much 
[image]

In this case the drift correction went from bad to completely broken, if you wish

pvvx, 2024-10-28:
This is THB1.
THB1 work fine until battery voltage drops to 1.9V.

[image]

Replaced the battery with a new one.
[image]

Checked OTA:

---

19:46:18: Ожидание соединения с THB1-4ACDD9
19:46:26: Model: THB1
19:46:26: Firmware: github.com/pvvx
19:46:26: Hardware: 0017
19:46:26: Software: V1.8
19:46:26: Device info # hw: 0017, sw: 0018, services: 000043B8, sd: 0000
19:46:26: Устройство подключено.
19:46:30: Загрузка firmware файла 'THB1_v18.bin'...
19:46:30: Файл: bin/THB1_v18.bin
19:46:30: Файл id:PHY6, Сегментов: 3, Старт: 0x"1FFF1838, : 50096
19:46:30: Размер файла: 50100 байт
19:46:30: Счетчик: 3132 блоков
19:46:32: Переключение на...
19:46:32: Переподключение
19:46:32: Устройство отключено.
19:46:32: Ожидание соединения с THB1-4ACDD9
19:46:35: Устройство отключено.
19:46:35: Ожидание соединения с THB1-4ACDD9
19:46:39: Model: THB1
19:46:39: Firmware: github.com/pvvx
19:46:39: Hardware: 0017
19:46:39: Software: B1.4
19:46:39: Device info # hw: 0017, sw: 0014, services: 000002A1, sd: 0000
19:46:40: OTA ver: 01
19:46:40: Устройство подключено.
19:46:42: Старт программирования...
19:47:35: Программирование завершено за 52.184 секунды
19:47:39: Устройство отключено.

---

Everything is fine.

What is problem?

universam1, 2024-10-28:
Thank you for checking with your devices! That proves that the code is working. And after clearing all flash with -a flag, most devices I have can be connected. 
However some still can not be connected, or the connection is interrupted as soon as data is sent. With nrf connect Android app I can reproduce that. 
Might be a hardware problem, I don't know if something can be done for that though?

pvvx, 2024-10-28:
Under the Tuya brand, devices are produced from various junk. Tuya devices do not have any guarantees or certificates.

The PHY documentation mentions the capacitance in the RF part power supply for RF stability. It is not known what they put in the device assembled from junk. It is also not known what PHY chips were used. They may be from a defective batch from the factory...

universam1, 2024-10-29:
I see your point that wasting time on cheap hardware doesnt pay off - however before I performed the -a full erase none of those devices did work. Now at least 80% work as expected. So I don't see what I should do wrong here.

Here is an example of a problematic device. Those devices light up the bluetooth icon for about 1s then the connection is dropped. No stable connection possible.

```
...\src\THB2> python rdwr_phy62x2.py -p COM5 -a -r wh bin\BOOT_THB1_v18.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
---- Segments Table -------------------------------------
Segment: 11003000 <- Flash addr: 00003000, Size: 00008f6c
Segment: 1fff0000 <- Flash addr: 0000bf6c, Size: 00000400
Segment: 1fff1838 <- Flash addr: 0000c36c, Size: 00002bd6
----------------------------------------------------------
Erase All Chip Flash... ok
Segment Table[03] <- Flash addr: 00002000, Size: 00000130
Write 0x00000130 bytes to Flash at 0x00002000... ok
Segment: 11003000 <- Flash addr: 00003000, Size: 00008f6c
Write 0x00002000 bytes to Flash at 0x00003000... ok
Write 0x00002000 bytes to Flash at 0x00005000... ok
Write 0x00002000 bytes to Flash at 0x00007000... ok
Write 0x00002000 bytes to Flash at 0x00009000... ok
Write 0x00000f6c bytes to Flash at 0x0000b000... ok
Segment: 1fff0000 <- Flash addr: 0000bf6c, Size: 00000400
Write 0x00000400 bytes to Flash at 0x0000bf6c... ok
Segment: 1fff1838 <- Flash addr: 0000c36c, Size: 00002bd6
Write 0x00002000 bytes to Flash at 0x0000c36c... ok
Write 0x00000bd6 bytes to Flash at 0x0000e36c... ok
----------------------------------------------------------
Write Flash from file: bin\BOOT_THB1_v18.hex - ok.
Send command 'reset' - ok
```

universam1, 2024-10-30:
This is powered at 3.3V via USB. Also tried lower voltages of 2.5V or even batteries. Same behaviour.

xyzzy42, 2024-10-30:
When I powered a BTH01 from the USB serial adapter it did not work reliably.  Interval between BLE beacons was inconsistent.  Attempting to connect would rarely succeed.  And once connected it would last a most a few seconds before being disconnected.

Just removing all soldered wires and changing to batteries (NiMH now at 2.65 V) has fixed it.  Interval between received beacons is now a multiple of 5 seconds.  Once connected it stays that way for hours.  Connecting is not perfectly reliable, but I think this is caused by the long connection interval with a too short timeout on the host.

universam1, 2024-10-30:
Interesting! So you are saying reducing the default setting of the interval should improve the likelihood for a successfull connection? 
Now unfortunately we have a chicken egg problem, that I cannot change the defaults without an initial connection... @pvvx would it be possible to fix comipile into the boot firmware a shor interval and longer timeout?

pvvx, 2024-10-30:
[README](https://github.com/pvvx/THB2?tab=readme-ov-file#main-features)
* _The button is used for quick connection to old BT adapters. Pressing the button switches the BLE advertising interval to a shorter period (1562.5ms). The action will continue for 60 seconds, then the interval will be restored to the one set in the settings._

xyzzy42, 2024-10-30:
Yes, button works for me.
With normal CI, I can connect with Nordic Connect app on Android, bluetoothctl on Linux, and Chrome on Android.  But Chrome on Linux will always fail.   And the ones that work are not 100%.
Using the button to reduce CI allow Chrome on Linux to connect to.  And success rate seems to be nearly 100% on other devices.

This is with batteries.  Same device on USB serial cable power was nearly useless.

There are lots of reasons that power from an USB-serial cable could cause problems beyond voltage.  Batteries provide a very clean low noise supply.  Cheap USB serial cable does not at all.

A good device will have a well designed power supply filter caps and work on less than perfect power from the USB serial adapter.  But will a Tuya device have extra capacitors that shouldn't be absolutely necessary on battery power?  Of course not.

pvvx, 2024-10-31:
There is no stability from the battery under pulse load. For example, a half-discharged CR2032 has an internal resistance of 100 Ohm. The current of the BLE device changes from 2 μA to 8 mA. Accordingly, the supply voltage jumps by 0.8 V (3.0V -> 2.2 V).

universam1, 2024-10-31:
The connection is stable for ~1s before it drops, visible via the icon.
Could it be that the 900ms(?) timeout is playing a role here?

pvvx, 2024-10-31:
The RC oscillator is corrected with a drift of 20%. Large deviations are not taken into account in the average deviation. 

The Bluetooth Core Specifications Version 5.0+ require the active clock to have a drift less than or equal to ±50 ppm.

It is impossible to get such accuracy in PHY62xx chips from RC oscillator.

pvvx, 2024-10-31:
> Could it be that the 900ms(?) timeout is playing a role here?

This is more related to the constants of different intervals in the SDK and the receiving adapter.
Each version of the PHY SDK contains different timing constants. The reason is the inaccuracy of the clock.
This is more related to the constants of different intervals in the SDK and the receiving adapter. Each version of the PHY SDK contains different synchronization constants. The reason is the inaccuracy of the clock, with a large dependence on the batch of chips. It is impossible to correct these deviations.

universam1, 2025-01-25:
so about 30% of my devices from the same reseller and batch (?) have this problem, looks like there is probably no QC at the manufacturer.

As you mention, they are cheap and low quality. Guess that is what you get for the money.
Hence closing the issue for now

closed 2025-01-25 by universam1