pvvx/THB2 #76: Help with "non-standard" TH05

booroondook-eng, 2024-11-09:
I have very strange TH05. Its board looks exactly like TH-05F on these pictures: 
https://pvvx.github.io/TH05F/img/TH05F_pcb1.jpg
and
https://pvvx.github.io/TH05F/img/TH05F_pcb2.jpg
And has the same mark as on photo - TH05Y_V1.1 2023-10-10 - and the same big chip - QD01/2332 NT/CJT 3180
But its display looks like TH-05_V1.4 as on this picture:
https://pvvx.github.io/TH-05/img/Box_sm.png
(The smiley has another form)

I tried to flash my device with TH-05F firmware but it does not work as a result.
Then I flashed it with TH-05 (without letter "F') firmware. And now my device works (it sends temperature and humidity values to my Home Assistant), but its display is off.

pvvx, 2024-11-10:
## Internal GPIO pinout:

| Firmware | Sensor SDA/SCL/Power | LCD SDA/SCL/Power | Key | LED |
|:---:|:---:|:---:|:---:|:---:|
| THB1 | P18/P20/- | P34/P33/- | P01 | - |
| THB2 | P18/P20/- | - | P07 | P26 |
| THB3 | P18/P20/- | P34/P33/- | P01 | - |
| BTH01 | P33/P34/P00 | - | P14 | P15 |
| TH05 | P33/P34/P00 | P32/P34/P02 | P14 | - |
| TH05D | P31/P32/- | P11/P14/- | P02 | - |
| TH05F | P33/P34/P00 | P26/P31/P02 | P14 | - |

booroondook-eng, 2024-11-10:
I resolved the problem by flashing BOOT v1.7 and APP v1.9 (both for TH05F)

pvvx, 2024-11-10:
Boot V1.9 works normally on TH05F. In previous versions there is an error if you assign another name to the device.