pvvx/THB2 #77: TH05 V1.6

PEPEct, 2024-11-10:
II have the TH05 v1.6 and I can't find it when updating the firmware, is there any possibility of updating it another way?
I have tried wiring the board but I only have Arduino Mega board, but I am new and I don't know how to link the board to be able to upload a .hex or .bin file to the TH05. Is it possible?

egzumer, 2024-11-11:
You need USB to UART converter, then follow: https://github.com/pvvx/THB2?tab=readme-ov-file#firmware

BTW: The readme says to connect RTS to RESET. This is wrong, DTR should be used instead of RTS.

pvvx, 2024-11-11:
> BTW: The readme says to connect RTS to RESET. This is wrong, DTR should be used instead of RTS.

[RTS is duplicated by the DTR signal.](https://github.com/pvvx/THB2/blob/master/rdwr_phy62x2.py#L221C1-L229C1)

```
def Connect(self, baud=DEF_RUN_BAUD):
		self._port.setDTR(True) #TM   (lo)
		self._port.setRTS(True) #RSTN (lo)
		time.sleep(0.1)
		self._port.flushOutput()
		self._port.flushInput()
		time.sleep(0.1)
		self._port.setDTR(False) #TM  (hi)
		self._port.setRTS(False) #RSTN (hi)
```

egzumer, 2024-11-11:
For some reason RTS doesn't work on my FT232RL. I've checked with an oscilloscope, there is no level change.

PEPEct, 2024-11-11:
> You need USB to UART converter, then follow: https://github.com/pvvx/THB2?tab=readme-ov-file#firmware
> 
> 
> 
> BTW: The readme says to connect RTS to RESET. This is wrong, DTR should be used instead of RTS.

Thanks, i'll buy a USB to UART. 
Btw, is there a posibility to use ArduinoMega board to use it as USB to UART?

egzumer, 2024-11-11:
> is there a posibility to use ArduinoMega board to use it as USB to UART?

In theory you probably could, you would have to write a program to mirror input serial port to some output port at a correct baud rate, but honestly, UART converter is a $2 thingie (CP2102 dongle on aliexpress), and you should have one either way if you play with microcontrollers, it is usefull for many things.