pvvx/THB2 #78: device does enter OTA mode

universam1, 2024-11-11:
This problem appears with v1.8 trying to upgrade to v1.9. The device is reconnected but not entering into OTA mode. 
Appears on THB1 and BTH01 devices I have. 

Example logs:

```
Time	Temperature	Humidity	Battery level
2024-11-11 11:21:47	21.62 °C	51.95 %RH	100 % (3.072 V)	In: 0, Out: 0, Cmf: 0, Te: 0, He: 0, rds: 0
ConfigOTAChartService
File:

APP Firmware: BTH01_v19.bin Update Boot: 
Purge log file State: Connected.
11:22:20: Reconnect
11:22:20: Waiting for connection to null
11:22:26: Model: BTH01
11:22:26: Firmware: github.com/pvvx
11:22:26: Hardware: 0014
11:22:26: Software: V1.8
11:22:27: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:22:27: Connected.
11:22:29: Download firmware file 'BTH01_v19.bin'...
11:22:29: File: bin/BTH01_v19.bin
11:22:29: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:22:29: Filesize: 48644 bytes
11:22:29: Counter: 3041 blocks
11:22:29: Switch to...
11:22:29: Reconnect
11:22:29: Device is disconnected.
11:22:29: Waiting for connection to null
11:22:29: Device is disconnected.
11:22:29: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
11:22:29: Reconnect 0 / 5
11:22:29: Waiting for connection to null
11:22:34: Model: BTH01
11:22:34: Firmware: github.com/pvvx
11:22:34: Hardware: 0014
11:22:34: Software: V1.8
11:22:34: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:22:34: Connected.
11:22:35: Download firmware file 'BTH01_v19.bin'...
11:22:35: File: bin/BTH01_v19.bin
11:22:35: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:22:35: Filesize: 48644 bytes
11:22:35: Counter: 3041 blocks
11:22:36: Switch to...
11:22:36: Reconnect
11:22:36: Device is disconnected.
11:22:36: Waiting for connection to null
11:22:36: Device is disconnected.
11:22:36: TypeError: Cannot read properties of null (reading 'getPrimaryService')
11:22:36: Reconnect 0 / 5
11:22:36: Waiting for connection to null
11:22:40: Model: BTH01
11:22:40: Firmware: github.com/pvvx
11:22:40: Hardware: 0014
11:22:40: Software: V1.8
11:22:41: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:22:41: Connected.
11:22:42: Download firmware file 'BTH01_v19.bin'...
11:22:42: File: bin/BTH01_v19.bin
11:22:42: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:22:42: Filesize: 48644 bytes
11:22:42: Counter: 3041 blocks
11:22:43: Switch to...
11:22:43: Reconnect
11:22:43: Device is disconnected.
11:22:43: Waiting for connection to null
11:22:43: Device is disconnected.
11:22:43: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
11:22:43: Reconnect 0 / 5
11:22:43: Waiting for connection to null
11:22:46: Model: BTH01
11:22:46: Firmware: github.com/pvvx
11:22:46: Hardware: 0014
11:22:46: Software: V1.8
11:22:46: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:22:46: Connected.
11:22:48: Download firmware file 'BTH01_v19.bin'...
11:22:48: File: bin/BTH01_v19.bin
11:22:48: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:22:48: Filesize: 48644 bytes
11:22:48: Counter: 3041 blocks
11:22:49: Switch to...
11:22:49: Reconnect
11:22:49: Device is disconnected.
11:22:49: Waiting for connection to null
11:22:49: Device is disconnected.
11:22:49: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
11:22:49: Reconnect 0 / 5
11:22:49: Waiting for connection to null
11:22:52: Model: BTH01
11:22:52: Firmware: github.com/pvvx
11:22:52: Hardware: 0014
11:22:52: Software: V1.8
11:22:52: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:22:52: Connected.
11:22:54: Download firmware file 'BTH01_v19.bin'...
11:22:54: File: bin/BTH01_v19.bin
11:22:54: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:22:54: Filesize: 48644 bytes
11:22:54: Counter: 3041 blocks
11:22:55: Switch to...
11:22:55: Reconnect
11:22:55: Device is disconnected.
11:22:55: Waiting for connection to null
11:22:55: Device is disconnected.
11:22:55: TypeError: Cannot read properties of null (reading 'getPrimaryService')
11:22:55: Reconnect 0 / 5
11:22:55: Waiting for connection to null
11:22:59: Model: BTH01
11:22:59: Firmware: github.com/pvvx
11:22:59: Hardware: 0014
11:22:59: Software: V1.8
11:23:00: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:23:00: Connected.
11:23:01: Download firmware file 'BTH01_v19.bin'...
11:23:01: File: bin/BTH01_v19.bin
11:23:01: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:23:01: Filesize: 48644 bytes
11:23:01: Counter: 3041 blocks
11:23:02: Switch to...
11:23:02: Reconnect
11:23:02: Device is disconnected.
11:23:02: Waiting for connection to null
11:23:02: Device is disconnected.
11:23:02: TypeError: Cannot read properties of null (reading 'getPrimaryService')
11:23:02: Reconnect 0 / 5
11:23:02: Waiting for connection to null
11:23:05: Model: BTH01
11:23:05: Firmware: github.com/pvvx
11:23:05: Hardware: 0014
11:23:05: Software: V1.8
11:23:05: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:23:05: Connected.
11:23:06: Download firmware file 'BTH01_v19.bin'...
11:23:06: File: bin/BTH01_v19.bin
11:23:06: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 48640
11:23:06: Filesize: 48644 bytes
11:23:06: Counter: 3041 blocks
11:23:07: Switch to...
11:23:07: Reconnect
11:23:07: Device is disconnected.
11:23:07: Waiting for connection to null
11:23:08: Device is disconnected.
11:23:08: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
11:23:08: Reconnect 0 / 5
11:23:08: Waiting for connection to null
11:23:11: Model: BTH01
11:23:11: Firmware: github.com/pvvx
11:23:11: Hardware: 0014
11:23:11: Software: V1.8
11:23:12: Device info # hw: 0014, sw: 0018, services: 00004398, sd: 0000
11:23:12: Connected.
```

Any idea?

pvvx, 2024-11-12:
OTA works only through Boot.
(Options for switching in Boot are specified in the README.)
Disconnect the battery. Press the button and connect the battery.

universam1, 2024-11-12:
What's the purpose of this button "Mode OTA" then?
[image]

universam1, 2024-11-12:
I realize, this button does not restart into OTA mode, unless I press Disconnect manually which causes the device to reboot. Weird

pvvx, 2024-11-12:
Based on your descriptions, it is not known why switching to Boot on command does not work.
There may have been failures in the Boot firmware or Boot from another version of the thermometer.
Therefore, try switching using the button on the thermometer and turning the battery on and off

universam1, 2025-01-25:
I found that from v20 the problem is solved. Thanks!

closed 2025-01-25 by universam1