pvvx/THB2 #81: Battery drainage (data recorded too often in Home Assistant)

RexioGit, 2024-11-14:
Hi @pvvx,
thank you for your great work!

I am using 2x TH05F and THB1. I flashed BOOT to them and after that I installed OTA. I left all settings default exept one - I turned off displays as I dont need them. I paired all 3 sensors to Home Assistant (installed at RPi 3B+) but after 2 days Voltage dropped from 3,3V to 2,89V so im guessing that this will be a short life of this batteries.

I saw that sometimes Home Assisntant is pulling data every 2-5s and for me is way to often. I tried to change something in settings to make delays longer (i would like to set something betwen 5-30minutes) but to be honest I dont know what am I doing as sidenotes are in RU and names of variables are not so clear for me (I am not electrician nor programmer). Can someone add examples and more descriptions to documentation of all available settings?

Can you help me to set it up and running? My gol is to run this sensors for at least 6 months at one battery.

pvvx, 2024-11-14:
Default settings:

[image]
[image]

RexioGit, 2024-11-14:
so basically if we can change the data measurement frequency we can run this sensor until the battery self-discharges.

can you tell me what to change to make measurements every 3-30 minutes? 

It would be great if you can add some more detailed notes in documentation or at your html. It is always easier for me when I can see some examples. Im not to bright ;) 

Thank you very much for you great work!

pvvx, 2024-11-30:
> a year would be good

It won't. Some thermometers with cheap sensor chips and LCD screen drivers consume quite a lot in sleep mode. And the PHY6222 itself is not the best in terms of consumption. As a result, with a processor doing nothing, etc., a year on a cheap CR2032 does not work out.

marvinlehmann, 2024-11-30:
> Öhm... how can i change from ms to s ? If i set the advertising to 60000ms = 1min, it corrects self to 15937.5.... whatever a max limit this is...

You can't. But the advertising interval will be multiplied with the measurement step so the sensor won't measure too often.

pvvx, 2024-11-30:
> If i set the advertising to 60000ms = 1min

The Bluetooth standard specifies a maximum of 10 seconds.