pvvx/THB2 #84: TH05Y_v3.1 with TH05_V1.4 screen - TEMPERATURE/HUMIDITY not working

Pinni3, 2024-11-19:
Flashed the unit(s) did try the firmwares BOOT_TH05_vNN.hex And **BOOT_TH05F_vNN.hex**
Is a **TH05_V1.4 screen** with the marking **TH05Y_v3.1** on pcb , chipset is **QD01 2340 NT CJT 3353C**
After OTA No screen nor TEMP or humidity info , battery is displayed 
[image]
[image]

**BOOT_TH05F_v18.hex** is running atm but still no data for temp and humidity 
[image]
[image]

Is there a new firmware needed or coud this be adjusted with actions from my side?

pvvx, 2024-11-20:
> Is a TH05_V1.4 screen with the marking TH05Y_v3.1 on pcb 

Perhaps this is a new version of the printed circuit board and the sensor layout is done on other Soc pins.

Pinni3, 2024-11-20:
How can i help in getting this on track ?
Pictures , measurements ?

Pinni3, 2024-11-20:
[image]
[image]

pvvx, 2024-11-20:
PCB layout is different from TH05Y_v1.1/v1.2

tomasz-kos, 2024-11-20:
Hi, I have the same device. On my device temperature sensor is CHT8320 (marking ML). From datasheet I got that it uses 0x88 I2C address which after short checking isn't use in that code. As I have few of these devices I will try to log I2C communication on not reprogrammed device.

Pinni3, 2024-11-20:
[image]

pvvx, 2024-11-20:
Sensylink 
| Order PN | Address | Accuracy | Green1 | Package | Marking ID2 | Packing | MPQ | Operation Temperature
|-- | -- | -- | -- | -- | -- | -- | -- | --|
|CHT8320ADNR | 0x88 | ±0.1℃±2.0%RH | Halogen free | DFN2X2-4 | MLYWXA | Tape & Reel | 3,000 | -40℃~+125℃

https://en.sensylink.com/html/web/product/wsdcgq/cwsdcgq/1694267611119996930.html

Datasheet does not describe how to work with it via I2C.
https://en.sensylink.com/upload/1/net.sensylink.portal/1717568489156.pdf

Pinni3, 2024-11-20:
If you need testing or firmware extract just let me know.

Pinni3, 2024-11-20:
> Right now Google only finds these descriptions: [CHT8310 Advanced Datasheet Ver1.1 20230927.pdf](https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8310%20Advanced%20Datasheet%20Ver1.1%2020230927.pdf) [CHT8315 Advanced Datasheet Ver1.6 20230927.pdf](https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8315%20Advanced%20Datasheet%20Ver1.6%2020230927.pdf) [CHT8305C Advanced Datasheet Ver2.5 2021-10-26.pdf](https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8305C%20Advanced%20Datasheet%20Ver2.5%202021-10-26.pdf)
> 
> (https://www.elektroda.com/rtvforum/viewtopic.php?p=21106861#21106861)
> 
> Copied to github...

I see but none of them is the same shape as the one used in the model i have picture of.
Shape is the same as in the non advanced pdf

pvvx, 2024-11-20:
> I see but none of them is the same shape as the one used in the model i have picture of.
> Shape is the same as in the non advanced pdf

To work with CHT8320 (mark "ML"), you need a description of the I2C protocol and coefficients for calculating temperature and humidity. But they are not in the shortened datasheet from the Sensylink website.

Pinni3, 2024-11-20:
I have checked on wechat, no extra documentation contact was only mail even via wechat.
I did send them (support) a mail, hoping on a reply from them.

pvvx, 2024-11-20:
Test firmware
[TH05F_v19_test.zip](https://github.com/user-attachments/files/17834257/TH05F_v19_test.zip)

Sensor ID unknown.
Run Service->Command with parameter "05" and publish the response from the log.

Pinni3, 2024-11-20:
18:39:50: Command response (05): ffffffff44
Also => Humidity is now 99%
[image]

Pinni3, 2024-11-20:
**TH05F_v19t2.bin :**
[image]
[image]
[image]

Pinni3, 2024-11-20:
[divadiow] from the [Elektroda forum](https://www.elektroda.com/rtvforum/viewtopic.php?p=21106861#21106861) is also asking from his side.
Lets hope on some quick emails with the advanced info in PDF

tomasz-kos, 2024-11-20:
If that helps, I2C communication with official software and monitor by phone (from power on and next 7 measurement every few seconds:
write to 0x44 ack data: 0x37 0x81 
read to 0x44 ack data: 0x59 0x59 0x53
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x64 0x69 0x64 0x76 0x0E 0xFF
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x64 0x34 0x56 0x77 0x5C 0x17
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x63 0xF7 0xEC 0x78 0x2D 0x47
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x63 0xDC 0x80 0x78 0x21 0x3A
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x64 0xFE 0xCA 0x79 0xF7 0x6C
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x6A 0xF8 0x01 0xCC 0x85 0x10
write to 0x44 ack data: 0x24 0x00 
read to 0x44 ack data: 0x66 0x97 0x20 0x93 0x70 0x19

Measurement shown on phone:
23.6 46
23.5 46
23.3 46
23.2 46
24 47
28.1 79
25.1 57

Pinni3, 2024-11-21:
[CHT832X Advanced  Datasheet Ver1_NO-watermark.pdf](https://github.com/user-attachments/files/17840405/CHT832X.Advanced.Datasheet.Ver1_NO-watermark.pdf)

Hope this contains the info you need.

pvvx, 2024-11-21:
[TH05F_v19t3.zip](https://github.com/user-attachments/files/17853301/TH05F_v19t3.zip)

Pinni3, 2024-11-22:
Think wrong IC2 adress ? 
[image]

[image]
and if needed the 05 service responce
[image]

pvvx, 2024-11-22:
I don't have a TH05Y v3.1 thermometer and I don't have the ability to check the firmware for all sensor variants. There are already 4 sensor types to service, plus subtypes... Polling different types during initialization can cause some sensors to fail... It is necessary to select the polling sequence by types during initialization, etc.

I'll try to make another version later...

Pinni3, 2024-11-22:
I understand the situation.
If i knew where to start and modify i would certainly help.
But i am clueless atm.

Just for identification purposes and other people (to be temporary warned) the came from this shop:
[RightSitu Official Store](https://www.aliexpress.com/store/1102474203?sortType=bestmatch_sort)
This[ item](https://nl.aliexpress.com/item/1005005234497671.html?spm=a2g0o.order_list.order_list_main.47.4cce79d2mB8XMP&gatewayAdapt=glo2nld)

If you need any help testing, etc just let me know ;-)

pvvx, 2024-11-22:
Apparently, the sensor did not respond to the MID query command, since other types were polled before and this confused it...

pvvx, 2024-11-22:
[TH05F_v19t4.zip](https://github.com/user-attachments/files/17867690/TH05F_v19t4.zip)

Pinni3, 2024-11-22:
i think we have a winner!
10:21:54: Commandoreactie (05): 5959832044
[image]
[image]
[image]

Pinni3, 2024-11-22:
So sensor is working, now just letting it sit next to original FW that came straight out of the packaging.
Waiting to see if there is a big difference.
Will let them sit for 1 or so and check back in.
[image]

Pinni3, 2024-11-22:
After 1 hour temps settled in and i can say it looks good!
[image]
Any more changes needed ?  because it is working fine now.
Thank you for the help!

pvvx, 2024-11-22:
It is necessary to measure average and peak consumption. Mainly duration of activity and what the sensor consumes - what contributes to the total consumption. The documentation does not clearly describe when the CHT832x sensor goes into sleep mode. It is advisable to do all measurements using PowerProfiler, not a tester... That is, a professional approach to measurement is needed...

pvvx, 2024-11-22:
[Version 2.0(beta) published.](https://github.com/pvvx/THB2)

tomasz-kos, 2024-11-22:
Hi,
I measured current with power profiler:
- average vith 5s update is 20uA (display enabled but not connected)
- sleep/idle about 14uA (as above) about 10uA with display disabled
- max for BT transmision about 22mA

I noticed 2 things:
- in normal mode BT transmission is every 5s (ok) but I2C communication is also every 5s once make conversion once read value, but in connected mode I2C is only every 10s - I assume that is expected behaviour
- I tried changing TX power but for all settings current for BT transmision is the same

pvvx, 2024-11-22:
> in normal mode BT transmission is every 5s (ok) but I2C communication is also every 5s once make conversion once read value, but in connected mode I2C is only every 10s - I assume that is expected behaviour

The measurement interval is set in BLE advertising periods.
Before the last advertising period, the sensor starts measuring - a measurement command is transmitted. 
In the next SoC activity cycle (after last advertising period) measurements are read from the sensor.

norbiv8, 2024-11-22:
Hi, my device just arrived, quickly flashed the latest FW. Measurement attached. The consumtion is pretty good IMHO.
[image]

pvvx, 2024-11-22:
10uA in sleep mode is a lot.
The main reason is Tuya - A high-power LCD driver chip is installed.

Normal consumption during sleep for thermometers with CR2032 is up to 6 μA. At the same time, the average, measured over several BLE periods and multiples of them, is no more than 14 μA. Otherwise, CR2032 will not work for even 8 months.

newwolf, 2024-11-22:
How did you flashed this version of the device?
Because I also received 1 this week and tried it the bleutooth methode and at the moment with 2 different usb connection on rx-tx and rx2-tx2 but still getting:
=================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!

Pinni3, 2024-11-22:
Set the RST-RESET connection also.
if you use the script adjust to the correct COM port that u are using.

newwolf, 2024-11-22:
Did that also and it resets because when connecting the screen when connected it resets.
Also twisted the pins no-go.
Maybe both are not working correct but also use them for esp32 etc.

tomasz-kos, 2024-11-25:
> If this is PowerProfiler II from nRF, then these are emissions from the range switch of the measurement circuit. In reality, there are peaks of about 8 mA (RF TX +0 dBm). These nRF glitches do not significantly affect the average measurement. But they distort the current picture. 

Yes I'm using Power Profiler Kit II and I'm doing that in my normal job (design battery operated sensors). I don't understand your  current graphs - I would suspect that measuring tool is broken.

My graphs:
10s measurement:
[image]

Zoom to transmision wakeup:
[image]

Sleep current:
[image]

Based on my experience with BT development that graft of current for transmission are correct (as 3 advertisement transmision on channels 37, 38 and 39)

pvvx, 2024-11-25:
Noted artifacts PowerProfiler II:

[image]

Real TH05Y_V1
[image]
On the simplest meter, INA226:
[image]

pvvx, 2024-11-25:
TX RF ~ 6 mA
[image]

Operation of the built-in PHY6222 DC-DC:
[image]

tomasz-kos, 2024-11-25:
> This is a measurement of the linearly increasing current signal on PPII from a precision generator (>24 bits linearity). [#84 (comment)](https://github.com/pvvx/THB2/issues/84#issuecomment-2494335946)

Thank you for explanation, now I understand.

pvvx, 2024-11-25:
Next, let's look at what distortions PPK2 introduces:
[image]
This is the current at the output of PPK2, with the range switch in PPK2:
[image]

paule999, 2024-12-05:
It looks like I have this particular variant.  I originally used BOOT_TH05_v19.hex &  TH05F_v19.bin and got the zerodisplay and came looking... saw that v20.bin fixes it (and it does) albeit temporarily before reverting to zeros shortly later. .. I'm guessing I need to replace boot firmware with v20 too... I notice there isn't currently a v20 OTA (update_boot) file available... is one likely or do I need to do it via uart again? 
Thanks for any help

pvvx, 2024-12-05:
There are no changes in the BOOT_TH05_V20.HEX for TH05.
https://github.com/pvvx/THB2/tree/master/bin

paule999, 2024-12-05:
Thank you for the reply! ...that's strange though... I tried using  BOOT_TH05_V20.HEX to update boot via OTA but kept getting "Inappropriate format of the firmware file size PHY6 OTA!" 

I assumed there was a different file for updating via OTA but there must be something else going on 😖.

pvvx, 2024-12-06:
You need TH05**F**.

[TH05F](https://pvvx.github.io/TH05F) | BOOT_**TH05F**_vNN.hex | TH05F_vNN.bin | TH05Y_V1.1, TH05Y_V1.2, TH05Y_V3.1 (chip: QD01 2332 NT)
-- | -- | -- | --

Beta -version 2.0 has not yet been completed.
There will be changes.

The repository is upgraded.

paule999, 2024-12-06:
> You need TH05**F**.
> [TH05F](https://pvvx.github.io/TH05F) 	BOOT_**TH05F**_vNN.hex 	TH05F_vNN.bin 	TH05Y_V1.1, TH05Y_V1.2, TH05Y_V3.1 (chip: QD01 2332 NT)
> 
> Beta -version 2.0 has not yet been completed. There will be changes.
> 
> The repository is upgraded.

Thank you very much for your time and effort with this! The F version you put up a little earlier worked perfectly. (Just in case you didn't know the smiley face is AWOL).... that's definitely not a criticism of any sort btw!... I'm very grateful you share all your hard work👍

pvvx, 2024-12-06:
> (Just in case you didn't know the smiley face is AWOL) 

https://dictionary.cambridge.org/ru/%D1%81%D0%BB%D0%BE%D0%B2%D0%B0%D1%80%D1%8C/%D0%B0%D0%BD%D0%B3%D0%BB%D0%B8%D0%B9%D1%81%D0%BA%D0%B8%D0%B9/awol

PS: I don't speak English and it's difficult and time-consuming for me to describe everything through Google Translate - it doesn't understand Russian well.

closed 2024-12-11 by pvvx