pvvx/THB2 #85: TH05Y Flashproblem

cheesemountain0815, 2024-11-23:
Ich bekomme beim Versuch zu verbinden diesen Fehler

C:\Python3>python rdwr_phy62x2.py -p COM8 -e -r wh BOOT_TH05_v19.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!

pvvx, 2024-11-23:
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!

pvvx, 2024-11-23:
The RTS or DTR pin from the adapter must be connected to "REST_N" on the thermometer board.

cheesemountain0815, 2024-11-23:
RX TX 3V3 is connected to the adapter 
ret_n must be plugged in? and then remove

pvvx, 2024-11-23:
> RX TX 3V3 is connected to the adapter

Is GND (minus power supply) connected?

> ret_n must be plugged in? and then remove

For automatic start of programming "REST_N" must be connected to the adapter's RTS or DTR pin.
https://github.com/pvvx/THB2?tab=readme-ov-file#firmware

cheesemountain0815, 2024-11-23:
GND is plugged in, doesn't work either

cheesemountain0815, 2024-11-23:
one more question, how can I change the build rate

cheesemountain0815, 2024-11-23:
now this issue

Connecting...
Traceback (most recent call last):
  File "C:\Python3\rdwr_phy62x2.py", line 738, in <module>
    main()
    ~~~~^^
  File "C:\Python3\rdwr_phy62x2.py", line 575, in main
    if not phy.Connect(args.baud):
           ~~~~~~~~~~~^^^^^^^^^^^
File "C:\Python3\rdwr_phy62x2.py", line 235, in Connect
    read = self._port.read(6);
  File "C:\Python3\Lib\site-packages\serial\serialwin32.py", line 295, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Zugriff verweigert', None, 5))

pvvx, 2024-11-23:
> one more question, how can I change the build rate

[BAUD rate](https://en.wikipedia.org/wiki/Baud)?
```
python rdwr_phy62x2.py -h
```

cheesemountain0815, 2024-11-23:
what is the exact command to change theBuilds Rate

pvvx, 2024-11-23:
> what is the exact command to change theBuilds Rate

```
> python rdwr_phy62x2.py -h
usage: rdwr_phy62x2.py [-h] [--port PORT] [--baud BAUD] [--allerase] [--erase]
                       [--reset] [--start START] [--write WRITE]
                       {wh,we,wf,er,ew,ea,rc,i} ...

PHY62x2 Utility version 11.03.24

positional arguments:
  {wh,we,wf,er,ew,ea,rc,i}
                        Run rdwr_phy62x2.py {command} -h for additional help
    wh                  Write hex file to Flash
    we                  Write bin file to Flash with sectors erases
    wf                  Write bin file to Flash without sectors erases
    er                  Erase Region (sectors) of Flash
    ew                  Erase Flash Work Area
    ea                  Erase All Flash (MAC, ChipID/IV)
    rc                  Read chip bus address to binary file
    i                   Chip Information

options:
  -h, --help            show this help message and exit
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Set Port Baud (115200, 250000, 500000, 1000000)
  --allerase, -a        Pre-processing: All Chip Erase
  --erase, -e           Pre-processing: Erase Flash work area
  --reset, -r           Post-processing: Reset
  --start START, -s START
                        Application start address for hex writer (default:
                        0x1fff1838)
  --write WRITE, -w WRITE
                        Flash starting address for hex writer (default:
                        0x00005000)
```

```
--baud  115200
--baud  250000
--baud  500000
--baud  1000000

--b  115200
--b  250000
--b  500000
--b  1000000
```

cheesemountain0815, 2024-11-23:
it doesn't work
pyserial-miniterm says

--- Available ports:
--- 1: COM1 'Communications Port (COM1)'
--- 2: COM8 'USB-SERIAL CH340 (COM8)'
--- Enter port index or full name: 2
--- Miniterm on COM8 9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

cheesemountain0815, 2024-11-23:
Have TeamViewer

joschi70, 2024-12-07:
The following solution worked for me:
https://github.com/pvvx/THB2/issues/49#issuecomment-2117516820

cata-lin, 2024-12-21:
I had programming issues too, below story hope to save time and frustration for other people - TLDR: change the programmer (USB-to-serial module) 
My target was TH05F (PHY6222), on the board (PCB) is written TH05Y_V1.2, my system is running Ubuntu:
-find first the serial port for the USB-to-serial conveerter - sudo dmesg | grep tty
-then replace default port with 7dev7ttyUSB0 in the parameters of the python script and run it (make sure you are on the right location to run the code) - sudo python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -e -r wh BOOT_TH05F_v20.hex  
-initially I used USB-to-serial module with PL2303 chip, I checled that is 3V3 logic. The module was used with success to program many ESP8266, ESP32.
-I could not get it to work no matter what, problem was that I could not reset the PHY6222 from the TH05F (tried to connect reset pad to RTS , DTR, I tried to reset it manually, I powered the TH05F from external 3V3, not from 3V3 of the PL2303 etc)
-eventually I tried another USB-to-serial module with CP2101, TH05F reset was connected to RTS and it was powered from 3V3 of CP2101 ('automatic' programming, no manual reset)
That worked without issues, I read all the advices from the github, but was not needed to change any python code since the USB-to-serial module was the problem.