pvvx/THB2 #86: ZTH05ZTU_V1.2 device support?

ccetin222222222222222222222222, 2024-11-28:
Hello, I have a tuya Zigbee thermometer I bought a year ago. It was having issues with z2m and ha, so I left it.

Screen looks the same as zth05 1.4, but the board looks a bit different.
[image]
[image]
[image]

pvvx, 2024-11-28:
It has a "Tuya MCU" chip installed.
This chip controls the LCD driver and sensor.
The firmware of the "ZTU" module will not change anything.

ccetin222222222222222222222222, 2024-11-28:
That sucks. I was hoping to flash a new firmware to this device so I can use it at all with my ha, right now it's completely useless. I will probably just throw it out and buy a different module. 

Thank you for the help anyway

pvvx, 2024-11-28:
https://pvvx.github.io/TS0601_TZE204/

gomble, 2024-12-07:
> https://pvvx.github.io/TS0601_TZE204/
> 
> > I will probably just throw it out and buy a different module.
> 
> The сhinese from Tuya did this to sell it, and then the user will throw it in the trash.

does it work with the firmware here?

pvvx, 2024-12-19:
https://github.com/pvvx/ATC_MiThermometer
+ [Modification for custom firmware](https://pvvx.github.io/TS0601_TZE204/)

closed 2024-12-19 by pvvx