pvvx/THB2 #89: How do I find out if it works?

bobybobybob, 2024-12-01:
How do I find out if it works?

My purchased sensor has the most similar layout to THB1 and THB3

Only with the installed THB3 (both files flashed via USB) a display with the values appears. 

With the web browser PHY62x2 BTHome I can establish a connection with Bluetooth and see the values for seconds, which then disappear again and no longer appear... unless I press the button and reconnect. 

In such a Youtube video, however, it can be seen that these values should be permanently visible

04:29:28: Searching for devices
04:30:34: Waiting for connection to THB3-A78721
04:30:37: Model: THB3
04:30:37: Firmware: github.com/pvvx
04:30:37: Hardware: 001A
04:30:37: Software: V1.9
04:30:37: Device info # hw: 001A, sw: 0019, services: 000043B8, sd: 0000
04:30:37: Connected.
04:30:41: Device is disconnected.

[image]
[image]
[image]
[image]
[image]

pvvx, 2024-12-01:
THB3 does not have a "BLE" icon on the screen.

bobybobybob, 2024-12-03:
Thanks for the hint. I thought it was also there because the packaging also shows the symbol in the example pictures.

closed 2024-12-03 by bobybobybob