pvvx/THB2 #91: THB2 flashing problem

polemius, 2024-12-03:
I am trying to flash THB2. I double checked that everything connected as should be.
And I am getting this error:

```
~ THB2 git:(master) ✗ python3 rdwr_phy62x2.py -p /dev/cu.usbserial-1420 -b 115200 -e -r wh BOOT_THB2_v19.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!
```

I wrote a test to check if it is even receiving something:

```python
ser = serial.Serial(
    port=port,
    baudrate=baud,
    timeout=1
)

print(f"\nTesting baud rate: {baud}")

ser.setDTR(True)  # TM (lo)
ser.setRTS(True)  # RSTN (lo)
time.sleep(0.1)
ser.flushOutput()
ser.flushInput()
time.sleep(0.1)
ser.setDTR(False)  # TM (hi)
ser.setRTS(False)  # RSTN (hi)

test_cmd = 'UXTDWU'
print(f"Sending: {test_cmd}")
ser.write(test_cmd.encode())

for _ in range(5):
    response = ser.read(10)
    if response:
        print(f"Received: {response}")
        print(f"Hex: {response.hex()}")
        print(f"ASCII: {response.decode('ascii', errors='replace')}")

ser.close()
```

and with that script I am getting:

```
~  THB2 git:(master) ✗ python3 test_baud.py
Testing port /dev/cu.usbserial-1420 at 115200

Testing baud rate: 115200
Sending: UXTDWU
Received: b'SDK Versio'
Hex: 53444b2056657273696f
ASCII: SDK Versio
Received: b'n ID 00030'
Hex: 6e204944203030303330
ASCII: n ID 00030
Received: b'102 \nMAX_N'
Hex: 313032200a4d41585f4e
ASCII: 102
MAX_N
Received: b'UM_LL_CONN'
Hex: 554d5f4c4c5f434f4e4e
ASCII: UM_LL_CONN
Received: b' 1 , GATT_'
Hex: 2031202c20474154545f
ASCII:  1 , GATT_
```

If I am adding line `print(read)` to line 236 after `read = self._port.read(6)` I am getting this:

```
~  THB2 git:(master) ✗ python3 rdwr_phy62x2.py -p /dev/cu.usbserial-1420 -b 115200 -e -r wh BOOT_THB2_v19.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
b'SDK Ve'
b'rsion '
b'ID 000'
b'30102 '
b'\nMAX_N'
b'UM_LL_'
b'CONN 1'
b' , GAT'
b'T_MAX_'
b'NUM_CO'
b'NN 2\nr'
b'fClk 0'
b' rcClk'
b' 1 sys'
b'Clk 2 '
b'tpCap['
b'33 2b]'
b'\nsizeo'
b'f(stru'
b'ct ll_'
b'pkt_de'
b'sc) = '
b'8, buf'
b' size '
b'= 1876'
b'\nsizeo'
b'f(g_pC'
b'onnect'
b'ionBuf'
b'fer) ='
b' 1876,'
b' sizeo'
b'f(pCon'
b'nConte'
b'xt) = '
b'644, s'
b'izeof('
b'largeH'
b'eap)=4'
b'096 \n['
b'REST C'
b'AUSE] '
b'0\n [00'
b'000000'
b'00 ty '
b'E][app'
b'_key.c'
b':179] '
b'------'
b'app_ke'
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!
```

Could someone help me with that? Thanks

pvvx, 2024-12-03:
"UXTDWU" is transmitted at 9600 baud rate.

polemius, 2024-12-03:
Thank you for so quick response!

On 9600 I am getting this

```
Sending: UXTDWU
Received: b'b\xff\xc7[\x92g[F)\xda'
Hex: 62ffc75b92675b4629da
ASCII: b��[�g[F)�
Received: b'\xc6\xf7\x7f\xcc\xf7\tI\xbed\x19'
Hex: c6f77fccf70949be6419
ASCII: ����     I�d
Received: b'\x92\xc5ak\xa6\xeb\x83)A\x1d'
Hex: 92c5616ba6eb8329411d
ASCII: ��ak���)A
Received: b'\xcb~?\xfe\x1f\xdf?\xfe\xfe\xbf'
Hex: cb7e3ffe1fdf3ffefebf
ASCII: �~?��?���
Received: b'??_\xfe\xbf\xbf??\xfd\x81'
Hex: 3f3f5ffebfbf3f3ffd81
ASCII: ??_���??��
```

and if I change back START_BAUD to 9600 (sorry, forget to mentioned that I changed it to 115200) I am getting this:

```
➜  THB2 git:(master) ✗ python3 rdwr_phy62x2.py -p /dev/cu.usbserial-1420 -b 115200 -e -r wh BOOT_THB2_v19.hex
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
b'c\xff\xff\xff\x98\x82'
b'S0f\x83\xb5\x7f'
b'\xa2\x85\xb7\xd9\xf5\xfe'
b"'\x86\xff\xffN3"
b'O\xc6\x11\xb3{]'
b''
b''
b'\x81\xd2\xf3Oi\xfe'
b'\x1f\xff\xff\xfe_\xfe'
b'~\xbf??_\xfe'
b'\xfe_?\xfe\xda"'
b'\xc5\x89\x85\xd1\xfe\xfe'
b'\xfe\xfe\xfe\xfe\xfe\xfe'
b'\xfe\xff\xff\xfe?\xf6'
b"\xc5\xe6'1\xfd"
b''
b''
b''
b'E;\x1b\xbf\xff\xc5'
b'\xfe\x89'
b''
b''
b''
b''
b''
b''
b''
b''
b'\xc5\xa6\x90S\xff\xa2'
b'+\x9b\xfd'
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
b''
PHY62x2 - Error Reset!
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!
```

Thanks

pvvx, 2024-12-03:
The TM pin on the chip should not be connected anywhere.
But you can try connecting TM to Vcc or GND and check other baud rates...

polemius, 2024-12-03:
The problem was that I was trying to do it on MacOS. Thank you for the help.

bgolab, 2024-12-20:
What is the name of the reset pin on the THB2 board?
Available pin as pictured are: VCC, GND, ADC, RX, TX, TM, TX2
Thank you

pvvx, 2024-12-20:
> What is the name of the reset pin on the THB2 board?

"RESET" 
[image]

bgolab, 2024-12-20:
Too big to notice;)
Thanks.

bgolab, 2024-12-20:
I wonder what is the timing: how long does the MCU waits for entering bootloader through the UART.
I cannot enter the reading mode (Win10, pyserial, 115200:

C:\Users\abg015\Desktop\tbh2>"C:\Program Files\Python313\python.exe" rdwr_phy62x2.py -p COM5 -b 115200 -r rc 0x11000000 0x80000 ff_thb2.bin
=========================================================
PHY62x2/ST17H66B Utility version 12.12.24
---------------------------------------------------------
Connecting...
PHY62x2: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset error! Response: b''
Check connection TX->RX, RX<-TX, RTS->RESET and Chip Power!

bgolab, 2024-12-20:
I am trying to reset the board manually...

bgolab, 2024-12-20:
What about this procedure:
https://developer.tuya.com/en/docs/iot/burn-and-authorize-phy6222-chip?id=Kaufzz2tosqam
Pull pin 8 (TM) high and power on the device to make the UART receive commands.
UART configuration: 115200 baud rate, 8 bit, 1 stop bit, none parity, and no flow control.

pvvx, 2024-12-20:
Connect RX, TX, +3.3V.
When starting the script, connect the GND wire from the adapter to the device.

bgolab, 2024-12-20:
Found the issue - replaced the adapter and now I can enter the flashing mode of the THB2 each time I release the RESET pin following the sequence:
-ground the RESET
-start the script
-release the reset (maybe it happen at the same time I run the script)

C:\Users\abg015\Desktop\tbh2>"C:\Program Files\Python313\python.exe" rdwr_phy62x2.py -p COM5 -b 15200 -r rc 0x11000000 0x80000 ff_thb2.bin
=========================================================
PHY62x2/ST17H66B Utility version 12.12.24
---------------------------------------------------------
Connecting...
PHY62x2: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
Reopen COM5 port 15200 baud... ok
Read 0x11006200...

pvvx, 2024-12-20:
> What about this procedure:

https://github.com/pvvx/THB2/issues/44#issuecomment-2075297537

bgolab, 2024-12-21:
Summarizing (finally successful efforts to flash THB2) for those who care :
1. The manual way of resetting the THB2 is OK. I could enter the Programming Mode almost each time I released the RESET pin following the sequence: 
-**ground the RESET** (used a milti-meter cable with alligator clip connected to the battery spring on one side but the sharp side of the cable was used to touch the pin - **this way my second hand was used to start the script**)
-**start the script** (remember to install pyserial using the pip3 earlier - NOT serial module)
-**release the RESET** pin **as soon as possible after starting the script**
2. It worked fine with cheap ESP-01 programmer (the one used to program ESP-01 ESP8266 modules). Only Vcc/GND and RX/TX is wired.