pvvx/THB2 #93: Error Flash Boot THB1

EMI2-2, 2024-12-05:
Hi, when I try to flash the boot, I get this error. The firmware backup, I have done well, but I can not put the boot:

$ python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -e -r wh BOOT_THB1_v19.hex 
=========================================================
PHY62x2 Utility version 11.03.24
---------------------------------------------------------
Connecting...
PHY62x2 - Reset Ok
Revision: b'001364c8 6222M005'
FlashID: 1364c8, size: 512 kbytes
PHY62x2 - connected Ok
Traceback (most recent call last):
  File "/home/emi/THB1/rdwr_phy62x2.py", line 738, in <module>
    main()
  File "/home/emi/THB1/rdwr_phy62x2.py", line 663, in main
    hp = ParseHexFile(args.filename)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/emi/THB1/rdwr_phy62x2.py", line 56, in ParseHexFile
    taddr = (int(hexstr[3:7],16))
             ^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 16: ''

pvvx, 2024-12-05:
Please download the firmware file correctly.

EMI2-2, 2024-12-05:
Thank you, but I don't understand.

I have downloaded BOOT_THB1_v19.hex and BOOT_THB1_v20.hex (Always the same error).

I have made a backup, installed the THB1_v19.bin firmware without errors, but the display is not showing. I have recovered the backup which is a bin and it works fine.

I think I have the problem with the .hex

What do I have to download again, my model is THB1.

regards

EMI2-2, 2024-12-05:
SOLVED, thank you.
I got it, I downloaded all the code and not just the firmware.

Thank you very much.

pvvx, 2024-12-06:
> I have downloaded BOOT_THB1_v19.hex and BOOT_THB1_v20.hex (Always the same error).

[image]

EMI2-2, 2024-12-06:
Thank you very much, everything works very well. Perfect.

closed 2024-12-06 by pvvx