pvvx/THB2 #94: Keychain with chip ST17H66 (iSearching)

pvvx, 2024-12-09:
### Key fob on chip ST17H66 with firmware "KEY2"

## Switching to “FindMy” mode

* Run [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html)

[image]

---

* To find a device in the connection list, press or hold the button on the key fob.
Select "Connect" and press the button on the key fob again.

[image]

---

* Set the "FindMy" key:
[image]

---

* Set the “FindMy” mode and the desired beacon transmission interval:
[image]

---

* Disconnect...

Hunter32R, 2024-12-09:
I have such key fobs, but they have ST17H66T chip. Is support planned?

pvvx, 2024-12-09:
ST17H66T is a chip without the ability to reflash. It uses one-time programmable memory, which is produced at the factory.

Hunter32R, 2024-12-09:
Thanks for the information.

biemster, 2024-12-09:
This looks great, going to try this as soon as I find my Lenze programming jig. Do you mind if I link to this on biemster/FindMy?

pvvx, 2024-12-09:
# Программирование брелка с ST17H66B

Потребуется адаптер USB-COM с выходами на 3.3В

### Талица соединений:

| USB-COM | PCB брелка |
|--|--|
| GND | GND |
| +3.3V | +3.3V |
| TX | P10 |
| RX | P9 |

### Пример строки запуска скрипта:

```
python rdwr_phy62x2.py -p COM5 -e -r wh BOOT_KEY2_v20.hex
```
Остальные варианты описаны в [README](https://github.com/pvvx/THB2) 

## Последовательность программирования.
1.	Включить USB-COM
2.	Произвести соединения согласно таблице.
3.	Запустить скрипт и быстро соединить провод питания +3.3В от USB-COM адаптера. Если прошивка не началась, отключить и снова подключить провод питания. Возможны и другие варианты  – при старте скрипта кратковременно отключать провод GND от адаптера к брелку.

pvvx, 2024-12-10:
# Интеграция в Home Assistant.

После прошивки брелка прошивкой “KEY2” в Home Assistant отобразится новое устройство:

[image]

Добавляем и нажимаем кнопку на брелке – появится новое Событие:  “Button”.

[image]

Брелок зарегистрирован.

## Переключение на шифрованную рекламу  BTHome BLE v2 (encrypted).

1. Производим соединение с брелком в [PHY62x2BTHome.html](https://pvvx.github.io/THB2/web/PHY62x2BTHome.html).
2. В меню “Service”  назначаем BindKey. Можно использовать изначально сгенерированный прошивкой случайный BindKey  -  тогда используем “Прочитать” BindKey. Копируем BindKey в буфер обмена.
3. В меню “Config” жмем `Прочитать`, включаем галку “Шифрованная реклама”, далее `Записать`.
4. Отключаем соединение: кнопка `Отключение`.
5. Жмем кнопку на брелке – в Home Assistant появится предложение установить BindKey. Копируем BindKey из буфера обмена.

[image]

На этом всё – теперь брелок работает с шифрованной рекламой.

olivluca, 2024-12-10:
My advertisement keys are 28 bytes (see [here](https://github.com/biemster/FindMy/blob/7bf6b5171746a258220d7680bf7688dbc6785427/generate_keys.py#L29) but when I try your flasher it complains that it must be 22 bytes.

olivluca, 2024-12-10:
In fact nrf connect shows 28 bytes, the first six  are 38 1f 8d 09 af  89 and the remaining 22 are the ones I put in your flasher.

pvvx, 2024-12-10:
Firmware (v2.0 beta4) and PHY62x2BTHome.html program (v1.8) have been updated.
The key is entered in the "Base64" format.
The FindMy beacon has been supplemented with battery status transmission.

olivluca, 2024-12-11:
Really nice, with the latest firmware flashed I actually got a report from Apple :+1: .
How does it compare to @biemster's  [implementation](https://github.com/biemster/FindMy/tree/main/Lenze_ST17H66) (regarding battery life and reach of the beacon)?

pvvx, 2024-12-11:
Depends heavily on the beacon transmission interval.
With the same interval, there is no difference with the firmware https://github.com/biemster/FindMy/tree/main/Lenze_ST17H66.

olivluca, 2024-12-11:
Cool, I set a 3s advertising interval but I see @biemster's code uses 5s, I'll change it.
Besides, I see that the device can do double duty (working both as a bthome button in ha and as a findmy tracker). Not that I'm going to use it that way but it's really cool nevertheless.

biemster, 2024-12-12:
Is there a way to protect OTA access, with a password or something? I would not like if someone else passes by and changes the key..

biemster, 2024-12-12:
> If the button is not pressed, it is impossible to connect. The FindMy beacon does not have a connection request reception...

ah sorry, I missed that! I just flashed an E2XT2319, as mentioned in the issue above, which went fine. But it does not have a button :D

pvvx, 2024-12-12:
Button processing (FindMy mode):

When the button is pressed, LED turns on, the FindMy beacon switches to transmitting BLE advertising with the AdvEventType = LL_ADV_CONNECTABLE_UNDIRECTED_EVT attributes. A first packet of BLE advertising events is transmitted in the quantity specified in "Number of event transmissions". The period of advertising events is 95 ms. Data in the packet is in BTHome format with "Button" = "1". 

After the packet has been transmitted for N*95ms, the LED goes out. If the button is still pressed, the first packet is transmitted again. If the button is released, the second packet of BLE "Number of Event Transmissions" announcements is transmitted, but with "Button" = "0".

At the end of the transmission of the second packet, the speaker quietly clicks and the transmission of the FindMy beacon with the attribute AdvEventType = LL_ADV_NONCONNECTABLE_UNDIRECTED_EVT begins.

omarkhali, 2024-12-12:
> @biemster - Now, to support "Find My" in Home Assistant, you'll have to fight with the writers of ["Bluetooth" integration](https://www.home-assistant.io/integrations/bluetooth). But there you'll be sent to "Bluez", and there you'll be sent to the kernel, and there's Linus Torvalds :P
[
](https://github.com/malmeloo/hass-FindMy)

omarkhali, 2024-12-12:
https://github.com/malmeloo/hass-FindMy

biemster, 2024-12-12:
> @biemster - Now, to support "Find My" in Home Assistant, you'll have to fight with the writers of ["Bluetooth" integration](https://www.home-assistant.io/integrations/bluetooth). But there you'll be sent to "Bluez", and there you'll be sent to the kernel, and there's Linus Torvalds :P

:sob:

omarkhali, 2024-12-12:
> @pvvx @biemster https://github.com/malmeloo/hass-FindMy

This integration works beautifully and I use it with hass. Many thanks to @biemster @pvvx @malmeloo On this hard work

biemster, 2024-12-12:
I forgot how frustrating it is to program these chips, I'm on it for three hours now and managed a grand total of 2!

The third one I flashed only half, @pvvx your OTA bootloader does not replace the entire bootloader right?

biemster, 2024-12-12:
I'm just installing BOOT_KEY2_v20.hex. Getting the chip to start in firmware upload mode has always been an issue for me, probably due to the hacky setup I'm using. When the flasher gets to `cmd>>` it actually finishes in seconds, but getting to that is very finicky.

The question was if flashing BOOT_KEY2_v20.hex only partially due to lost connection will brick the chip?

pvvx, 2024-12-12:
Flash writing on PHY62x2/ST17H66B chips is always available.
UART Boot is in ROM.

pvvx, 2024-12-12:
Flash bootloader program via UART located in ROM can be blocked by encryption key in Flash.
Then the response will not be `cmd>>:`, but `fct>>:`.
With `fct>>:` only command to erase entire Flash works. Then will be `cmd>>:` again.

biemster, 2024-12-12:
That's interesting, I read the code of your flasher and found a lot more functionality then I extracted from LeKit :)

My issue however is that the flashing procedure overloads my USB uart, which is also powering the chip:
```
[25046.566999] cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
```
This gets spammed in dmesg k times when I apply the 3v3 of the usb-uart to the chip.

olivluca, 2024-12-12:
> Firmware installation via USB-COM adapter takes several minutes with soldering of wires.

No need to solder using [some pogo pins](https://github.com/biemster/FindMy/issues/14#issuecomment-1438305392) (I'm using a much more rudimentary version but it works).
@biemster I use the usb to ttl part of an esp32 board (keeping the EN pin connected to GND), apparently its 3.3V regulator is powerful enough to flash the keychain.

biemster, 2024-12-12:
Just finished flashing three of those:
[image]
which have an E2XT2319 (?). No need for a button, when you add the battery after flashing you can connect and set the key + interval, then all is set.

This is absolutely brilliant.

lovelyelfpop, 2024-12-13:
I bought 4 iTags, none of them is ST17H66B, so sad

pvvx, 2024-12-13:
> I use the usb to ttl part of an esp32 board (keeping the EN pin connected to GND), apparently its 3.3V regulator is powerful enough to flash the keychain.

The reason for the high consumption of "iSearching" during programming is in the electrical circuit. The speaker key (Buzzer) is connected to the P9 pin. During programming, P9 is the input of the "RX" adapter and the speaker is on 90% of the time, which creates a high current consumption.

lovelyelfpop, 2024-12-13:
Anyway to make the buzzer work? AirGuard app can found airtags around, it can make them play sound. This fork https://github.com/AeroX2/st17h66_FindMy support buzzer, but it surely not the way airtags do it.

lovelyelfpop, 2024-12-13:
And it would be greate if this firmware implement key rotation. I found that if a tag with this firmware and an iPhone meet at the same place every morning, the location of the tag will not be reported by the iPhone. Even if the iPhone and the tag have been to other places the day before. I guess this is related to key rotation. The location of my other tags with nRF5x firmware(50 keys) get updated more frequently.

pvvx, 2024-12-13:
There is no description of the key rotation algorithm yet.

biemster, 2024-12-13:
There is a description in the [openhaystack paper](https://arxiv.org/abs/2103.02282), and also [FindMy.py](https://github.com/malmeloo/FindMy.py/) is able to deduce the current airtag key from the registered data on macOS, but what @lovelyelfpop probably meant is uploading N keys and just start broadcasting the next after let's say 15 minutes.

Since uploading a bunch of keys might be cumbersome with the web flasher, we could also use one key as base, and after every time interval either add the curve generator to it (basically private key +1), or multiply by 2 (private key *2) with the latter being easier to implement. Although since this will be done very rarely efficiency should not be an issue.

omarkhali, 2024-12-13:
> This integration does not receive the Find My beacon.
> There is no way to determine that the Find My carrier has appeared at home or in the yard, in a specific room ...
> The "bluetooth" integration does not accept beacons without "flags" in the PDU. At the same time, in the standard, unspecified flag keys are accepted as the value 0 by default. But Bluez and kernel (Linus Torvalds) have their own standards.
> This also involves "D-Bus"... And it is impossible to move this entire chain. Especially since Linus Torvalds has gone into politics and imposed sanctions on the Russians :)
> In Linux, in "Bluez", in "Bleak" Bluetooth version 5.0+ is still not supported since 2016.

HI @pvvx Is this the love you are looking for?

https://github.com/agittins/bermuda

pvvx, 2024-12-13:
Added musical accompaniment :)
[image]
I'm too lazy to write music - [I copied some pieces](https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/buzzer.c#L84)...

pvvx, 2024-12-13:
> HI @pvvx Is this the you are looking for?
> 
> https://github.com/agittins/bermuda

The link offers an unnecessary device that consumes several watts - _using **ESPHome bluetooth_proxy** devices_

omarkhali, 2024-12-13:
> Added musical accompaniment :) ![image](https://private-user-images.githubusercontent.com/12629515/395610085-24282790-12c7-4928-aeb7-dc6d6c927d84.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQxMTI3NjEsIm5iZiI6MTczNDExMjQ2MSwicGF0aCI6Ii8xMjYyOTUxNS8zOTU2MTAwODUtMjQyODI3OTAtMTJjNy00OTI4LWFlYjctZGM2ZDZjOTI3ZDg0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjEzVDE3NTQyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwOTc1YTdjZWIxN2I0ZThjYmMzOGYyODhkOGY0NDkwNmQ4OWI0MTA1ZjcwNzg5NTQ3MDUyYjhjMDJmYjdhODgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mgdB5nKLtWc66s5_BQPFcuzfpmUPucS7WoE3UCBFfYc) I'm too lazy to write music - [I copied some pieces](https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/buzzer.c#L84)...
> 
> The buzzer is turned off by the button release event.

Is it possible to add button buzzer in homeassistant

pvvx, 2024-12-13:
> Is it possible to add button buzzer in homeassistant

To do this, you need to write some kind of integration for "HA". The main problem with integrations for "HA" is that it requires constant support. "HA" is constantly changing and users always have thousands of questions. Support takes a lot of time and not everyone has it.

pvvx, 2024-12-14:
I poked around in "Passive BLE Monitor Integration" and:
[image]
[image]
:)

biemster, 2024-12-14:
I'm not a HA user, but this might change my mind

pvvx, 2025-01-01:
Unknown.

yousaf465, 2025-01-01:
> Unknown.

should I program with RX and TX on board only?

nobodyspecial, 2025-01-01:
What is the model of the chip? (small one in the second pic) I have the same keychain and unfortunately mine has the ST17H66T chip which is not flashable.

nobodyspecial, 2025-01-01:
> > What is the model of the chip? (small one in the second pic) I have the same keychain and unfortunately mine has the ST17H66T chip which is not flashable.
> 
> Can't read at least with my phone camera, let me try with my microscope I do have a program pad under p06 in the second pic.

Yeah, I had to use a magnifying glass and a flashlight lol.

pvvx, 2025-01-01:
[Option with OTP on PHY6230.](https://pvvx.github.io/iSearching3/)
Has one-time programming by the manufacturer.

pvvx, 2025-01-01:
> Can't read at least with my phone camera, let me try with my microscope

The markings don't say anything. Especially on OTP chips. The chip marking may indicate the OTP firmware order number.

yousaf465, 2025-01-01:
> Yeah, I had to use a magnifying glass and a flashlight lol.
BK3431 
CG5111f

[image]
[image]

pvvx, 2025-01-01:
> I have the same keychain and unfortunately mine has the ST17H66T chip which is not flashable.

ST17H66T is PHY6230.

[image]

[image]

nobodyspecial, 2025-01-01:
> ST17H66T is PHY6230? [...]

Earlier in this thread you said:

>ST17H66T is a chip without the ability to reflash. It uses one-time programmable memory, which is produced at the factory.

Is that not true?

pvvx, 2025-01-01:
Phy6230, having OTP, meets this ID:
```
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'00000000 6230CK05'
```

biemster, 2025-01-01:
> BK3431 CG5111f
 
that's a beken 3431, and should be programmable. I'm working on those, but did not receive mine yet

pvvx, 2025-01-01:
PHY6xxx is a Chinese SoC design. It is sold to other companies or labeled for the customer.

yousaf465, 2025-01-01:
> that's a beken 3431, and should be programmable. I'm working on those, but did not receive mine yet

Should I flash with PVVX firmware? Use TX and RX pins only?

pvvx, 2025-01-01:
> Should I flash with PVVX firmware? Use TX and RX pins only?

There is no firmware for the BK3431 chip.

biemster, 2025-01-01:
> > Should I flash with PVVX firmware? Use TX and RX pins only?
> 
> There is no firmware for the BK3431 chip.

@yousaf465 an SDK seems to be here: https://github.com/yumzhi/ble
I'll start working on this once my chips arrive (although they are bk3432, but I'm hoping they are similar enough)

MartinEssink, 2025-01-07:
> [Option with OTP on PHY6230.](https://pvvx.github.io/iSearching3/) Has one-time programming by the manufacturer.

[image]
I have one with a similar layout, but the pinout seems slightly different. Did you read the data from the SOP8 one over uart, and if so, what pins did you use?

pvvx, 2025-01-07:
[image]

khongpt, 2025-01-23:
> > [Option with OTP on PHY6230.](https://pvvx.github.io/iSearching3/) Has one-time programming by the manufacturer.
> 
> [image] I have one with a similar layout, but the pinout seems slightly different. There is no marking on the chip. Did you read the data from the SOP8 one over uart, and if so, what pins did you use?

Looking at the PCB, it seems similar to the one I have. It looks like it's the OTP type that can't be re-flashed.

[image]

pvvx, 2025-01-26:
> прошивка BOOT_KEY2_v20  через адаптер USB-COM заливаеться и все работает как описано у вас
> но после заливки через веб страницу KEY2_v20 брелок перестаёт работать

Что "заливается" через веб страницу?

rif63, 2025-01-26:
конекчусь через блютус 
20:45:54: Waiting for connection to KEY-B27243
20:45:59: Model: KEY
20:45:59: Firmware: github.com/pvvx
20:45:59: Hardware: 0020
20:45:59: Software: B2.0
20:46:00: Device info # hw: 0020, sw: 0020, services: 00600009, sd: 0000
20:46:00: OTA ver: 01
20:46:00: Connected.
20:46:00: Set device time...
20:46:00: Set device time (23089F9667)...

далее ота app firmware : key2_v20.bin start
20:45:59: Model: KEY
20:45:59: Firmware: github.com/pvvx
20:45:59: Hardware: 0020
20:45:59: Software: B2.0
20:46:00: Device info # hw: 0020, sw: 0020, services: 00600009, sd: 0000
20:46:00: OTA ver: 01
20:46:00: Connected.
20:46:00: Set device time...
20:46:00: Set device time (23089F9667)...
20:48:21: Download firmware file 'KEY2_v20.bin'...
20:48:22: File: bin/KEY2_v20.bin
20:48:22: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 46560
20:48:22: Filesize: 46564 bytes
20:48:22: Counter: 2911 blocks
20:48:25: Download firmware file 'KEY2_v20.bin'...
20:48:25: File: bin/KEY2_v20.bin
20:48:25: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 46560
20:48:25: Filesize: 46564 bytes
20:48:25: Counter: 2911 blocks
20:48:27: Download firmware file 'KEY2_v20.bin'...
20:48:27: File: bin/KEY2_v20.bin
20:48:27: File id:PHY6, Segments: 3, Start: 0x"1FFF1838, : 46560
20:48:27: Filesize: 46564 bytes
20:48:27: Counter: 2911 blocks
20:50:00: Start OTA flashing...
20:50:47: Flash finished in 46.39 seconds
20:50:51: Device is disconnected.

 и все тухнет 
сканер не видит брелок
хотя с прошивкой воот видел

pvvx, 2025-01-26:
Проверил только что:
```
18:55:42: Поиск устройств
18:55:57: Ожидание соединения с KEY-C84EB4
18:56:18: Model: KEY
18:56:18: Firmware: github.com/pvvx
18:56:18: Hardware: 0020
18:56:18: Software: V2.0
18:56:18: Device info # hw: 001E, sw: 0020, services: 00600008, sd: 0000
18:56:18: Устройство подключено.
18:56:43: Загрузка firmware файла 'KEY2_v20.bin'...
18:56:43: Файл: bin/KEY2_v20.bin
18:56:43: Файл id:PHY6, Сегментов: 3, Старт: 0x"1FFF1838, : 46560
18:56:43: Размер файла: 46564 байт
18:56:43: Счетчик: 2911 блоков
18:56:48: Переключение на...
18:56:48: Переподключение
18:56:48: Устройство отключено.
18:56:48: Ожидание соединения с KEY-C84EB4
18:56:52: Устройство отключено.
18:56:52: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
18:56:52: Переподключение 0 / 5
18:56:52: Ожидание соединения с KEY-C84EB4
18:56:56: Model: KEY
18:56:56: Firmware: github.com/pvvx
18:56:56: Hardware: 0020
18:56:56: Software: B2.0
18:56:56: Device info # hw: 001E, sw: 0020, services: 00600009, sd: 0000
18:56:56: OTA ver: 01
18:56:56: Устройство подключено.
18:56:59: Старт программирования...
18:57:46: Программирование завершено за 45.449 секунды
18:57:50: Устройство отключено.
18:57:54: Переподключение
18:57:54: Ожидание соединения с KEY-C84EB4
18:57:57: Model: KEY
18:57:57: Firmware: github.com/pvvx
18:57:57: Hardware: 0020
18:57:57: Software: V2.0
18:57:57: Device info # hw: 0020, sw: 0020, services: 00600008, sd: 0000
18:57:57: Устройство подключено.
```

[53 more replies]