pvvx/THB2 #95: Serial Flash Problem - ModuleNotFoundError: No module named 'serial'

AndyW999, 2024-12-11:
e:\Users\Andy\Downloads\THB2-master\THB2-master>pip3 install -r requirements.txt
Requirement already satisfied: serial in e:\users\andy\appdata\local\programs\python\python39\lib\site-packages (from -r requirements.txt (line 1)) (0.0.97)
Requirement already satisfied: future>=0.17.1 in e:\users\andy\appdata\local\programs\python\python39\lib\site-packages (from serial->-r requirements.txt (line 1)) (1.0.0)
Requirement already satisfied: pyyaml>=3.13 in e:\users\andy\appdata\local\programs\python\python39\lib\site-packages (from serial->-r requirements.txt (line 1)) (6.0.1)
Requirement already satisfied: iso8601>=0.1.12 in e:\users\andy\appdata\local\programs\python\python39\lib\site-packages (from serial->-r requirements.txt (line 1)) (2.1.0)

e:\Users\Andy\Downloads\THB2-master\THB2-master>pip3 install -r requirements.txt
Requirement already satisfied: pyserial in e:\users\andy\appdata\local\programs\python\python39\lib\site-packages (from -r requirements.txt (line 1)) (3.5)

e:\Users\Andy\Downloads\THB2-master\THB2-master>python3 rdwr_phy62x2.py
Traceback (most recent call last):
  File "e:\Users\Andy\Downloads\THB2-master\THB2-master\rdwr_phy62x2.py", line 5, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'

Any ideas?

AndyW999, 2024-12-13:
I never thought I would say this but it was easier with Linux!

Used an RPi4 and the bootloader went straight in.

Then Edge on Win10 for the update.

My board is TH05Y_v1.2 24-01-15 and TH05F_v20 works fine.

I also get a Count and Opening Sensors showing on HA.

Thanks

Andy.

pvvx, 2024-12-13:
You don't have PATH set for "Python\Scripts" on Windows.

AndyW999, 2024-12-13:
OK - thanks!

closed 2024-12-13 by AndyW999