pvvx/THB2 #99: Idea: raw screen acces

dakhnod, 2024-12-24:
Thanks for your great work on this!

It would be awesome if we could get raw access to the screen via BLE.
Here is a suggestion how we could send the data to a characteristic:
- byte 0-4: timeout in milliseconds (after this time the screen displays temperature and humidity again)
- byte 4-x: the screen data, one bit per screen segment

This way, we could render text to the LCD segments in the application, and just send the data to the screen!

What do you think?

pvvx, 2024-12-25:
There is no idea here. Everything was planned long ago.

All firmwares have the "60" function.

"60xxxxxxxxxx" - output screen buffer to LCD.
The screen buffer contains a bitmap of LCD segments.
"60" - disable.

But so far only the TH05F firmware disables the measurement output when the "60" function is displayed on the screen.

The function "22" will be added in the next version. See the implementation https://github.com/pvvx/ATC_MiThermometer

dakhnod, 2024-12-25:
Oh, my bad I didn't know.
I was thinking of includnig something like [gpioASM](https://github.com/dakhnod/gpioASM).
This would allow for setting a timeout for a bitmap.
Furthermore, this would allow for sending sequences of bitmaps, for example to have a longer text running on the screen.

Any thoughts on that?

pvvx, 2024-12-25:
The LCD screen cannot display most letters. The LCD is designed to display numbers and a few symbols.

dakhnod, 2024-12-25:
Okay, a few questions regarding this:
1. How can I set a timeout on a bitmap sent to the device?
2. How can I, after sending a bitmap, return the display to normal operation and display temperature?

pvvx, 2024-12-25:
"60xxx...xxx" - output screen buffer "xx..xx" to LCD.
The screen buffer contains a bitmap of LCD segments.
"60" without data - disable this function.

But so far only the TH05F firmware disables the measurement output when the "60" function is displayed on the screen.

dakhnod, 2024-12-25:
So, can I set a timeout for the bitmap?

pvvx, 2024-12-25:
Not yet. -> _The function "22" will be added in the next version._
Any add-ons increase battery consumption. And not everyone needs such functions.
You are provided with source codes to create your own functions. Create what you need.

dakhnod, 2024-12-25:
I understand, thanks for the discussion!

Speaking about the battery life, what is the expected battery life of the TH05F? That should be the V3.1, right?

pvvx, 2024-12-25:
The functions "60" and "22" will be implemented in the coming days for all versions with LCD.

dakhnod, 2024-12-25:
What is the expected battery life of the TH05F?

pvvx, 2024-12-25:
More than 8 months and default settings and:
1. When using the cheapest CR2032 battery.
2. Without frequent connections to the device.
3. When the external adapter is operating in passive scanning mode.

pvvx, 2024-12-25:
TH05F with CHT8305 transmits incorrect measurements when the battery voltage is below 2.5V. The CHT8305 sensor has a working voltage of up to 2.5V and is not suitable for use with CR2032.
[image]
Due to its low cost and simplified design, the use of CR2032 in Tuya thermometers is limited to 40% of the battery capacity.

dakhnod, 2024-12-25:
How about the TH05Y_V3.1?

pvvx, 2024-12-25:
The difference is not big. No more than 10%.
All claims to Tuya.

dakhnod, 2024-12-25:
thanks tuya

dakhnod, 2024-12-25:
So, turning off the LCD helps?

pvvx, 2024-12-25:
By changing all settings it is possible to increase the operating time approximately twofold. But the dependence on the quality of the CR2032 battery is still greater.
If you modify the circuit to install the correct capacitor in the power supply, it will increase the operating time by 50%.

pvvx, 2024-12-25:
MJWSD05MMMC (CR2450) when working in the "Le Long Range" mode (200 meters with obstacles). Consumption when transferred 6..8 times more (by transfer time, but not by power): 
[image]

dakhnod, 2024-12-25:
> Le Long Range

LE coded?

pvvx, 2024-12-25:
Comparison "LE Long Range" with WiFi.

Ble receiver - USB -BT RTL8761BU without an external antenna. Located in the house. The signal passes through 2 walls.
The Wi -fi router is in the attic, in direct sight with Wi -fi "Smart Plug".

WiFi "Smart Plug" and Th03z are located in one place in the garage. The outer wall of the garage is a thin metal sheet.
WiFi "Smart Plug" was chosen with the maximum quality of receipt and transmission of a dozen different ones. The rest of the "Smart Plug" did not support work at such a distance...
[image]

dakhnod, 2024-12-25:
Should I just add a big enough capacitor to the battery in the thermometer? That should dampen the spikes, right?

pvvx, 2024-12-25:
> Should I just add a big enough capacitor to the battery in the thermometer? That should dampen the spikes, right?

 https://github.com/user-attachments/files/18102754/swra349.pdf

pvvx, 2024-12-25:
The "22" function has been added to the latest version 2.0.

[image]

"227b002d003c0000" 
Show: 

12.3  
 45 

Display 60 seconds.

closed 2024-12-25 by pvvx