pvvx/THB2 #100: TH05Y v3.1 display not working

madzur, 2024-12-26:
After having flashed my newly bought TH05 (PCB says TH05Y_V3.1) with the firmware version BOOT_TH05_v18.bin were originally showing all zeroes and the onboard display was not working properly. Having brought the device online, I was able to flash the current firmware v19 and the beta firmware v20. With both versions, the temperature and humidity readings over BThome in home assistant work fine, however the onboard display is still not working.

I tried swapping the internals with another TH05 PCB with the factory firmware and the display works fine with the factory internals.

Attached, I am sending the chips, that are onboard of the TH05Y PCB

[image]
[image]
[image]

After digging through the documentation, I realized it is most likely a TH05F, so I flashed the corresponding TH05F firmware. However I have been unsuccessfully trying to flash with several combinations of BOOT and APP and I am getting one of 3 outcomes:

1. I can get the display working, however the temperature and humidity readings both on the LCD and over BTHome read all zeroes (TH05F_v18.bin and TH05F_v19.bin)
2. I can get the temperature and humidity readings working in BThome, however the display stays blank (TH05F_v20.bin)
3. Flashing any version of the BOOT firmware usually gets the temperature and humidity readings working, however the display stays blank.

ewpa, 2024-12-26:
I was about to raise the same issue.  Reverting to previous commit 09f742e2 v20 resolved this in my case.

pvvx, 2024-12-26:
Fixed, TH05F v2.0 firmware updated.

closed 2024-12-26 by pvvx

ewpa, 2024-12-26:
> Fixed, TH05F v2.0 firmware updated.

Thanks @pvvx, confirmed fixed in my case with commit 1484313.

madzur, 2024-12-27:
Thanks. Works for me as well.