pvvx/THB2 #102: TH05Y Bluetooth Not Advertising

rspeir, 2024-12-28:
I have a TH05Y board. The silkscreen on it reads "TH05Y_v3.1 2024-10-26".

I successfully flashed the custom bootloader onto the board (BOOT_TH05F_v20.hex), but when I attempt to connect to it via the PHY62x2BTHome.html website, I do not find it in the list of available bluetooth devices. 

The display reads "BOT 20" when I power it on with the pushbotton pressed, and it will then display the temperature and humidity, but I do not see any bluetooth devices named "TH05*" advertised anywhere. 

I was also able to successfully manually load the OTA firmware via serial, so now when the device boots, the display shows "APP 20", but again, no bluetooth. 

Does anyone have suggestions?

Is there a serial terminal that will print debugging information or error logs?

pvvx, 2024-12-28:
> I successfully flashed the custom bootloader onto the board (BOOT_TH05F_v20.hex), but when I attempt to connect to it via the PHY62x2BTHome.html website, I do not find it in the list of available bluetooth devices.

You did not specify in which operating system the actions take place and which BT adapter is used.
Linux may require changing settings to comply with Bluetooth standards, rather than the default settings that do not comply with standards.

> The display reads "BOT 20" when I power it on with the pushbotton pressed, and it will then display the temperature and humidity

The display shows "APP nn" or "BOT nn" when the power is turned on, not when a button is pressed.

rspeir, 2024-12-29:
After scanning for BT devices and refreshing several times, I was able to connect to the device and get it talking to Home Assistant. Unfortunately, the battery ran empty after 3 hours. I think I have a bad hardware revision that's not suitable for this firmware, so I will revert it to the stock FW.

closed 2024-12-29 by rspeir