pvvx/THB2 #103: "Close" event a counter?

bgolab, 2024-12-29:
Hi,
I am an happy user of THB2 running your firmware. 
I wonder if you could consider adding "Close" event counter to the already available stats.

PS. Assumed that the "OUT" field shown in the stats is not the "Close" event counter - maybe I am wrong?

Thank you.

pvvx, 2024-12-29:
Why do need an additional counter if have "Open" and "Closed" events?
"Count" contains the "Open" count.

[image]

[image]

bgolab, 2024-12-29:
BLE events can be missed (I know you repeat them 5 times)

I would like to use the THB2 to know if something is Opened or Closed without tracking BLE events for the above reason.

I could track the counters to conclude what is the current state of the OR better I would like to read the current state of the pin input... You could also advertise the state of a pin periodically.

pvvx, 2024-12-29:
> BLE events can be missed (I know you repeat them 5 times)

5..255
[image]

> I would like to use the THB2 to know if something is Opened or Closed without tracking BLE events for the above reason.

How is it possible to know the status without tracking events?

> You could also advertise the state of a pin periodically.

The Open/Close sensor always transmits its state periodically.

Are you creating something completely incomprehensible?

bgolab, 2024-12-30:
If the  Open/Close sensor is transmitted periodically that's perfectly fine. I missed this point when was looking for BTHome packet structure here.

Writing about avoiding tracking events I assumed (wrongly) that the "input change" is transmitted once so missing such event would be catastrophic. In case the input state is transmitted periodically everything is perfectly fine.

Yes, I may use unreliable hw - currently RPI 0 is capturing BLE advertisements using your hcitooladv tool. Yes, RPI0 is busy with other tasks.

Talking about my plans: I would like to add these kind of BLE devices to my HA for other tasks than T/H reporting. So adding the digital capabilities like "pin state" is one of the plan (to get a feedback on some important devices' state).

Another idea is to read analog input (TX/P20 in the case of THB2 is an analog input) - I have not studied the schematic to see what are the circuit limitation here but it would be interesting enhancement to such devices I guess - It would make them more capable IoT sensors.

Thank you.

bgolab, 2024-12-30:
I was considering to compile the project for me but it requires some time to prepare the dev env so I am trying to re-use what you have now (the builds).

As you noticed it required to prepare small lab to measure power consumption, etc.

At the moment I am considering the THB2 one of the best choice as IoT sensor platform (cheap, has decent temp sensor, really low power consumption, etc). 

As DIY alternative for me is also IN100 based dfrobot Fermion:
https://www.dfrobot.com/product-2765.html?srsltid=AfmBOopuIM3VeKuSVaNKBPj7Rjxq814nJO6Hv71x_ZwFeBkYoeD8-QWd

It is more generic (you can add arbitrary sensor to the IN100), you can define the BLE packets without explicit programming.
Tested, works fine.

bgolab, 2024-12-30:
The feature I need is already implemented (the pin status is advertised periodically.)

closed 2024-12-30 by bgolab

pvvx, 2024-12-30:
Filename:      thb2_main.h
```
// adv. event
#define RDS_EVENT_STEP_SEC		1800 // transfer step (state duplication) RDS 30 minutes without events
#define RDS_EVENT_DOUBLE_SEC	(RDS_EVENT_STEP_SEC - 12) // transmit (state duplication) RDS 12 sec after event
#define RDS_EVENT_START_SEC		(RDS_EVENT_STEP_SEC - 120) // transmission (state duplication) of RDS from power on - 2 minutes
```

bgolab, 2024-12-30:
Out of curiosity - I wonder what was the original use case of advertising RDS status quite rarely?

Why not to include the 'Opening' status in every regular packet just following other data? 
This would potentially open new possibilities as the status would be considered almost real-time for some applications?

The battery won't be affected here?

bgolab, 2024-12-30:
Currently the packet size is around 62 bytes, adding 2 bytes for the purpose of advertising the input state increases the size by 3.3%, and I guess consequently the transmission time by the same percent... Of course I have not taken into account the linearity you mentioned...

Not sure about the battery life of the CR2032 but the IN100 chip current consumption is real/true (there are some products on the market which are badly designed e.g. earlier version of the DFROBOT module using IN100 chip - the power chip is wrongly implemented there and this particular board consumes more than 1mA; other boards - two from SparkFun and the latest DFROBOT Fermion is OK - they keep promises from the IN DS).

Anyway, thank you for the discussion.

pvvx, 2024-12-30:
It says there - several years.
The internal resistance of the CR2032 battery by 40..60% of its service life can already be 100 Ohm.
In the datasheet for the IN100 chip - the transfer current is 10 mA.
10 mA 100 Ohm = drop (sag) of the supply voltage from 3V to 2V.
As a result, the battery is only 40% functional.
What years are we talking about there?

bgolab, 2024-12-31:
Not sure why we are discussing CR2032 here - I only mentioned IN100 chip. 

Power issues and internal resistance of the power source are pretty well known - see super cap solutions for car batteries or other cases where we need to add capacitor to mitigate power issues.

I believe IN100 has no worse power requirements than other BLE chips(good when in sleep mode, and similar when in active mode in 10mA range). I used it with AAA batteries.

ESP32 power requirements are well known. Mostly affected by bad power chip consuming a lot of power when in low power mode (see comparison between XIAO EPS32C3 vs C6 where changing the power chip helped to save a lot of power - charts are available in many places in the Internet.

In the Low Power designs the average power consumption matter (sleep mode vs active mode rates) - yes, yes I remember the peak power issues. Spent many hours measuring it using different methods.

Let's stop the discussion here. I believe we have similar experiences in this area. Or you are much better than me in this area if you will.

Thank you for the discussion.

bgolab, 2024-12-31:
One more thing: I really appreciate what you have done here in this project.

pvvx, 2024-12-31:
I read what was indicated by your link to the N100 advertising. And surprised at such a deception.