pvvx/THB2 #104: Attempting to connect via the webpage...

trippme, 2025-01-01:
When I scan for devices my BTH01 (at least that's what it's labeled as at [Aliexpress](https://www.aliexpress.us/item/3256807158732570.html)) identifies itself as a TH05 so when I connect it fails.

Is there a way to work around this?
[image]
[image]
[image]

pvvx, 2025-01-01:
You use firmware from another device.

trippme, 2025-01-01:
Thanks for the response, appreciate the help, I'm very NOOB on this!

I didn't select the firmware this is just what shows up when I try to discover the device.

How would I use firmware from another device?

brianfinley, 2026-01-18:
> Thanks for the response, appreciate the help, I'm very NOOB on this!
> 
> I didn't select the firmware this is just what shows up when I try to discover the device.
> 
> How would I use firmware from another device?

I have the same device (eight of them) and am experiencing the same issue.  Will try to write via serial port, but would be nice to be able to do OTA.

pvvx, 2026-01-18:
> Will try to write via serial port, but would be nice to be able to do OTA.

Please provide technical documentation on BLE OTA for this device.