pvvx/THB2 #105: BTH_01 battery voltage adjustable

ExploWare, 2025-01-02:
First: thanks for your work.
The advice with the bad battery management of this specific device did make me rethink it and modify the powersource:
I have a rechargeable outdoor PIR light with a decent LiPo battery. 
This 3.7v (avg) is apparently not a problem,  it's been working for a while without issues. 
Now, the firmware reports a battery level in Volts and in Percents.
This percentage is calculated to the 2x AAA level. This of course is no longer valid. 
So, my question/feature requeat:
Would there be a feasible way to have this adjustable over Bluetooth or perhaps have a second branch of firmware releases? (I assume this is produced easily by CI/CD? That part I could assist)

Thanks for reading this

LarenPL, 2025-08-20:
I agree with this request. The integrated circuit itself supports voltage measurement up to 3.6V, and this is sufficient for 100% and 3V for 0%. Please share the compiled code. I know which file to change this in, but I can't compile it. Please help or provide instructions.

pkkrusty, 2025-08-26:
Related note: Can BTH-01 handle a full 4.2v Lithium cell?

pvvx, 2025-08-27:
> Related note: Can BTH-01 handle a full 4.2v Lithium cell?

No.  Only up to 3.6V.

LarenPL, 2025-09-04:
I need to use the BTH-1 to monitor multiple power banks. I'm primarily interested in low-battery information. Can you help pvvx?

pkkrusty, 2025-09-05:
Very interesting that it is working at 4.2v. Thus taping a 18650 to it, it will last for many years probably.

LarenPL, 2025-09-05:
Sure, but I want to use it to monitor power banks. I'll also make the concession that the voltage measurement is limited to 3.6V. My main concern is informing HomeAssistant that a power bank connected to another device needs to be charged.

pvvx, 2025-09-06:
> It has also been operating stably at a supply voltage of 4.2V for several weeks now. 

Then plug it into the 110/240V network. Maybe it will work too :)

There are other battery monitoring projects.

Controller using INA226 to provide a standalone Zigbee router - Measures battery voltage and current:
[image]
Firmware for TB-03F module: [TB_INA226_v53.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/bin/TB_INA226_v53.bin)

Uninterruptible power supply controller for the system with "Home Assistant":

[image]

Manages both charging and balancing of the battery.
Monitors all voltages and currents.
https://github.com/pvvx/ATC_MiThermometer/issues/638

pvvx, 2025-09-06:
> Very interesting that it is working at 4.2v. Thus taping a 18650 to it, it will last for many years probably.

Most 18650 have high self-discharge and the battery protection circuit consumes more than 3uA.
The chip will degrade at more than 3.6V....

LarenPL, 2025-09-07:
Can You help with this ?

`00:50:41 **** Incremental Build of configuration Default for project bthome_phy6222 ****
make all 
CC: source/main.c
process_begin: CreateProcess(NULL, arm-none-eabi-gcc -mcpu=cortex-m0 -mthumb -mthumb-interwork -Os -g3 -ggdb -W -Wall --std=gnu99 -fno-diagnostics-show-caret -fdata-sections -ffunction-sections -funsigned-char -funsigned-bitfields -fms-extensions -specs=nosys.specs -Wl,--gc-sections -D__GCC -DCFG_SLEEP_MODE=PWR_MODE_SLEEP -DDEBUG_INFO=0 -DMTU_SIZE=240 -DMAX_NUM_LL_CONN=1 -DDEF_GAPBOND_MGR_ENABLE=0 -DTEST_RTC_DELTA=1 -DLL_DEBUG_NONE=1 -DCLK_16M_ONLY=1 -DSTACK_MAX_SRAM=1 -DBROADCASTER_CFG=0x01 -DOBSERVER_CFG=0x02 -DPERIPHERAL_CFG=0x04 -DCENTRAL_CFG=0x08 -DHOST_CONFIG=0x04 -DARMCM0 -DOSAL_CBTIMER_NUM_TASKS=1 -DENABLE_LOG_ROM_=0 -DOSALMEM_METRICS=0 -DPHY_MCU_TYPE=MCU_BUMBEE_M0 -DUSE_FS=0 -DOTA_TYPE=OTA_TYPE_NONE -I./source -I./SDK/misc -I./SDK/misc/CMSIS/include -I./SDK/components/arch/cm0 -I./SDK/components/ble/controller -I./SDK/components/ble/include -I./SDK/components/ble/hci -I./SDK/components/ble/host -I./SDK/components/inc -I./SDK/components/osal/include -I./SDK/components/common -I./SDK/components/profiles/ota_app -I./SDK/components/profiles/DevInfo -I./SDK/components/profiles/SimpleProfile -I./SDK/components/profiles/Roles -I./SDK/components/driver/adc -I./SDK/components/driver/bsp_button -I./SDK/components/driver/clock -I./SDK/components/driver/dma -I./SDK/components/driver/flash -I./SDK/components/driver/gpio -I./SDK/components/driver/i2c -I./SDK/components/driver/key -I./SDK/components/driver/kscan -I./SDK/components/driver/led_light -I./SDK/components/driver/log -I./SDK/components/driver/pwm -I./SDK/components/driver/pwrmgr -I./SDK/components/driver/qdec -I./SDK/components/driver/spi -I./SDK/components/driver/spiflash -I./SDK/components/driver/timer -I./SDK/components/driver/uart -I./SDK/components/driver/voice -I./SDK/components/driver/watchdog -I./SDK/components/libraries/crc16 -I./SDK/components/libraries/cliface -I./SDK/components/driver/watchdog -I./SDK/misc/CMSIS/device/phyplus -c source/main.c -o ./build/source/main.o, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:297: source/main.o] Error 2

00:50:52 Build Finished (took 10s.866ms)`

pvvx, 2025-09-08:
```
make -s clean
make -s -j PROJECT_NAME=BTH01_LarenPL PROJECT_DEF="-DDEVICE=DEVICE_BTH01"
```