pvvx/THB2 #107: An attempt to build a the project for THB2 - differences in binary size

bgolab, 2025-01-03:
Hi,

I tried to build the project on WSL2 using probably the latest arm gcc and I get different binary size than yours. I wonder if we use different arm gcc verions?

Some details:
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bthome_phy6222$ apt list | grep gcc-arm-none-eabi
[...]
gcc-arm-none-eabi-source/jammy 15:10.3-2021.07-4 all
gcc-arm-none-eabi/jammy,now 15:10.3-2021.07-4 amd64 [installed]

Executed make for the latest repo:
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bthome_phy6222$ make -s -j PROJECT_NAME=THB2_v20 POJECT_DEF="-DDEVICE=DEVICE_THB2"
LD: build/THB2_v20.elf
OBJCOPY: build/THB2_v20.hex
OBJDUMP: build/THB2_v20.asm
size:
   text    data     bss     dec     hex filename
  49036    1913   11084   62033    f251 build/THB2_v20.elf
  49036    1913   11084   62033    f251 (TOTALS)
Make: build/THB2_v20.bin

Elf file type is EXEC (Executable file)
Entry point 0x1fff1841
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x010000 0x1fff0000 0x1fff0000 0x00400 0x00400 RW  0x10000
  LOAD           0x011838 0x1fff1838 0x1fff1838 0x026d0 0x026d0 R E 0x10000
  LOAD           0x013f08 0x1fff3f08 0x1fff3f08 0x00379 0x02ac8 RW  0x10000
  LOAD           0x000100 0x11010100 0x11010100 0x095b0 0x095b0 R E 0x10000
  LOAD           0x0096b0 0x110196b0 0x110196b0 0x0030c 0x0030c R   0x10000

 Section to Segment mapping:
  Segment Sections...
   00     .jumptbl
   01     .textentry .text
   02     .data .bss
   03     .xip
   04     .rodata

=========================================================
PHY62x2 OTA Utility version 25.02.24
---------------------------------------------------------
---- Segments Table -------------------------------------
Segment: 11010100 <- Flash addr: 00010100, Size: 000098bc
Segment: 1fff0000 <- Flash addr: 000199bc, Size: 00000400
Segment: 1fff1838 <- Flash addr: 00019dbc, Size: 00002a49
---- File Structure -------------------------------------
Segment Table[03] <- Flash addr: 00010000, Size: 00000100
Segment: 11010100 <- Flash addr: 00010100, Size: 000098bc
Segment: 1fff0000 <- Flash addr: 000199bc, Size: 00000400
Segment: 1fff1838 <- Flash addr: 00019dbc, Size: 00002a54
---------------------------------------------------------
Write to file: ./build/THB2_v20.bin 51220 bytes - ok.
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bthome_phy6222$ ls build
SDK  THB2_v20.asm  THB2_v20.bin  THB2_v20.elf  THB2_v20.hex  THB2_v20.map  source

What I got:
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bthome_phy6222$ ls -l build
total 6784
drwxr-xr-x 5 bogdan bogdan    4096 Jan  3 07:43 SDK
-rw-r--r-- 1 bogdan bogdan 4952699 Jan  3 08:02 THB2_v20.asm
-rw-r--r-- 1 bogdan bogdan   51220 Jan  3 08:02 THB2_v20.bin
-rwxr-xr-x 1 bogdan bogdan 1055152 Jan  3 08:02 THB2_v20.elf
-rw-r--r-- 1 bogdan bogdan  143384 Jan  3 08:02 THB2_v20.hex
-rw-r--r-- 1 bogdan bogdan  747850 Jan  3 08:02 THB2_v20.map
drwxr-xr-x 2 bogdan bogdan    4096 Jan  3 07:43 source
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bthome_phy6222$

What you have:
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bin$ ls -al THB2_v20.bin
-rw-r--r-- 1 bogdan bogdan 49044 Dec 28 14:45 THB2_v20.bin

I wonder what is your opinion about this difference.
Thank you,
Bogdan

bgolab, 2025-01-03:
bogdan@ABG015-Z15G8:~$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/10.3.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --enable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:10.3-2021.07-4 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile,aprofile CFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -flto=auto -ffat-lto-objects -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -flto=auto -ffat-lto-objects -fstack-protector-strong' DFLAGS=-frelease FCFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -flto=auto -ffat-lto-objects -fstack-protector-strong' FFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -flto=auto -ffat-lto-objects -fstack-protector-strong' GCJFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -flto=auto -Wl,-z,relro' OBJCFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -flto=auto -ffat-lto-objects -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -ffile-prefix-map=/build/gcc-arm-none-eabi-hYfgK4/gcc-arm-none-eabi-10.3-2021.07=. -flto=auto -ffat-lto-objects -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip SED=/bin/sed SHELL=/bin/sh BASH=/bin/bash CONFIG_SHELL=/bin/bash
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.3.1 20210621 (release) (15:10.3-2021.07-4)

bgolab, 2025-01-03:
-probably your arm gcc version:
GNU_Tools_ARM_Embedded\13.2.rel1\bin
Version 13.2.Rel1
Released: October 30, 2023 

-my arm gcc: 
Version 10.3-2021.10
Released: October 21, 2021 

OK. Need to update it.

pvvx, 2025-01-03:
[GNU_Tools_ARM_Embedded 13.2.rel1](https://github.com/pvvx/THB2/blob/master/bthome_phy6222/mk_windows.cmd#L1)

bgolab, 2025-01-03:
Do you use the arm gcc under windows or linux?

mk_windows-only-THB2.cmd - points to windows (@set PATH=D:\MCU\GNU_Tools_ARM_Embedded\13.2.rel1\bin;%PATH%) but it fails  - MakeFile refers to linux commands (rm etc).

BTW. I do no tuse eclipse as my version (2024.3 fails) for some reason when I wanted to add arm gcc toolchain so wanted to use cli ony. The linuc arm gcc builds the project without error.

bgolab, 2025-01-04:
I refreshed the linux arm-gcc to:
gcc version 13.2.1 20231009 (15:13.2.rel1-2)

And still got different size of the binary file:
bogdan@ABG015-Z15G8:~/thb2/THB2-master/bthome_phy6222$ ls -l build/THB2_v20.bin
-rw-r--r-- 1 bogdan bogdan 51220 Jan  4 10:20 build/THB2_v20.bin

But ok, maybe it is not worth investigating.

bgolab, 2025-01-04:
Not so important

closed 2025-01-04 by bgolab