pvvx/THB2 #109: Error returned when trying to flash BTH01 device

sjoram, 2025-01-05:
OK I know I am not lucky buying devices at this stage!

But I now have some BTH01s and am trying to flash the bootloader using:

    python3 rdwr_phy62x2.py -p /dev/ttyAMA0 -e -r wh BOOT_BTH01_v20.hex

But I am getting the errors below. I have the latest v20 and the latest rdwr_phy62x2.py from the website.

Anybody know what I could be doing wrong?

Cheers

=========================================================
PHY62x2/ST17H66B Utility version 20.12.24
---------------------------------------------------------
Connecting...
PHY62x2: Release RST_N if RTS is not connected...
ST17H66B: Turn on the power...
Chip Reset Ok. Response: b'cmd>>:'
Revision: b'001340c4 6222M005'
FlashID: 1340c4, size: 512 kbytes
PHY62x2 - connected Ok
Traceback (most recent call last):
  File "/home/stephen/Downloads/rdwr_phy62x2.py", line 725, in <module>
    main()
  File "/home/stephen/Downloads/rdwr_phy62x2.py", line 650, in main
    hp = ParseHexFile(args.filename)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stephen/Downloads/rdwr_phy62x2.py", line 56, in ParseHexFile
    taddr = (int(hexstr[3:7],16))
             ^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 16: ''

pvvx, 2025-01-05:
No file `BOOT_BTH01_v20.hex` or downloaded incorrectly

sjoram, 2025-01-05:
> No file `BOOT_BTH01_v20.hex` or downloaded incorrectly

Yes - I just found it was not downloaded properly. Apologies for the entry

closed 2025-01-05 by pvvx