pvvx/THB2 #110: TH05Y_V3.1 - Not seen after successfull uploading of BOOT_TH05F_v20.hex

Bingo600, 2025-01-05:
Hello & Thank you for the FW :-)

I received two Tuya TH05F v3.1 - with the Sensor: CHT8320 
Like bottom pics here - https://pvvx.github.io/TH05F/

I'm using Linux Mint & a FT-232 dongle for programming.
I'm powering the TH05F from the FT-232 , and have wired : 

Tuya           FTDI
TX              RX
RX              TX
NRST          RTS

I'm using this command to flash the bootloader (without issues)
python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -e -r wh ./bin/BOOT_TH05F_v20.hex

The first adapter was flashed w. the bootloader, and i had no issues.
On powerup i saw it without issues in Linux BT.
And also using Brave browser on the "web page" : 
https://pvvx.github.io/THB2/web/PHY62x2BTHome.html

I used the Webpage to perform an OTA without issue ... All is fine - (Thank you)

*** But ***
My 2'nd adapter seems to have an issue - I can not see it in BT after flahing the Bootloader.

I connected the 2'nd TH05F (same PCB V3.1) , and flashed the bootloader without any issues.
Everything looks like the first one.
But when i BT search/scan for it , i never see it.
It's like the BT Xmitter isn't working/active.
I have been trying for hours .... Nothing seen.

I then tried to restore the "Full Tuya" fw you have on the page : 
python3 rdwr_phy62x2.py -p /dev/ttyUSB0 -b 250000 -r we 0 ./ff_tuya_th05f.bin

And now the 2'nd adapter is seen in Linux BT
[Tuya-org-fw.pdf](https://github.com/user-attachments/files/18311004/Tuya-org-fw.pdf)

It's not a big issue, and I would have accepted that it was a "semi" dead Tuya, if it didn't work with the Original firmware.
But it does work with the original firmware ?????

It's strange ....
The pvvx firmware worked super on the first one , but not on the 2'nd one.

I did cleanup the pads, and assembled the 2'nd , to see the display.
It shows Temp + Humi , but no "face"  (it just has bootloader in it).

Any ideas ?

TIA
Bingo

pvvx, 2025-01-05:
1. Before scanning on older adapters, you must press a button to connect.
2. Operating system not specified.

Bingo600, 2025-01-05:
Re 1:
I did not have to do that on the working Tuya
But i have tried that too.

Re 2:
I specified Linux mint above 
Version : Linux Mint 21.3 Virginia 
VERSION_CODENAME=virginia
UBUNTU_CODENAME=jammy

pvvx, 2025-01-05:
Set Bluez parameters according to Bluetooth standards, not as defined by its authors.

Bingo600, 2025-01-05:
Re: Set Bluez parameters according to Bluetooth standards, not as defined by its authors.
I agree it could be that - But why can i then see one of them , and not the other ??

Well - I have turned on a Win11 PC.
After quite some attempts i succeded in seeing the device , and after many more attempts i have connected to the device.
I now have OTA updated the Tuya ... Took around 180 sec.

But it's still strange that my Linux have no issues - Seeing & Connecting to one of the two devices.
And have never seen the other device ...

But i have to agree with you, in that the other/problem device could be seen & detected in Windows.

Thank you for taking your time to help me out here

Bingo

Ps: 
Do you have any idea why my Linux can see one of them , and not the other - They seems to have same hardware.
And why the problem device works with the original Tuya firmware ?

pvvx, 2025-01-05:
> I agree it could be that - But why can i then see one of them , and not the other ??

Because the restriction on the period of the BLE advertising and in other settings that do not correspond to Bluetooth LE is set.
And to save power, the BLE advertising period is set in 5 seconds. The standard describes up to 10 seconds.

Bingo600, 2025-01-05:
If i want "some kind of" gateway/receiver for these BT temp/humi devices , to interface (LAN) with a HA Server.
I expect to have : 8..12 of them

I have ordered a few of these : Low-cost Realtek RTL8761 BT adapters

But i had hoped to use either a Raspberry PI or in the beginning an ESP32.
But i think you wrote that an ESP32 is too slow for passing the data.
And now you write that Linux BT doesn't work reliably.

What do i get/setup ???

pvvx, 2025-01-05:
Realtek RTL8761 BT works for receiving BLE advertising according to the BT5.2 standard. And “Le Long Range” supports the “patch” - switching the adapter through the HCI protocol. But it does not support the connection to “Le Long Range” due to the available API in Linux.

Bingo600, 2025-01-05:
Sigh .....
I have a few ESP32-C3 - Risc-V ??

Well it was a fun excersize, but i'll probably drop these.

If i get some MIJA and convert to Zigbee - Is that better supported on linux ?

pvvx, 2025-01-05:
Linux does not have support for Zigbee. This is done by installing additional software that works through the standard USB-COM drivers to connect the coordinator.

Bingo600, 2025-01-05:
I'll have a look at a zigbee coordinator.
Did you not have some Xiaomi fw, that converted some MIJA temp sensors to zigbee.

And i'll get some more of these :
https://en.lacrossetechnology.fr/P-17-A1-WSTX29D-IT.html
I'm already using these with an RFM-12B 868MHz receiver, connected to a DIY Arduino Gateway. 
They work excellent.
Uses 2 x AA Batteries , and lasts more than a year.

I just have a linux python program that query the Arduino via serial every 30 sec, and saves the sensor data as CSV.

pvvx, 2025-01-05:
> Did you not have some Xiaomi fw, that converted some MIJA temp sensors to zigbee.

All firmware is adapted to the existing software for Home Assistant. And that's all they are limited to.

pvvx, 2025-01-05:
> Uses 2 x AA Batteries , and lasts more than a year - I use NIMH (IKEA or ENERLOOP).

The main page has a diagram showing LYWSD03MMC operating outdoors from a CR2032 since August 10, 2023. 

[image]

There are also graphs showing MJWSD05MMC operating for over a year and a half on one battery (not dead yet) at a distance of 200 meters (LE Long Range)...

Bingo600, 2025-01-05:
That is COOL !!
Does the  LYWSD03MMC run zigbee ??

Since i want a Gateway to poll the devices, and linux (Raspberry) sucks at BT.
Then i have no idea of what to get, to get a decent BT LE Gateway.

Then i might give Zigbee a try.
Do you have any experience with this Zigbee GW - SLZB-06 ?
https://www.aliexpress.com/item/1005006284356429.html?

pvvx, 2025-01-05:
With the same amount of data transfers, Zigbee consumes significantly more than BLE.
Zigbee cannot compare in communication distance with BLE in the "LE Long Range" mode. The "LE Long Range" communication distance is 2 times greater.
And the Zigbee network is not reliable.
That's why I don't use Zigbee for sensors. Only for actuators that run on mains power.

Bingo600, 2025-01-05:
It is not the consumption that worries me.
It is the problem of receiving BT LE and/or BT LE LR.

If i choose to run BT devices, i need to get a device that can receive the above BT protocols and transmit it (LAN) to my HA server.
I have no idea what to get ....

pvvx, 2025-01-05:
That's your problem.
I use only one server with three USB-BTs connected and a ZigBee coordinator to cover a 500 meter radius area and about 200 devices.

pvvx, 2025-01-05:
2 x USB-BT:
[image]
One in "legacy" mode, the second in "LE Long Range" mode.
The third in USB-BT of another integration:
[image]

Bingo600, 2025-01-05:
And those adapters are just running directly on your Home Assistant server ?
What OS do you run your HA under ?

I was hoping to be able to move my HA to Proxmox 
But i suppose i might be able to pass the BT Adapters through

pvvx, 2025-01-05:
Daily HA database write data flow to NVME:
[image]

Bingo600, 2025-01-05:
So that's around 20GB per day 

Well i was hoping to get some info about your  server hardware and OS, that supports BT API & Protocols so nice.
But that's ok.

Thank you for an educational conversation , and thank you for your time spend on the various firmwares.

Have a nice day

pvvx, 2025-01-05:
The previous HA server destroyed a 128GB SSD in 2 years. And this is with filtering out some unnecessary data.

pvvx, 2025-01-05:
> your server hardware and OS

NanoPi-R5S in maximum configuration.

Bingo600, 2025-01-05:
What do you mean self assembled Armbian ??
Did you write your own BT Stack or ?

I have a few Odroids & OrangePi running armbian

pvvx, 2025-01-05:
> Did you write your own BT Stack or ?

No. It's just that the Armbian website doesn't have the latest kernel version and stuff...
Only HA is installed on this device. And it doesn't require connection to BLE devices. The BLE stack is only needed for connection.

Bingo600, 2025-01-05:
What device has your Realtek BT adapters connected ?

pvvx, 2025-01-05:
USB: two USB 3.2 Gen 1 Type-A ports -> usb3.0 hub -> USB2.0 cords -> USB-BT/Zigbee adapters :)

Bingo600, 2025-01-05:
Ahh ... 
So your USB & Zigbee adapters are connected to the nanopi-rs5 using Armbian.

Could you share your Armbian (linux) Bluetooth configuration ?
I mean what modifications you have made to get linux BLUEZ to work with for example TH05's ?

I still can just see one of the TH05F's ....
But i'm using  : 
Bus 003 Device 003: ID 8087:0029 Intel Corp. AX200 Bluetooth

pvvx, 2025-01-05:
No additional changes are required to receive BLE advertising from PHY62x2 devices.
Currently, the firmware does not contain a switch to "LE Long Range" and PHY62x2 operates in BT4.2 compatibility mode.

> Will the Realtek adapters perform better ?

RTL8761BU has a good reception level. In LE Long Range up to -115 dBm. But poor performance - cannot receive hundreds of BLE ads per second...
The theoretical maximum for Legacy mode in BLE is up to 300 BLE ads per second. Only a special sniffer can provide this. Approaching such indicators gives https://github.com/pvvx/ADV_BLE2UART.

That's why one of my USB-BT adapters is CSR8510A10. It's the most productive adapter of the cheap ones, but it only has BT4.2.

Bingo600, 2025-01-06:
Closed 
Apparently Linux has issues connecting to some of the devices, using this firmware.
I succeded in windows.

closed 2025-01-06 by Bingo600