pvvx/THB2 #124: TH05F_v20 and max Advertizing interval

astrogene1000, 2025-01-24:
Hi,

First off, thank you very much for your work on this!

T0 decrease battery usage I have been changing the Advertizing interval

I have only used whole numbers, i.e. 9000, 10000, 11000 up to 15000

What I see is anything over whole number 10000 and the bootup screen, either APP or BOT, never leave the given display and no advertisements are made.

Note I also changed the battery survey interval to match (in whole seconds) the advertising interval.

Is there an upper limit that is supported? Example I see if I change interval to 16000 the setting drops to 15937.5 and 
Measurement steps show 31875.0 msms

9:49:21 AM: Configuration data: 55440000003f**a0**0010020**a0**000
9:49:32 AM: Configuration data: 55440000003f**b0**0010020**b0**000
9:49:39 AM: Disconnect
9:49:39 AM: Device is disconnected.
9:50:57 AM: Reconnect
9:50:57 AM: Waiting for connection to TH05-2
9:51:05 AM: Model: TH05F
9:51:05 AM: Firmware: github.com/pvvx
9:51:05 AM: Hardware: 0019
9:51:05 AM: Software: V2.0
9:51:05 AM: Device info # hw: 0019, sw: 0020, services: 000043B8, sd: 0000
9:51:05 AM: Connected.
9:51:12 AM: Configuration data: 55440000003f**b0**0010020**b0**000
9:51:23 AM: Configuration data: 55440000003f**a0**0010020**a0**000

Gene

astrogene1000, 2025-01-24:
One more piece of the puzzle maybe.
Holding the unit in my hand and measuring the time between when the unit shows a change in temperature.
Setting the advertising to 10000 and battery sample interval to 10 I see a consistent approx 19 seconds between when unit shows the temperature change.

Again, thank you very much for this work!

Gene

pvvx, 2025-01-24:
In the Bluetooth/BLE standard, the maximum interval is up to 10 seconds.

astrogene1000, 2025-01-24:
Thank you!

Closing now!

closed 2025-01-24 by astrogene1000